tasmota/mgos32-to-tasmota32 #7: shelly plus 1

LinuxfarmerHH, 2023-09-24:
Got a new Shelly Plus 1 that did not work pretty over MQTT with hassio. So i tried tasmota and followed point by point the instructions here. At the "optional" point the partition wizard told me that only 64kb of mem is unused, so i skipped this step.

Then it seems i did something evil. Switchet the GPIO4 switch_n to button_ as this is my setup on the shelly.

Tasmota enforced a reboot and did not come back, button was out of order and no WifiAP from the device was there. It's dead.

Jason2866, 2023-09-27:
The safeboot conversion was done? Without next reboot will fail, exactly what you have.
The only way to recover is serial flashing.

LinuxfarmerHH, 2023-09-28:
Every single step from the list was done, the device did reboot 4 times but the GPIO setting was the kill deal. The only thing i missed, was the "optional" step to partition wizard again, because only 64kB was available. Maybe setting the GPIO to button_n exceed the flash memory?
Will keep the device, because i have two USB flashers here.

Jason2866, 2023-09-28:
@LinuxfarmerHH The issue you see is not from the convert process. Probably different board revision which needs different GPIO config. Some tinker work left for you.

Jason2866, 2023-09-28:
Closing since not related to convert.

closed 2023-09-28 by Jason2866