tasmota/mgos32-to-tasmota32 #8: Shelly PlusPMMini: no Partiton Wizard, no Power

daUser, 2023-09-24:
Hi,

im stuck on Step 10:
There is no Partition Wizard.

Also my PlusPMMini doesn't show me any Power. And the Temperature isn't correct.

Are there any solutions?

[image]

Jason2866, 2023-10-01:
@daUser this is needed `so21 1`

closed 2023-10-01 by Jason2866

daUser, 2023-10-04:
Hi,
some Time is gone,.. it's difficult to find the time at the Moment...

I have found the Partition-Wizard - it was behind the Button "Consoles", my fault.

So i have done Step 10 and 11. And then i had this GUI:
[image]
I have done an upgrade to 13.1.0.4 "tasmota32**s3**.bin" from https://ota.tasmota.com/tasmota32/.
After this a second Time `so21 1`.
Now i get the Power Informations, a little bit calibration and this is good.

But: the Analog1 Temperature shows 46,1°C, in reality i have 21,5°C. An at this difference i can't work with 'TemperatureOffset'.