tasmota/mgos32-to-tasmota32 #9: Stuck at Step 10

skl283, 2023-10-01:
Hi,

i following every step, after Step 10, the shelly 1 plus starts and creates a "Tasmota" wlan. If i connect to it, i cant open the webinterface.

Any hints what to do now?

br
sash

Jason2866, 2023-10-01:
Connected to the Tasmota AP? Entered http://192.168.4.1 ?

skl283, 2023-10-01:
Exactly. No Webfrontend opens.

skl283, 2023-10-02:
Sorry, at first i didn't recognized the other question!

I've tried to flash a shelly 1 Plus. Because the first boot, after i put the shelly 1 plus zip to flash, was no peroblem, i don't think that this is an issue with my Device. But i tried also 2 Device (iPhone with Safari and macbook with chrome) - both didn't work. 

WLAN Connection work, but not webinterface at 192.168.4.1 opens. I've cliecked "Start Migration" and then the device restarts (with a new tasmota wlan).

I've tried also to reset the device by pressing the Button - this also didn't work. 

Do you have maybe in addition any hints, what i could do now? Yesterday i've also bought an usb2ttl device - perhaps i could flash directly the device? i

Jason2866, 2023-10-03:
Easiest to solve is to do a wired flash. Hard to say from distance how to solve (if possible) the behaviour you see currently

skl283, 2023-10-03:
Thank you, will try i wired flash as soon as the flasher arrive.

For helping - is it possible to provide logs for further investigation?

Jason2866, 2023-10-03:
If the Flasher has arrived it may help to capture the serial log when the device starts.
Detail log enabled with `seriallog 3`

skl283, 2023-10-04:
hmmm, i don't know if this helps? Since now its the first time working with an esp and flashing it... this comes, if i connect it:

`[21:56:55]00:00:00.002 HDW: ESP32-U4WDH-S 
[21:56:55]00:00:00.005 CFG: Loaded from NVS, Count 6
[21:56:55]00:00:00.010 Project tasmota - Tasmota Version 13.1.0(safeboot)-2_0_11(2023-08-15T07:58:36)
[21:56:55]00:00:00.010 WARNING This version does not support persistent settings
[21:56:55]00:00:00.273 WIF: WifiManager active for 3 minutes
[21:56:56]
[21:56:56]Brownout detector was triggered
[21:56:56]
[21:56:56]ets Jul 29 2019 12:21:46
[21:56:56]
[21:56:56]rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[21:56:56]configsip: 188777542, SPIWP:0xee
[21:56:56]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[21:56:56]mode:DIO, clock div:1
[21:56:56]load:0x3fff0030,len:184
[21:56:56]load:0x40078000,len:12736
[21:56:56]load:0x40080400,len:2716
[21:56:56]entry 0x40080594
[21:56:56]`

Jason2866, 2023-10-04:
okay, you are stuck since your USB Serial adapter does not power the device enough.
`Brownout detector was triggered` tells this and restarts the device.
Use the in Tasmota Documentation recommended USB serial adapter

closed 2023-10-05 by Jason2866