tasmota/mgos32-to-tasmota32 #17: Shelly Plus Wall Dimmer

kunaalm, 2023-10-17:
I am attempting to migrate a Shelly Plus Wall Dimmer to Tasmota (I know it's untested!). I don't see any GPIO mapping after autoconfig. Is the template known / defined somewhere?

Jason2866, 2023-10-17:
You are the first. So no.

tkffaul, 2024-04-21:
I have one as well that I would like to convert. I have the boot loader up and running but still no GPIO mapping?

Jason2866, 2024-04-24:
Since no feedback from users which have the device how the GPIOs are assigned, I can't add them.

tkffaul, 2024-04-24:
I know this is probably more involved, but I'm not a complete noob... can you guide me on how to check GPIO and maybe I can help provide a map? I have a temporary rig setup for this already, i have a serial over USB connector I can hook on to it... I just don't know how to proceed to map the pin sets.

Jason2866, 2024-04-24:
Without open the device this can be a useless attempt. If dimming is done via a second MCU the train stops here.
In this case a complete reverse engineering is needed or a dimming firmware needs to be done. Like it was done for the Shelly Dimmer.
If all work is done by the ESP32 the journey goes on :-)

tkffaul, 2024-04-24:
So if i crack the case and take some photos, that would be helpful? I know shelly has stated that this is very different than the original dimmers... so (just an assumption) i think it might do everything from the ESP32 now.

Jason2866, 2024-04-24:
Yes, photos will help. So no guessing about "who" does the dimming. Maybe it can be seen which GPIOs goes to the FETs

tkffaul, 2024-04-24:
[image]
Back of Switch

tkffaul, 2024-04-24:
[image]
Inside

tkffaul, 2024-04-24:
[image]
Zoom in of the ESP32

tkffaul, 2024-04-24:
[image]

tkffaul, 2024-04-24:
[image]
[image]
[image]

tkffaul, 2024-04-24:
Hope that helps!

Jason2866, 2024-04-25:
Yes, it does help. I think this IC is an MCU
[image]
since it is placed on the AC power PCB it probably does the dimming job and the ESP32 "just tells" to do so.
Did not find any data sheet for this IC

Looks like flashing Tasmota does lead to nowhere. Thx for the photos!!

RavenSystem, 2024-05-09:
It is an STM32S003. Best way is to use a logic analyzer to read UART communication between ESP32 and it, with the original Shelly firmware loaded, of course.

Jason2866, 2024-12-16:
Closing support not planned.

closed 2024-12-16 by Jason2866