tasmota/mgos32-to-tasmota32 #22: Partition Wizard Blank Screen

Gk2E7Aaojf3KDCQ, 2023-11-12:
I have flashed a Shelly Plus Plug US with Tasmota 12.5.0 (mgos32-to-tasmota32 13.1, using Plug US, not Plus Plug as detailed in issue #20). I have followed the guide, using the Plug US (no Plus Plug US available) auto-configuration. I got to the step concerning the parition wizard and hit a brick wall. Whenever I load the partition wizard from the Consoles tab, I get a blank screen and no buttons, options, or anything.

I've tried several versions of the partition_wizard.tapp from the tasmota repository and got the same result every time. Partition_manager.tapp gave similar results, although that at least loaded a warning saying these settings could brick the device. There were no buttons there either. 

Error output from the Console after trying to load the partition wizard for Tasmota 12.5.0:

> 1:41:43.454 BRY: Exception> 'type_error' - unsupported operand type(s) for <=: 'nil' and 'int'
> 01:41:43.459 stack traceback:
> 01:41:43.460 	<unknown source>: in function `factory_migrate_eligible`
> 01:41:43.462 	<unknown source>: in function `show_migrate_to_factory`
> 01:41:43.463 	<unknown source>: in function `page_part_mgr`
> 01:41:43.464 	<unknown source>: in function `<lambda>`

I also attempted to use the Berry console to just list the partitions.

`import partition_core var p=partition_core.Partition() print(p)`

Results:

> <instance: Partition([
> ],
>   <instance: Partition_otadata(ota_active:ota_0, ota_seq=[,], ota_max=1)>
> )>

From what I've read, there should be more partitions. Seems my plug is stuck in partition limbo.

Jason2866, 2023-11-12:
The Partition Wizard is installed automatically when doing the Autoconfig.
When this step went wrong or an update is tried before this step is done it ends in a soft brick. Only serial flash can recover the device

mlschuh, 2023-11-12:
> I have flashed a Shelly Plus Plug US with Tasmota 12.5.0 (mgos32-to-tasmota32 13.1, using Plug US, not Plus Plug as detailed in issue https://github.com/tasmota/mgos32-to-tasmota32/issues/20). I [...]

Can confirm this same problem with both of my Plug US. Followed the instructions exactly, ended up with the same error and soft brick both times. I'll pull out my serial connection and try to flash, mostly wanted to say that there looks to be something not quite right.

Jason2866, 2023-11-12:
Can't explain. but two cases are strange. Which Shelly version had the plug before converting?

mlschuh, 2023-11-12:
They were updated to the latest, I believe 1.0.8. Looks to have been very recent (within the last month) if I'm remember the datestamp on the firmware.

Gk2E7Aaojf3KDCQ, 2023-11-13:
I flashed it from 1.0.8 as well. I did so just the day before 13.2.0 was confirmed tested for the Plus Plug US.

I managed to solder some leads to the TP pads on the little board the ESP32 chip lives on. I'm also thinking the ESP32 chip I have is a dual core chip, as there's a D in the spot in the model number. I had read elsewhere, I think on the Tasmota Github page, that the D says it's dual core. Apparently, Shelly swapped these in around Dec. 2021.

I managed to erase and flash the tasmota.factory.bin for the initial setup, but I can't connect via serial connection. I don't think the chip is booting up correctly. I got thinking I should open the Plus Plug US .zip file and see if I can manually flash the bootloader.bin or other .bin files. Is there a way to manually flash Mongoose OS, then plug the plug in, connect via WiFi, then migrate to Tasmota?

ajkistner, 2023-11-16:
I have the same issue here basically. I had updated my Shelly firmware to 1.0.8 because the version that it came with so from 2022 and didn't even haven an option to upload a zipfile. At this point, I'm going to consider that one a loss because I'm not going to try to figure out how to solder to those tiny leads any time soon. 

But I have another one and I'm not sure what to do now - is there a way to update it via url if it's on build 20221107-134831/0.11.4-plugusprod3 or should I wait until more testing it done on 1.0.8 and update once it's been confirmed? Or is there a url for a newest version of the shelly firmware that's supported?

Jason2866, 2023-11-16:
At the moment, waiting is the recommendation. Didn't had the time to test and investigate.

Jason2866, 2023-12-05:
Closing since issue is not reproducable. #26

closed 2023-12-05 by Jason2866

Gk2E7Aaojf3KDCQ, 2023-12-23:
I have a few more Plus Plug US devices now. I tried flashing one of them and ran into the same problem, despite using the updated files based on yesterday's commit (after the events of #31). I updated to Shelly stock 1.10, then tried going to Tasmota. I tried using the dedicated PlusPlugUS version, ran into the name mismatch in the Shelly stock firmware logs of name mismatch mentioned in #29. I used the PlugUS.zip, which (as I understand it) the same as the PlusPlugUS.zip, just with the name changed in the manifest.json file.

I got to the ESP32-Dev on the plug, and could connect to WiFi. Just like my original issue, the partition wizard is a blank screen. I'm not sure what I'm doing wrong. My one thought is I chose the PlugUS autoconfig, as there was no PlusPlugUS autoconfig option. I never have even seen Tasmota Safeboot.

I can connect to the device via WiFi, and see it has a fs_0 and fs_1 pair of partitions in the Information screen. I manually put in ajkistner's into a manual configuration, and have basic control. But this is definitely unfinished.

The device seems to boot into fs_1 every time. I'm not sure how to proceed. The logs show the same berry errors when attempting to open partition wizard as my original problem. I hear of other people getting this to work, I'm not sure what I'm doing wrong or if I'm just unlucky to have the mislabeled Plus Plugs...

brad07x, 2024-01-24:
Sorry to post in a closed issue, but this seems like the most relevant one to the main issue I'm experiencing.

I can also confirm what @Gk2E7Aaojf3KDCQ reported in this issue and others in #29 mentioned pertaining to partitions on my Plus Plug US device as well. 

My plug also started on Shelly 1.10 and had the 'Wrong app name' issue as described in #29. I was able to use PlugUS.zip and the info & template from that issue to flash and configure to the point that basic controls and power monitoring are working.

The issue is that I also have the errors reported in this issue exactly when trying to use the Partition Wizard to convert to safeboot:

```
06:39:00.035 BRY: Exception> 'type_error' - unsupported operand type(s) for <=: 'nil' and 'int'
06:39:00.036 stack traceback:
06:39:00.036 	<unknown source>: in function `factory_migrate_eligible`
06:39:00.037 	<unknown source>: in function `show_migrate_to_factory`
06:39:00.038 	<unknown source>: in function `page_part_mgr`
06:39:00.039 	<unknown source>: in function `<lambda>`
```

The following platform info is also reported on the information screen:
```
ESP Chip Id	1798856 (ESP32-U4WDH-D rev.3)
Flash Chip Id	0x164020 (DIO)
Flash Size	4096 KB
Program Flash Size	0 KB
Program Size	1518 KB
Free Program Space	1536 KB
Free Memory	127.2 KB (frag. 16%)
 
Partition app_0*	1536 KB (used 98%)
Partition fs	448 KB
Partition app_1	1536 KB
Partition fs	448 KB
```

kevin-david, 2024-08-01:
Reporting another instance of this issue today - I should have done more reading before buying the Shelly Plus Plug US.

Any chance someone has figured out a way out in the past few months beyond cracking the device open? I'd also take a method back to stock so I can return it and get something else.

I find it interesting that the info screen reports partition info but `import partition_core var p=partition_core.Partition() print(p)` returns nothing ...

ajkistner, 2024-08-02:
Not that it helps at all but I sort of gave up and resigned myself that this Shelly plug will be stuck at 12.5. Since I had a 4 pk I had 4 tries, followed the directions perfectly, could never fix the partition wizard to work. I actually wish the table was updated to say this device is not upgradable past 12.5

If you want a recommendation for an ESP32 device with US plugs that can be updated past 12.5 then I suggest the Switchbot Mini Plug.