tasmota/mgos32-to-tasmota32 #23: shelly PLUS plug us not flashing.

aesir2, 2023-11-12:
I have a shelly PLUS plug us that would not accept a flash of any file to tasmota. I then tried the shelly plug us zip and that installed fine, even showing the cpu as esp32, however the pinout is different(there isn't a pinout or config for the plus plug us) so other than reading the voltage from the wall, nothing happens. I've tried release release of 13.X.X for the PLUS plug us and every time i go through tasmota firmwrae on the webui, it uploads and then returns that it failed with "invalid signature". I've downloaded and redownloaded every release, i've tried the ota from the tasmota webui. everything fails.

Jason2866, 2023-11-12:
Please provide Logs. Set log Level to `weblog 3`

aesir2, 2023-11-13:
> Please provide Logs. Set log Level to `weblog 3`

00:00:00.482 RUL: POWER1#BOOT performs "backlog ledpower1 1; ledpower2 1"
00:00:00.483 SRC: Rule
00:00:00.484 CMD: Grp 0, Cmd 'BACKLOG', Idx 1, Len 24, Pld -99, Data 'ledpower1 1; ledpower2 1'
00:00:00.485 Project tasmota - Tasmota Version 12.5.0(solo1)-2_0_7(2023-04-17T08:07:06)
00:00:00.496 ETH: No ETH MDC and/or ETH MDIO GPIO defined
00:00:00.672 TAP: Loaded Tasmota App 'Partition_Wizard.tapp'
00:00:01.548 BRY: No 'autoexec.be'
00:00:01.550 SRC: Backlog
00:00:01.551 CMD: Grp 0, Cmd 'LEDPOWER', Idx 1, Len 1, Pld 1, Data '1'
00:00:01.553 RSL: RESULT = {"LedPower1":"ON"}
00:00:02.183 SRC: Backlog
00:00:02.184 CMD: Grp 0, Cmd 'LEDPOWER', Idx 2, Len 1, Pld 1, Data '1'
00:00:02.186 RSL: RESULT = {"LedPower2":"ON"}
00:00:02.197 WIF: Attempting connection...
00:00:02.628 WIF: Connecting to AP1 xxx Channel 1 BSSId xx:xx:xx:xx:xx:xx in mode 11n as tasmota-xxxxxx-4640...
00:00:05.155 WIF: Connected
00:00:05.417 HTP: Web server active on tasmota-xxxxxx-4640 with IP address 192.168.1.42
00:00:08.937 RSL: INFO1 = {"Info1":{"Module":"ShellyPlugUS","Version":"12.5.0(solo1)single-core","FallbackTopic":"cmnd/DVES_091220_fb/","GroupTopic":"cmnd/tasmotas/"}}
00:00:08.953 RSL: INFO2 = {"Info2":{"WebServerMode":"Admin","Hostname":"tasmota-xxxxxx-4640","IPAddress":"192.168.1.42","IP6Global":"","IP6Local":"xxx::xxx:xxx:xxx:xxx"}}
00:00:08.967 RSL: INFO3 = {"Info3":{"RestartReason":"Vbat power on reset","BootCount":6}}
00:00:08.971 RSL: RESULT = {"POWER":"ON"}
00:00:08.972 RSL: POWER = ON
00:00:08.992 QPC: Reset
03:36:04.868 RSL: STATE = {"Time":"2023-11-13T03:36:04","Uptime":"0T00:00:11","UptimeSec":11,"Heap":152,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":54,"MqttCount":0,"Berry":{"HeapUsed":22,"Objects":331},"POWER":"ON","Wifi":{"AP":1,"SSId":"xxx","BSSId":"xx:xx:xx:xx:xx:xx","Channel":1,"Mode":"11n","RSSI":100,"Signal":-46,"LinkCount":1,"Downtime":"0T00:00:04"}}
03:36:04.901 RSL: SENSOR = {"Time":"2023-11-13T03:36:04","ENERGY":{"TotalStartTime":"1970-01-01T00:00:00","Total":0.000,"Yesterday":0.000,"Today":0.000,"Period":0,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":122,"Current":0.000}}
03:36:44.060 RSL: RESULT = {"SerialLog":{"2":{"Active":"2"}}}
03:36:44.081 RSL: RESULT = {"WebLog":3}
03:36:44.088 SRC: Backlog
03:36:44.089 CMD: Grp 0, Cmd 'MQTTLOG', Idx 1, Len 1, Pld 0, Data '0'
03:36:44.091 RSL: RESULT = {"MqttLog":0}
03:36:44.097 SRC: Backlog
03:36:44.098 CMD: Grp 0, Cmd 'SYSLOG', Idx 1, Len 1, Pld 0, Data '0'
03:36:44.099 RSL: RESULT = {"SysLog":{"0":{"Active":"0"}}}
03:36:44.111 SRC: Backlog
03:36:44.112 CMD: Grp 0, Cmd 'LOGHOST', Idx 1, Len 1, Pld 1, Data '1'
03:36:44.113 CFG: CR 402/699, Busy 0
03:36:44.115 RSL: RESULT = {"LogHost":""}
03:36:44.126 SRC: Backlog
03:36:44.128 CMD: Grp 0, Cmd 'LOGPORT', Idx 1, Len 3, Pld 514, Data '514'
03:36:44.129 RSL: RESULT = {"LogPort":514}
03:36:44.136 SRC: Backlog
03:36:44.137 CMD: Grp 0, Cmd 'TELEPERIOD', Idx 1, Len 3, Pld 300, Data '300'
03:36:44.139 RSL: RESULT = {"TelePeriod":300}
03:36:45.004 CFG: Saved, Count 36, Bytes 4096
03:36:45.016 RSL: STATE = {"Time":"2023-11-13T03:36:45","Uptime":"0T00:00:52","UptimeSec":52,"Heap":148,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":29,"MqttCount":0,"Berry":{"HeapUsed":22,"Objects":331},"POWER":"ON","Wifi":{"AP":1,"SSId":"xxxx","BSSId":"xx:xx:xx:xx:xx:xx","Channel":1,"Mode":"11n","RSSI":100,"Signal":-45,"LinkCount":1,"Downtime":"0T00:00:04"}}
03:36:45.048 RSL: SENSOR = {"Time":"2023-11-13T03:36:45","ENERGY":{"TotalStartTime":"1970-01-01T00:00:00","Total":0.000,"Yesterday":0.000,"Today":0.000,"Period":0,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":122,"Current":0.000}}
03:36:47.290 HTP: Configure Logging
03:36:53.109 HTP: Configuration
03:37:00.807 HTP: Main Menu
03:37:04.158 HTP: Firmware Upgrade
03:37:07.210 HTP: Upgrade started
03:37:07.212 SRC: WebGui from 192.168.1.49
03:37:07.213 CMD: Grp 0, Cmd 'OTAURL', Idx 1, Len 54, Pld -99, Data 'http://ota.tasmota.com/tasmota32/release/tasmota32.bin'
03:37:07.214 CFG: CR 402/699, Busy 0
03:37:07.215 RSL: RESULT = {"OtaUrl":"http://ota.tasmota.com/tasmota32/release/tasmota32.bin"}
03:37:07.324 SRC: WebGui from 192.168.1.49
03:37:07.325 CMD: Grp 0, Cmd 'UPGRADE', Idx 1, Len 1, Pld 1, Data '1'
03:37:07.334 RSL: RESULT = {"Upgrade":"Version 12.5.0 from http://ota.tasmota.com/tasmota32/release/tasmota32.bin"}
03:37:07.631 CFG: Saved, Count 37, Bytes 4096
03:37:09.452 UPL: http://ota.tasmota.com/tasmota32/release/tasmota32.bin 12.5.0(solo1)
03:37:11.701 OTA: Connection timeout after 1006 ms
03:37:13.528 UPL: http://ota.tasmota.com/tasmota32/release/tasmota32.bin 12.5.0(solo1)
03:37:15.132 OTA: Connection timeout after 1001 ms
03:37:16.911 UPL: http://ota.tasmota.com/tasmota32/release/tasmota32.bin 12.5.0(solo1)
03:37:18.272 OTA: Connected in 762 ms, stack low mark 4204
03:37:18.354 HTP: Main Menu
03:37:20.385 UPL: http://ota.tasmota.com/tasmota32/release/tasmota32.bin 12.5.0(solo1)
03:37:21.644 OTA: Connected in 623 ms, stack low mark 4204
03:37:21.680 HTP: Consoles
03:37:23.494 HTP: Console
03:37:25.605 RSL: UPGRADE = {"Upgrade":"Failed "}
03:37:25.618 UPP: Multicast (re)joined
03:37:40.733 HTP: Consoles
03:37:43.750 HTP: Main Menu
03:37:47.128 HTP: Information
03:37:49.005 HTP: Main Menu
03:37:50.560 HTP: Information
03:37:51.440 HTP: Main Menu
03:37:52.687 HTP: Configuration
03:37:54.504 HTP: Configure Logging
03:38:43.639 HTP: Configuration
03:38:46.416 HTP: Main Menu
03:38:47.937 HTP: Consoles
03:38:50.394 HTP: Console

Jason2866, 2023-11-13:
Wrong firmware change your OTA URL to `http://ota.tasmota.com/tasmota32/release/tasmota32solo1.bin`

aesir2, 2023-11-13:
I had that url originally but when i clicked update, it would fail and say invalid signature. i pasted that url again and clicked update and it didn't fail. I used the newest release to install tasmota and the webui says it failed because of invalid signature. looking at the information tab i get this:

Program Version | 12.5.0(solo1)single-core
-- | --
build date and time 2023-04-17T08:07:06
core/sdk version 2_0_7/4.4.4.20230218

ESP Chip Id	594464 (ESP32-U4WDH-D rev.3)
Flash Chip Id	0x164020 (DIO)
Flash Size	4096 KB
Program Flash Size	0 KB
Program Size	1518 KB
Free Program Space	1536 KB
Free Memory	140.4 KB (frag. 18%)

console output is:
00:27:08.966 HTP: Firmware Upgrade
00:27:14.794 UPP: Multicast disabled
00:27:15.006 CFG: Saved, Count 45, Bytes 4096
00:27:15.008 UPL: File mgos32-to-tasmota32-PlusPlugUS.zip
00:27:15.048 UPP: Multicast (re)joined
00:27:23.938 HTP: Upload done
00:27:25.528 HTP: Main Menu
00:27:28.077 HTP: Consoles
00:27:28.213 APP: Serial logging disabled
00:27:29.830 HTP: Console

however it seems the pinout configuration is for the shelly plug us and not the plus plug us, assuming there is a difference, since on the main screen it shows volts but no other value is populated.

Jason2866, 2023-11-19:
Your OTA URL is wrong `00:27:15.008 UPL: File mgos32-to-tasmota32-PlusPlugUS.zip`
you have to provide this `http://ota.tasmota.com/tasmota32/tasmota32solo1.bin`

closed 2023-11-25 by aesir2