tasmota/mgos32-to-tasmota32 #24: Bricked 2 Shelly Plus1 trying OTA conversion and Safeboot install

wunter8, 2023-11-13:
I was able to upload the custom firmware to get into the Tasmota Web UI the first time. I auto-configured for the Shelly Plus1, then I tried doing the Safeboot migration. Both times (on two different Shelly Plus1 devices), after starting the Safeboot migration, the device never connects to the Wifi AP again. The Tasmota fallback AP appears again, but when I connect to it, the page (192.168.4.1) just returns net::ERR_EMPTY_RESPONSE.

Both times I tried this, I had the Shelly Plus1 connected to 12v DC power. Was I not supposed to do it that way?

I tried flashing via a serial connection and broke off the small headers on the Plus1 since they're so small and don't work with normal Dupont connectors. I'm thinking I might have to solder small leads to the serial connection pads to get it flashed.

Is there anything else I can do before pulling out the soldering iron?

Thanks!

Jason2866, 2023-11-16:
Soldering is the way.

NoFace3D, 2023-11-16:
Same issue here, after Safeboot Conversion wifi access point is reachable, but shelly is not responding.
I am guessing, that it was required to specify different firmware for the update and I might have single-core MCU.
If that's the case I'd suggest to update docs, that during this step might be needed to change default URLs.

Jason2866, 2023-11-16:
If a dual core firmware is flashed on a single core esp32, nothing really nothing will work, since device is boot looping.

wunter8, 2023-11-19:
I was able to fix both devices by soldering a UART adapter to the RX/TX pins and flashing ESPHome (sorry for abandoning Tasmota... 😬)

closed 2023-11-19 by wunter8