tasmota/mgos32-to-tasmota32 #26: Success on Shelly Plus Plug S

teknoid, 2023-12-05:
- updated Original shelly to 1.0.8
- tried update via /ota?url=xxx   --> worked and gave me Tasmota 12.5
- tried partition wizard bricked the Tasmota and got me original Shelly Firmware back
- then followed your instructions and that was successful
- now I have Tasmota 13.2.0 with new safeboot layout

Mikecodex, 2024-03-30:
@teknoid Where did you get Original Shelly Firmware 1.0.8?

teknoid, 2024-03-31:
The devices updated automatically to this version. There was no need to download it manually.
Probably this FW is now outdated?

Mikecodex, 2024-03-31:
I dont know why it dosent work, every time I drag a file the plug ignores it and goes to the clowd to get updates and if I make a server with the file to update it goes to the clowd, fails, gives an error on the capability of getting the file from my server and nothing.
[link server.txt](https://github.com/tasmota/mgos32-to-tasmota32/files/14815072/link.server.txt)

[drag file.txt](https://github.com/tasmota/mgos32-to-tasmota32/files/14815065/drag.file.txt)

I give up from ota, going serial wish me luck.

Jason2866, 2024-03-31:
A last try to recover the device. **Download** the update package zip from Shelly update server. Drag and drop this file **local** in update area. Hit update. **If** this update worked, do the same again. Needed to have the Shelly firmware in both OTA firmware slots.
After that you can try to drag and drop the Shelly convert package again.

Mikecodex, 2024-03-31:
@Jason2866 Shelly update server, i dont know what it is. I know the firmware archive but they don't have for the plugs, I found an old firmware that was for 0.14.XX in an old forum but that did not work. I think its hard to find firmware for this device in the internet and, not to mention, I have shelly plug plus S v2. I don't know what v2 means in terms of firmware.

Jason2866, 2024-03-31:
@Mikecodex Whats the exactly device name shown in Shelly firmware page?
Please post or post a screenshot. Will have a look for the download URL from Shelly update server.

Mikecodex, 2024-04-01:
This one https://kb.shelly.cloud/knowledge-base/shelly-plus-plug-s-v2 (SNPL-0112EU) i have installed by serial esphome In one plug have 4 to go, the casing is destroyed using this method . If this server has older version try getting 1.0.8 so I can replicate the top post steps.

RavenSystem, 2024-04-01:
@Mikecodex Device name shown in your Shelly firmware web page is needed.

Mikecodex, 2024-04-01:
Is this what you requested?
[image]

Jason2866, 2024-04-01:
The URL to download the Shelly Update package is https://fwcdn.shelly.cloud/gen2/PlusPlugS/f27ae4e9cc7bbbe58cf73764c4cf445871b6dad357d5d352b75f367ea3677120
generated from this URL https://updates.shelly.cloud/update/PlusPlugS

Use update from file to try to flash it. It it works (device restarts) do it again! After this you can try the Shelly convert package.

Jason2866, 2024-04-01:
Well it is in the description for a reason. 
> Navigate to Settings > Firmware and drag & drop the convert "zip" in the firmware update area (do NOT update via URL)

Not following the guide has a very high chance for bricking the device.

closed 2024-04-01 by Jason2866