tasmota/mgos32-to-tasmota32 #28: Will these work on Gen3?

tbrasser, 2023-12-18:
Just received my first batch of minis. Got the new Gen3 ones, see attached images. Are the released firmwares safe to use? [image]

[image]

[image]

Jason2866, 2023-12-18:
Should. Not tested, since we don't have.

Mugga6315, 2023-12-20:
I've also got some Shelly Plus 1PM Mini Gen3 here, but sadly update isnt working.

I just get in JS Console the following:
`(index):20 Error: no update in progress
    at DeviceFactory.processMessage_ ((index):29:16364)
    at DeviceFactory.handleMessage_ ((index):29:15917)
    at DeviceFactory.refresh ((index):29:13237)`

Anything I can help to better debug this?

Jason2866, 2023-12-20:
What Shelly firmware version is installed?

Mugga6315, 2023-12-20:
It's latest from shelly: 20231219-133952/1.1.0-g34b5d4f
And web build id is: 231218152112-8372d8c8-HEAD

Jason2866, 2023-12-20:
So, it is a newer firmware. v1.10. Known working is v1.08
The log you have, that's all when you drop the convert.zip in the update area?
Nothing else?

Mugga6315, 2023-12-20:
Yeah, thats all I have. I browse or drag n drop the file "mgos32-to-tasmota32-Plus1PMMini.zip" and after hitting "Update from file" just the error I posted.

But I also got a fresh one which I haven't yet updated. This one got the following version:
Firmware 20231121-110955/1.1.99-minig3prod1-ga898543
Web Build 231121102904-aad9cbe1-HEAD

But getting the same console error like with the newest firmware

Jason2866, 2023-12-20:
Thanks. Nothing more can be done. Need the device to do investigation. Not saying a solution can be found. The train maybe stops forever for OTA convert.
To be clear i am not going to buy the new generation. It is a hobby and i can't buy everything.

Mugga6315, 2023-12-20:
Okay, got it. Totally understandable.
I've written via discord, if I can assists you in anyway.

Mugga6315, 2023-12-21:
Just one thing I wanted to add which maybe helps. The js error seems to arise in that function:
`async updateViaFile(e) {
                    if (!e)
                        return !1;
                    let t = await new Promise((t=>{
                        const o = new FileReader;
                        o.onload = ()=>t(new Uint8Array(o.result)),
                        o.readAsArrayBuffer(e)
                    }
                    ));
                    await this.device.rpcRequest({
                        method: "OTA.Start",
                        params: {
                            size: t.length
                        }
                    });
                    let o = new Array(Math.ceil(t.length / _Firmware.CHUNK)).fill(0).map(((e,o)=>({
                        start: o * _Firmware.CHUNK,
                        end: Math.min((o + 1) * _Firmware.CHUNK, t.length)
                    }))).map((e=>({
                        offset: e.start,
                        data: btoa(String.fromCharCode(...t.slice(e.start, e.end)))
                    })));
                    try {
                        for await(let e of o)
                            await this.device.rpcRequest({
                                method: "OTA.Write",
                                params: e
                            })
                    } catch (r) {
                        console.error(r)
                    }
                    return !0
                }`

tbrasser, 2023-12-21:
I guess this can be closed? Will try the new release tomorrow!

tbrasser, 2023-12-22:
We could try this? https://github.com/tasmota/mgos32-to-tasmota32/commit/ffd16849fddf2a564d631b8e653f2c9877d8d164

Obviously needs the 8MB C3 files generated

Jason2866, 2023-12-22:
@tbrasser We need the files you reference. Possible to get from a complete dump of the flash or from a update package. The source URL is visible in the log when starting the update or search process for a newer firmware.

tbrasser, 2023-12-22:
https://updates.shelly.cloud/update/Mini1G3

Jason2866, 2023-12-22:
@tbrasser give it a try https://github.com/tasmota/mgos32-to-tasmota32/releases/tag/v13.3.1-Mini1G3 used your change

tbrasser, 2023-12-22:
Cool! Will probably be able to test it tomorrow.

Jason2866, 2023-12-22:
If you like you can look at this, similar what is needed for the G3 https://github.com/tasmota/autoconf/tree/main/raw/esp32c3/Shelly_Plus1PMMini
You can use as boilerplate for a PR for the MiniG3 ;-)

tbrasser, 2023-12-22:
Tbh I'm using this as a stepping stone to esphome, have 10 of these so I am willing to be less careful with 1 of them. Thanks for the info, will take a look at it when I can!

Jason2866, 2023-12-23:
The step replacing bootloader is needed. Without it get stuck either in Shelly or Tasmota firmware.

tbrasser, 2023-12-23:
Somehow the manifest lists platform as esp32 instead of esp32c3 and thus getting `shelly_ota.cpp:311 Aborting update: bad arch: esp32` in the logs.

Jason2866, 2023-12-23:
@tbrasser re-released under same tag.
s-hadinger was faster and commented already the Autoconfig PR

tbrasser, 2023-12-23:
Works! :tada:  Will update autoconf PR

tbrasser, 2023-12-23:
Also, not sure what part of this is read/probed from the esp or comes from the firmware, but dropping it here anyway:

```
ESP Chip Id	3995060 (ESP32-C3 rev.4)
Flash Chip Id	0x1740C8 (QIO)
Flash Size	8192 KB
Program Flash Size	8192 KB
Program Size	1567 KB
Free Program Space	2560 KB
Free Memory	190.1 KB (frag. 13%)
	 
Partition app_0	2560 KB
Partition fs	1024 KB
Partition app_1*	2560 KB (used 61%)
Partition fs	1024 KB
```

Jason2866, 2023-12-23:
> Works! 🎉 Will update autoconf PR

Great!! After the PR "Autoconfig" is updated i will merge the changes from you in main branch and do a new release.

Jason2866, 2023-12-23:
About the info log you posted. This is the result of the still used partition scheme from Shelly.
The partition scheme will be aligned to Tasmota partition scheme when the safeboot converting is done. When using a other firmware than Tasmota, it has to work with that scheme or do a repartitoning like the Tasmota safeboot convert process.

Jason2866, 2023-12-23:
@tbrasser Your Autoconfig PR has been merged. Thx!
You should now be able to do the Autoconfig for the gen3 devices.
This will replace the bootloader and you can update the firmware.

tbrasser, 2023-12-23:
Not sure what the correct place is, but opened the issue here: https://github.com/tasmota/autoconf/issues/37

Jason2866, 2023-12-23:
No fast solution for this. Needs big changes in Berry script, to get Tasmota running with safeboot scheme. Have you seen a message like "bootloader flashed / replaced"?
If so you can try to upgrade anyway.

AntonWert, 2023-12-28:
Hi,
I do also have a G3 version of the Plus PM Mini here and like to test. Can I use the pre-release?

And one question: If something fails, and I do have a "brick" can I solder (I know it might be fiddely) the RX TX Pins and upload manually as on the "normal" Shellys?

Jason2866, 2023-12-28:
@AntonWert  First step backup the orig. firmware. So you can go always back to Shelly firmware. The flash size is 8MB. No OTA does not work since the Partition layout is completely different to anything else before.
Yes, you can flash Tasmota via wire, and it will work. Just the correct config (Template) needs to be found.

AntonWert, 2023-12-28:
> @AntonWert First step backup the orig. firmware. So you can go always back to Shelly firmware. The flash size is 8MB. No OTA does not work (yet) since the Partition layout is completely different to anything else before. Yes, you can flash Tasmota via wire, and it will work. Just the correct config (Template) needs to be found.

Thanks for the quick answer.
<< First step backup the orig. firmware
Can I do this via the (shelly) UI? Or how can I download the current FW? (Well I do not need it, since I want to use Tasmota)

<<No OTA does not work (yet) since the Partition layout is completely different to anything else before.
Wwuld it help, if I wait until I can test it? I do have not the need to do the Tasmota conversion now, and would offer to help...

The config from there should work?
https://templates.blakadder.com/shelly_plus_pm_mini.html

Jason2866, 2023-12-28:
idk if any existing Template "fits". We don't have the device, and you are the first doing wired flashing of Tasmota.
Downloading of the complete Shelly firmware is only possible via wire. Best tool for is `esptool.py` via command `esptool.py --baud 115200 read_flash 0x00000 0x800000 Backup_8MB.bin`
If you provide the backup of the firmware to us, we can investigate and do the needed changes in the Berry script.

AntonWert, 2023-12-28:
@Jason2866 :
what do you mean with fits?
<<idk if any existing Template "fits".

Jason2866, 2023-12-28:
If it has the correct GPIO assignments.

AntonWert, 2023-12-28:
Are you shure with 8MB flash?
ESPTool tells me 4MB
Connecting....
Detecting chip type... ESP32-C3
Chip is ESP32-C3 (QFN32) (revision v0.4)
Features: WiFi, BLE, Embedded Flash 4MB (XMC)
Crystal is 40MHz

It seems at least I do have soldered the correct pins.

Maybe you do not have seen my question above, witch is the correct version to flash

Jason2866, 2023-12-28:
Shelly has posted the info 8MB flash on there Web site. Until now there are no C3 devices out there with 8MB flash. esptool.py does not "know" this variant, so the info is most probably wrong. As it is just an info, the read command for 8MB will work.

Mugga6315, 2023-12-28:
> idk if any existing Template "fits". We don't have the device, and you are the first doing wired flashing of Tasmota. Downloading of the complete Shelly firmware is only possible via wire. Best tool [...]

I did wired flashing with one of the 1PM Mini, did work without any issues and I just used the autoconfig template for the "normal"  1PM Mini. No issues on my side

AntonWert, 2023-12-28:
@Jason2866 
What do you want to know from the backup? I do not know if i can post the backup here publically. 
But If you tell me what to look for, I can tell you the result :-)

Jason2866, 2023-12-28:
It would be great if you zip the file and post it on Tasmota Discord via PM.
My name is at Discord @Jason2866 too.
Link to Discord https://discord.gg/Ks2Kzd4

AntonWert, 2023-12-28:
> It would be great if you zip the file and post it on Tasmota Discord via PM. My name is at Discord @Jason2866 too. Link to Discord https://discord.gg/Ks2Kzd4

I sent you there a request, allowing sending direct PM messages

AntonWert, 2023-12-28:
@Mugga6315 
I was now able to flash Tasmota, and it seems to work.
Only the Temerature Reading is far off. Did your's work? Do you have to calibrate for temperature?

Mugga6315, 2023-12-28:
> @Mugga6315 I was now able to flash Tasmota, and it seems to work. Only the Temerature Reading is far off. Did your's work? Do you have to calibrate for temperature?

Very good. Yeah mine was off to, but its written on the tasmota page that you have to calibrate it. But I dont use the temp sensor anyway, so I didnt care. Did you manage to get a dump? If not I can also try to get a shelly original firmware dump.

AntonWert, 2023-12-28:
Jason2866 has a dump.
I have to search how to calibrate the temp sensor

tbrasser, 2023-12-28:
Nice! I also tried the autoconfig template manually before creating that PR and that worked already, forfot to mention that. And yeah temp seemed low but I counted that on bare on the table vs behind a switch in the wall. Curious to compare them in a similar physical situation running tasmota/esphome vs stock. 

Did you get the dump already?

Jason2866, 2023-12-30:
@tbrasser @AntonWert The flashing process from `mgos32-to-tasmota32` is already working as it should. Please try latest Autoconfig. Before remove every file in Tasmota  Filesystem.

AntonWert, 2023-12-30:
Should I try with the [Mini gen3 support](https://github.com/tasmota/mgos32-to-tasmota32/releases/tag/v13.3.1-Mini1G3) released last week?

Jason2866, 2023-12-30:
> Should I try with the [Mini gen3 support](https://github.com/tasmota/mgos32-to-tasmota32/releases/tag/v13.3.1-Mini1G3) released last week?

Yes

AntonWert, 2023-12-31:
Sorry for the delay, but here ist the result:
I used a stock "Shelly Plus PM Mini"
- Used the AP to setuo my WiFi
- Used the now to internet connected device to upate to the newest version
- uploaded the File
- Did the auto config
- Started Safeboot Conversion, the Web Page with "Migration process will start in 5 seconds Magic is happening, leave device alone for 3 minutes." stood there for 6 minutes
- nothing happened any more , no LEDs flashed, no access
- I power cycled the devices, Tasmota 13.3.0(tasmota32) came online, everything seems to work (maybe I need calibration of Temperature and power)

So I can confirm "Shelly Plus PM Mini" GEN 3 is working accoring the manual

closed 2023-12-31 by Jason2866

tbrasser, 2024-01-01:
> @tbrasser @AntonWert The flashing process from `mgos32-to-tasmota32` is already working as it should. Please try latest Autoconfig. Before remove every file in Tasmota  Filesystem.

I did this, result is no more partition wizard.

So tried a "fresh" Mini1G3, but seemingly same result:
- connect to wifi
- update Shelly to 1.1.0
- upload Mini1G3 mgos32-to-tasmota32 and flash
- connect to wifi
- run autoconf, device restarts

Then I have this in logs:
`failed to run compiled code 'syntax_error' - /Partition_Wizard.tapp#/partition_wizard.be:813: expected 'end' before 'EOS'`

Jason2866, 2024-01-01:
Grrr, the TAPP file seems corrupt in repo Autoconf