tasmota/mgos32-to-tasmota32 #29: How to work around manifest.json mismatch with deviceInfo for Shelly Plug US Plus identifying as Plug US

morrowwm, 2023-12-19:
I just received a Shelly Plug US, the box has "Plus" in the upper right. The Shelly Web UI has "Shelly PlugUS" in its footer. I popped open the case, it has an ESP32-U4WD on the daughterboard. So I'm sure it is a "Plus" device.

However Shelly deviceInfo reports this as simply PlugUS, no "Plus":
```
  "deviceInfo": {
    "name": "shelly03",
    "id": "shellyplugus-ccdba7d11c10",
    "mac": "CCDBA7D11C10",
...
    "batch": "2245-Frankever",
    "fw_sbits": "00",
    "model": "SNPL-00116US",
    "gen": 2,
    "fw_id": "20231106-160237/1.0.8-gdba0ee3",
    "ver": "1.0.8",
    "app": "PlugUS", <--------------------- ???
    "auth_en": false,
    "auth_domain": null
  }
```

I tried flashing tasmota32solo1.factory.bin over the air, with no luck. On the recommendation of experts on the Tasmota discord chat, I then tried mgos32-to-tasmota32-PlusPlugUS.zip, totalling 2043697 bytes. Simultaneously with loading the firmware, I had the Shelly diagnostic log open in another browser window. It showed:
```
shos_rpc_inst.c:230 ota.start via WS_in 192.168.0.11:52350
shelly_ota.cpp:304 Starting update (0x3ffe4818), timeout=600 commit_timeout=0 ignore_same=0
shos_rpc_inst.c:230 ota.write via WS_in 192.168.0.11:52350
shelly_ota.cpp:403 'manifest.json' 1793 bytes (hdr 71 desc 0 crc eb0f6367)
shelly_ota.cpp:523 fw: PlusPlugUS ver 12.5.0 20231029-135017/tasmota-12.5.0
shelly_update.cpp:305 incoming fw signatures: 00
shelly_update.cpp:297 Wrong app name 'PlusPlugUS' <---------- ??
shelly_ota.cpp:311 Aborting update: aborted by user
shelly_ota.cpp:234 OTA status error, 0 %
shelly_ota_source_rp:21 leftover RPC source 0x3ffe1f78
shos_rpc_inst.c:230 ota.write via WS_in 192.168.0.11:52350
```
I unzipped mgos32-to-tasmota32-PlusPlugUS.zip and found the manifest.json file:
```
{
  "name": "PlusPlugUS",
  "platform": "esp32",
  "version": "12.5.0",
  "build_id": "20231029-135017/tasmota-12.5.0",
  "build_timestamp": "2023-10-29T13:50:17Z",
```
I changed
```
"name": "PlusPlugUS",
```
to
```
"name": "PlugUS",
``` 
Then zipped all the files together again and attempted to load this. In the log:
```
Zip it all together after this change. Installing, and the log gives:
shos_rpc_inst.c:230 ota.start via WS_in 192.168.0.11:52350
shelly_ota.cpp:304 Starting update (0x3ffe4534), timeout=600 commit_timeout=0 ignore_same=0
shos_rpc_inst.c:230 ota.write via WS_in 192.168.0.11:52350
shelly_ota.cpp:403 'bootloader-tasmota.bin' 15648 bytes (hdr 80 desc 0 crc 160e0a45)
shelly_ota.cpp:311 Aborting update: manifest.json was not first in archive   <----------------- !!!
shelly_ota.cpp:234 OTA status error, 0 %
shelly_ota_source_rp:21 leftover RPC source 0x3ffe3334
shelly_notification:209 Event from sys: {"component":"sys", "event":"ota_error", "msg":"error", "ts":1702904137.62}
shos_rpc_inst.c:230 ota.write via WS_in 192.168.0.11:52350
```
I rebuilt the archive in a better order, using no compression, and forcing manifest.json to be first. These are linux command line:
```
zip -0 poop.zip manifest.json 
zip -u -0 poop.zip bootloader-tasmota.bin fs-4MB-tasmota.img otadata-tasmota.bin partition-table-tasmota.bin tasmota32solo1.bin 
```
Uploading poop.zip worked.

There remains issues getting the template configuration correct, but that's a second issue.

Jason2866, 2023-12-19:
This one https://github.com/tasmota/mgos32-to-tasmota32/releases/download/v13.2.1/mgos32-to-tasmota32-PlugUS.zip is now in release section. So there is now aversion with "plus" and without for the same device.
Shelly messed up there own naming convention.

morrowwm, 2023-12-20:
jason2866, your new firmware worked fine. It did spew a lot of messages in the Shelly diagnostic log, I'm guessing one for every chunk of the update? My Shelly PlugUS Plug is now an ESP32-DevKit running Tasmota 12.5.0.

morrowwm, 2023-12-24:
I traced circuits from the BL0937 on the main board through to the ESP32-U4WD on the daughter board.
```
BL0937 pin   daughter board jumper  ESP32-U4WD pin 
VDD       1                J15      3V3 bus       
CF        6                J19      GPIO25     14
CF1       7                J20      GPIO26     15
SEL       8                J21      GPIO27     16
```
Unfortunately, I get no voltage, current, power measurements. All zeroes.  I might have destroyed the BL0937 doing earlier tests, though. CF and CF1 are outputs on the BL0937. Earlier, I used them as inputs. There is a pull-up resistor on these though. Would that protect it?

I don't have any other PlugUS PLus plugs right now. Could someone try this template?

```
{"NAME":"Shelly PlugUS Plus","GPIO":[0,0,0,0,32,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,2720,2656,2592,0,0,0,0,224,320,0,0,0,0,0,0],"FLAG":0,"BASE":1}
```

Gk2E7Aaojf3KDCQ, 2023-12-24:
Sure. Let me know if you have any others to verify. I have another plug that I opened and tried serial flashing. I did solder the terminal pads (TP #), I know which pads map to which pin on the ESP32 chip. The TP pads are the same order as the pins in the Plus 1PM (https://templates.blakadder.com/shelly_plus_1PM.html). I can also try to verify your wire traces if that will help.

b3schni, 2024-01-09:
> I just loaded your custom config into a still sealed Plus Plug US. I'm stuck at Tasmota 12.5.0, where I load the automatic configuration (still stuck with a blank partition wizard screen #22). But I can verify it's not just your plug. I'm getting all zeroes on the voltage, current, and power.

I run into the exact same behavior.
Shelly Plug flashed fine with [PlugUS](https://github.com/tasmota/mgos32-to-tasmota32/releases/download/v13.3.1/mgos32-to-tasmota32-PlugUS.zip).
[PlusPlugUS](https://github.com/tasmota/mgos32-to-tasmota32/releases/download/v13.1.1/mgos32-to-tasmota32-PlusPlugUS.zip) didn't work. Nothing would happen in the UI when clicking the Update button.
Partition Wizard is blank.
Stuck on version 12.5.0
 
I also traced the pinout, which seems to be different from @morrowwm. I checked the datasheet of both to match the pins to the GPIOs.
`BL0937	pin	ESP32-U4WD	pin
SEL	8	GPIO27		16
CF1	7	GPIO14 (MTMS)	17
CF	6	GPIO12 (MTDI)	18`

I got the energy monitor working like this:
`{"NAME":"Shelly Plus Plug US","GPIO":[0,0,0,0,32,0,0,0,2624,0,2656,0,0,0,0,0,0,0,0,0,0,0,320,2720,0,0,0,0,224,544,0,0,0,0,0,0],"FLAG":0,"BASE":1}`

The only load I have to test is a 150W bulb which reads as 178W at a power factor of 1. Seems ok, but if someone could confirm if these settings are good?

brad07x, 2024-01-24:
@b3schni - It appears I am in the exact same situation as you. 

My Plus Plug US started on Shelly 1.10 and had the wrong app name error as well; used PlugUS.zip to flash mgos32-to-tasmota32 successfully.

From there, I was able to use your template to get my Plus Plug US working with basic controls and power monitoring verified against a test load.

Unfortunately I was also not successful in applying autoconfig or using the partition wizard to convert to safeboot as you and @Gk2E7Aaojf3KDCQ reported - mine is also blank as described in #22 and am stuck at v12.5.0.

optio50, 2024-03-18:
Hoping there might be some movement on this issue. 
I have the same symptoms as @brad07x mentioned above

closed 2024-12-16 by Jason2866