tasmota/mgos32-to-tasmota32 #30: Shelly Plus Plug S firmware update not going through

maurice-p, 2023-12-20:
I connected the Shelly Plus Plug S via WiFi and then uploaded the file "mgos32-to-tasmota32-PlusPlugS.zip" from the last release and then pressed "Update from file".
The update is now running, the socket flashes red. The progress display then runs through in the UI. The UI remains responsive and the socket is permanently red.
According to the description, step 5 should now be carried out:
--> 5. If Shelly Web frontend is back after the update, repeat the steps above.
 
Unfortunately I'm in a loop, new firmware updates have no effect. I cant get past the Shelly UI.

The current Shelly firmware (1.1.0), more precisely 20231219-134003/1.1.1.0-g34b5d4f, is still installed.

Can you guys please help me?

RavenSystem, 2023-12-21:
This issue is the same as https://github.com/tasmota/mgos32-to-tasmota32/issues/31

Jason2866, 2023-12-21:
Fixed with release v13.3.0

closed 2023-12-21 by Jason2866