tasmota/mgos32-to-tasmota32 #31: Shelly stock firmware v1.1.0 prevents to install Tasmota

RavenSystem, 2023-12-20:
https://github.com/phiten/mgos32-to-tasmota32/issues/27

(Sorry, I don't know where is the best place to submit an issue.)

Jason2866, 2023-12-20:
Any logs?

RavenSystem, 2023-12-20:
```
shos_rpc_inst.c:230     ota.write via WS_in 10.0.0.248:51721
shelly_ota.cpp:403      bootloader-tasmota.bin: 15648 bytes (hdr 80 desc 0 crc 160e0a45)
shelly_ota.cpp:311      Aborting update: backend 0x3ffe239c aborts update at file bootloader-tasmota.bin: -4: app / fs already written, can't update boot
shelly_ota.cpp:234      OTA status error, 76 %
shelly_ota_source_rp:21 leftover RPC source 0x3ffe0c04
```

Do you want complete log?

Jason2866, 2023-12-20:
Thx, yes complete log would be good to have an idea where it stops

RavenSystem, 2023-12-20:
[screenlog.txt](https://github.com/tasmota/mgos32-to-tasmota32/files/13732301/screenlog.txt)

RavenSystem, 2023-12-20:
Ahh, I use a development board, so ignore all errors about temperature and hardware.

Jason2866, 2023-12-20:
Can you try this one https://github.com/Jason2866/mgos32-to-tasmota32/releases/download/v13.3.0/mgos32-to-tasmota32-Plus1PM.zip
with your Plus1PM?

RavenSystem, 2023-12-20:
Now fails at partition table.

```
shelly_ota.cpp:403      partition-table-tasmota.bin: 3072 bytes (hdr 85 desc 0 crc 5991bd2d)
shelly_ota.cpp:311      Aborting update: backend 0x3ffe2e98 aborts update at file partition-table-tasmota.bin: -4: app / fs already written, can't update pt
```

[screenlog.txt](https://github.com/tasmota/mgos32-to-tasmota32/files/13732722/screenlog.txt)

Jason2866, 2023-12-20:
Can substitute this one with orig. file too. But when the firmware file needs correct signing too, that's the end of OTA convert. New version to try in a few minutes.

Jason2866, 2023-12-20:
Next one https://github.com/Jason2866/mgos32-to-tasmota32/releases/download/v13.3.1/mgos32-to-tasmota32-Plus1PM.zip

If this one fails with Tasmota firmware -> RiP or maybe Shelly accepts orig firmware to downgrade to v1.0.8

RavenSystem, 2023-12-20:
v13.3.1 fails again at bootloader:

```
shelly_ota.cpp:403      bootloader.bin: 25728 bytes (hdr 72 desc 0 crc 1f976993)
shelly_ota.cpp:311      Aborting update: backend 0x3ffe299c aborts update at file bootloader.bin: -4: app / fs already written, can't update boot
shelly_ota.cpp:234      OTA status error, 75 %
```

Jason2866, 2023-12-20:
Thx makes no sense to do further tries, will install a dumped Shelly PM1 firmware on a dev board and update to latest firmware. Don't know why this happens now. Will repack a orig. update firmware package to see if this one is accepted and runs through.

RavenSystem, 2023-12-20:
> Thx makes no sense to do further tries, will install a dumped Shelly PM1 firmware (tomorrow) on a dev board and update to latest firmware. Don't know why this happens now. Will repack a orig. update firmware package to see if this one is accepted and runs through.

Many thanks!

RavenSystem, 2023-12-21:
@Jason2866 Shelly OTA now has a stupid requirement: Files must be added to ZIP in a specific order.

I took original manifest.json and I changed app bin file with mine from HAA firmware:

```
{
 "name": "Plus1PM",
 "platform": "esp32",
 "version": "1.1.0",
 "build_id": "20231219-133934/1.1.0-g34b5d4f",
 "build_timestamp": "2023-12-19T13:39:34Z",
 "parts": {
  "boot": {
   "type": "boot",
   "src": "bootloader.bin",
   "addr": 4096,
   "size": 25728,
   "cs_sha1": "3735c2d6c4f5fae10d2c3043ddab2dd31b084363",
   "cs_sha256": "1182108aeb648b6dba26e43db3bd80c79c2bad49a8846c4a2c127e6bd6aaed7e",
   "encrypt": true,
   "min_version": "0.0.0"
  },
  "pt": {
   "type": "pt",
   "src": "partition-table.bin",
   "addr": 32768,
   "size": 4096,
   "cs_sha1": "a8ee7fd30729b2d50a3f062070e17dc3f414145b",
   "cs_sha256": "bc91205f649b82825c365d9f3646a8e5c308d5d0755e48a31e56ef6c8899784b",
   "encrypt": true
  },
  "otadata": {
   "type": "otadata",
   "src": "otadata.bin",
   "size": 8192,
   "cs_sha1": "471af83fac6e5bd49355879b3acbed10bab6f423",
   "cs_sha256": "dd76e54da59f6824fb93d2b92b36d10932b06801d167b25c08ebc27e78391b6c",
   "encrypt": true,
   "ptn": "otadata"
  },
  "nvs": {
   "type": "nvs",
   "size": 16384,
   "fill": 255,
   "ptn": "nvs"
  },
  "app": {
   "type": "app",
   "src": "Plus1PM.bin",
   "size": 654016,
   "cs_sha1": "54ac1d7adbd60f487de13e673de5e3d6eea11dcb",
   "cs_sha256": "3e70bc1fbae19655efb23df5d142405c0af71722d926c12c78d62724ce795091",
   "encrypt": true,
   "ptn": "app_0"
  },
  "fs": {
   "type": "fs",
   "src": "fs.img",
   "size": 458752,
   "cs_sha1": "0d98936b53a58c5fd3e46622f9f3c7ea0572ce3f",
   "cs_sha256": "6f80ea7523c89a4ef0cfa639a32136828b9005a9281d55c21b21ea6d923743ad",
   "encrypt": true,
   "ptn": "fs_0",
   "fs_size": 458752
  }
 }
}
```

Then, I added all files to `mytest.zip` file, one by one, in the same order as manifest.json:
```
iMac-de-Jose:mgos2HAA jose$ zip -0 mytest.zip workdir/manifest.json
  adding: workdir/manifest.json (stored 0%)
iMac-de-Jose:mgos2HAA jose$ zip -0 mytest.zip workdir/bootloader.bin
  adding: workdir/bootloader.bin (stored 0%)
iMac-de-Jose:mgos2HAA jose$ zip -0 mytest.zip workdir/partition-table.bin
  adding: workdir/partition-table.bin (stored 0%)
iMac-de-Jose:mgos2HAA jose$ zip -0 mytest.zip workdir/otadata.bin
  adding: workdir/otadata.bin (stored 0%)
iMac-de-Jose:mgos2HAA jose$ zip -0 mytest.zip workdir/Plus1PM.bin
  adding: workdir/Plus1PM.bin (stored 0%)
iMac-de-Jose:mgos2HAA jose$ zip -0 mytest.zip workdir/fs.img
  adding: workdir/fs.img (stored 0%)
```

And flashing it from Shelly web UI did the work.

RavenSystem, 2023-12-21:
For example, this error is saying that `boot`part must be before `app` and `fs` parts:

`aborts update at file bootloader.bin: -4: app / fs already written, can't update boot`

Jason2866, 2023-12-21:
@RavenSystem  Great finding!

RavenSystem, 2023-12-21:
Pack files into ZIP in this order and all will work fine again:

```
manifest.json
bootloader-tasmota.bin
partition-table-tasmota.bin
otadata-tasmota.bin
tasmota32solo1.bin
fs-4MB-tasmota.img
```

flinknet, 2023-12-21:
As I understand it, the compression method must be "store". On most zip-applications the default is "deflate".

TD-er, 2023-12-21:
So it is more like a tar file instead of a zip file?

gabi1324, 2023-12-21:
> As I understand it, the compression method must be "store". On most zip-applications the default is "deflate".

Thank you! This solved the issue.

> So it is more like a tar file instead of a zip file?

I am using winrar on windows and when adding the files to the zip you need to set it from normal to store.

Jason2866, 2023-12-21:
it is a zip without compression. Changed in branch development the json create. But still wrong order.
Being bad in bash scripting, i think this part is damaging the needed order
```
generatezip () {

shelly_device=$1
app_file=$2
fs_file=$3
if [[ $4 -eq 2 ]]
then
  platform="esp32c3"
  boot_file="bootloader-tasmota-c3.bin"
  boot_file_addr=0
  partition_file="C3_4MB_partition-table.bin"
else
  platform="esp32"
  boot_file="bootloader-tasmota.bin"
  boot_file_addr=4096
  partition_file="partition-table-tasmota.bin"
fi
```

Jason2866, 2023-12-21:
Currently it generates this manifest.json

```
{
  "name": "Plus1PM",
  "platform": "esp32",
  "version": "12.5.0",
  "build_id": "20231219-131446/tasmota-12.5.0",
  "build_timestamp": "2023-12-19T13:14:46Z",
  "parts": {
    "app": {
      "type": "app",
      "src": "tasmota32solo1.bin",
      "size": 1555216,
      "cs_sha1": "1303331c42f73dbba205d0ec5f4aed154495147d",
      "cs_sha256": "891903b57caffd17578cdffbff4fa5cb780f42e692acd4c5c6aa6cb298a5f940",
      "encrypt": true,
      "ptn": "app_0"
    },
    "boot": {
      "type": "boot",
      "src": "bootloader-tasmota.bin",
      "addr": 4096,
      "size": 15648,
      "cs_sha1": "83f3bdd324ac53c2240dab74753923a05f916293",
      "cs_sha256": "bc429d1ac7076e701e69341dbee3ca2144cddb205849a3e486ba8e9272222093",
      "encrypt": true,
      "update": true
    },
    "fs": {
      "type": "fs",
      "src": "fs-4MB-tasmota.img",
      "size": 458752,
      "cs_sha1": "d672dab8aa71b2d55b29d0835ffbaacb256c8b95",
      "cs_sha256": "d622e534b6246cf920cf688f53cbaa5adefa476d23616b0c1bc05352804f6f60",
      "fs_size": 458752,
      "encrypt": true,
      "ptn": "fs_1"
    },
    "nvs": {
      "type": "nvs",
      "size": 16384,
      "fill": 255,
      "ptn": "nvs"
    },
    "otadata": {
      "type": "otadata",
      "src": "otadata-tasmota.bin",
      "size": 8192,
      "cs_sha1": "471af83fac6e5bd49355879b3acbed10bab6f423",
      "cs_sha256": "dd76e54da59f6824fb93d2b92b36d10932b06801d167b25c08ebc27e78391b6c",
      "encrypt": true,
      "ptn": "otadata"
    },
    "pt": {
      "type": "pt",
      "src": "partition-table-tasmota.bin",
      "addr": 32768,
      "size": 3072,
      "cs_sha1": "7b8dc7e877846775ea51215bb0294d070d5dabcb",
      "cs_sha256": "a395fa078f94250f482f23f0ef2625d913bb465c7458efc93616e5eb3f40b1d8",
      "encrypt": true
    }
  }
}
```

Jason2866, 2023-12-21:
Still failing with this manifest
```
{
  "name": "Plus1PM",
  "platform": "esp32",
  "version": "12.5.0",
  "build_id": "20231221-152737/tasmota-12.5.0",
  "build_timestamp": "2023-12-21T15:27:37Z",
  "parts": {
    "boot": {
      "type": "boot",
      "src": "bootloader-tasmota.bin",
      "addr": 4096,
      "size": 15648,
      "cs_sha1": "83f3bdd324ac53c2240dab74753923a05f916293",
      "cs_sha256": "bc429d1ac7076e701e69341dbee3ca2144cddb205849a3e486ba8e9272222093",
      "encrypt": true,
      "min_version": "0.0.0"
    },
    "pt": {
      "type": "pt",
      "src": "partition-table-tasmota.bin",
      "addr": 32768,
      "size": 3072,
      "cs_sha1": "7b8dc7e877846775ea51215bb0294d070d5dabcb",
      "cs_sha256": "a395fa078f94250f482f23f0ef2625d913bb465c7458efc93616e5eb3f40b1d8",
      "encrypt": true
    },
    "otadata": {
      "type": "otadata",
      "src": "otadata-tasmota.bin",
      "size": 8192,
      "cs_sha1": "471af83fac6e5bd49355879b3acbed10bab6f423",
      "cs_sha256": "dd76e54da59f6824fb93d2b92b36d10932b06801d167b25c08ebc27e78391b6c",
      "encrypt": true,
      "ptn": "otadata"
    },
    "nvs": {
      "type": "nvs",
      "size": 16384,
      "fill": 255,
      "ptn": "nvs"
    },
    "app": {
      "type": "app",
      "src": "tasmota32solo1.bin",
      "size": 1555216,
      "cs_sha1": "1303331c42f73dbba205d0ec5f4aed154495147d",
      "cs_sha256": "891903b57caffd17578cdffbff4fa5cb780f42e692acd4c5c6aa6cb298a5f940",
      "encrypt": true,
      "ptn": "app_0"
    },
    "fs": {
      "type": "fs",
      "src": "fs-4MB-tasmota.img",
      "size": 458752,
      "cs_sha1": "d672dab8aa71b2d55b29d0835ffbaacb256c8b95",
      "cs_sha256": "d622e534b6246cf920cf688f53cbaa5adefa476d23616b0c1bc05352804f6f60",
      "encrypt": true,
      "ptn": "fs_1",
      "fs_size": 458752
    }
  }
}
```

RavenSystem, 2023-12-21:
@Jason2866 You must add files one by one into the ZIP file in the right order, as the bottom of https://github.com/tasmota/mgos32-to-tasmota32/issues/31#issuecomment-1865993407

Jason2866, 2023-12-21:
@RavenSystem I dont see whats wrong in the script https://github.com/tasmota/mgos32-to-tasmota32/blob/development/build-zip.sh

```
shelly_ota.cpp:403      manifest.json: 1798 bytes (hdr 71 desc 0 crc 0de5347f)
shelly_ota.cpp:523      fw: Plus1PM ver 12.5.0 20231221-160724/tasmota-12.5.0
shelly_update.cpp:305   incoming fw signatures: 00
shelly_ota_esp32_ba:477 Will write to slot 1
shelly_ota.cpp:403      bootloader-tasmota.bin: 15648 bytes (hdr 80 desc 0 crc 160e0a45)
shelly_notification:209 Event from sys: {"component":"sys", "event":"ota_begin", "msg":"in_progress", "ts":1703175078.61}
shos_rpc_inst.c:230     ota.write via WS_in 192.168.2.202:54349
shos_rpc_inst.c:230     ota.write via WS_in 192.168.2.202:54349
shos_rpc_inst.c:230     ota.write via WS_in 192.168.2.202:54349
shos_init.c:86          New min heap free: 109772
shos_rpc_inst.c:230     ota.write via WS_in 192.168.2.202:54349
shos_rpc_inst.c:230     ota.write via WS_in 192.168.2.202:54349
shos_rpc_inst.c:230     ota.write via WS_in 192.168.2.202:54349
shos_rpc_inst.c:230     ota.write via WS_in 192.168.2.202:54349
shos_rpc_inst.c:230     ota.write via WS_in 192.168.2.202:54349
shelly_ota_esp32_ba:681 Boot: cur 00040127, new 00000000, min 00000000, update? 0
shelly_ota.cpp:403      partition-table-tasmota.bin: 3072 bytes (hdr 85 desc 0 crc 5991bd2d)
shos_rpc_inst.c:230     ota.write via WS_in 192.168.2.202:54349
shos_rpc_inst.c:230     ota.write via WS_in 192.168.2.202:54349
shelly_ota.cpp:311      Aborting update: backend error while processing data: -1 (-10: fs_1 not found in the new PT)
shelly_ota.cpp:234      OTA status error, 1 %
shelly_ota_source_rp:21 leftover RPC source 0x3ffe22dc
shelly_notification:209 Event from sys: {"component":"sys", "event":"ota_error", "msg":"error", "ts":1703175079.79}
shos_rpc_inst.c:230     ota.write via WS_in 192.168.2.202:54349
```

RavenSystem, 2023-12-21:
@Jason2866

Try this script:
```shell
#!/bin/bash

#Philipp "3D" ten Brink
#2023-01-19

#GNU General Public License v3.0 see LICENSE

mkdir output

#tasmota_version=$(curl --silent "https://api.github.com/repos/arendst/Tasmota/releases/latest" | grep '"tag_name":' | sed -E 's/.*"([^"]+)".*/\1/')
tasmota_version="12.5.0"

curl -o build-files/tasmota32.bin http://ota.tasmota.com/tasmota32/release-12.5.0/tasmota32.bin
curl -o build-files/tasmota32solo1.bin http://ota.tasmota.com/tasmota32/release-12.5.0/tasmota32solo1.bin
curl -o build-files/tasmota32-lvgl.bin http://ota.tasmota.com/tasmota32/release-12.5.0/tasmota32-lvgl.bin
curl -o build-files/tasmota32c3.bin http://ota.tasmota.com/tasmota32/release-12.5.0/tasmota32c3.bin

generatezip () {

shelly_device=$1
app_file=$2
fs_file=$3
if [[ $4 -eq 2 ]]
then
  platform="esp32c3"
  boot_file="bootloader-tasmota-c3.bin"
  boot_file_addr=0
  partition_file="C3_4MB_partition-table.bin"
else
  platform="esp32"
  boot_file="bootloader-tasmota.bin"
  boot_file_addr=4096
  partition_file="partition-table-tasmota.bin"
fi

#General:
otadata_file="otadata-tasmota.bin"

#Generated:

build_id=$(date '+%Y%m%d-%H%M%S')
build_date=$(date '+%Y-%m-%dT%H:%M:%SZ')
app_file_size=$(wc -c build-files/$app_file | awk '{print $1}')
app_file_cs1=$(shasum -a1 build-files/$app_file | awk '{print $1}')
app_file_cs256=$(shasum -a256 build-files/$app_file | awk '{print $1}')
boot_file_size=$(wc -c build-files/$boot_file | awk '{print $1}')
boot_file_cs1=$(shasum -a1 build-files/$boot_file | awk '{print $1}')
boot_file_cs256=$(shasum -a256 build-files/$boot_file | awk '{print $1}')
fs_file_size=$(wc -c build-files/$fs_file | awk '{print $1}')
fs_file_cs1=$(shasum -a1 build-files/$fs_file | awk '{print $1}')
fs_file_cs256=$(shasum -a256 build-files/$fs_file | awk '{print $1}')
otadata_file_size=$(wc -c build-files/$otadata_file | awk '{print $1}')
otadata_file_cs1=$(shasum -a1 build-files/$otadata_file | awk '{print $1}')
otadata_file_cs256=$(shasum -a256 build-files/$otadata_file | awk '{print $1}')
partition_file_size=$(wc -c build-files/$partition_file | awk '{print $1}')
partition_file_cs1=$(shasum -a1 build-files/$partition_file | awk '{print $1}')
partition_file_cs256=$(shasum -a256 build-files/$partition_file | awk '{print $1}')

JSON_STRING=$( jq -n \
                    --arg name "$shelly_device" \
                    --arg platform "$platform" \
                    --arg version "$tasmota_version" \
                    --arg build_id "$build_id/tasmota-$tasmota_version" \
                    --arg build_timestamp "$build_date" \
                    --arg app_file "$app_file" \
                    --argjson app_file_size $app_file_size \
[... 37 lines cut ...]

ShellyPlus=( PlusHT PlusPlugS PlusPlugUK PlusPlugIT PlusPlugUS PlugUS PlusI4 PlusWallDimmer Plus1PM Plus1 Plus2PM Plus10V )
for i in "${ShellyPlus[@]}"
do
    generatezip $i "tasmota32solo1.bin" "fs-4MB-tasmota.img" 1
done

ShellyPro8MB=( Pro1 Pro1PM Pro2 Pro2PM Pro3 )
for i in "${ShellyPro8MB[@]}"
do
    generatezip $i "tasmota32.bin" "fs-8MB-tasmota.img" 1
done

ShellyPro8MBlvgl=( Pro4PM )
for i in "${ShellyPro8MBlvgl[@]}"
do
    generatezip $i "tasmota32-lvgl.bin" "fs-8MB-tasmota.img" 1
done

ShellyPro16MB=( Pro3EM )
for i in "${ShellyPro16MB[@]}"
do
    generatezip $i "tasmota32.bin" "fs-16MB-tasmota.img" 1
done
```

Jason2866, 2023-12-21:
@RavenSystem  Same error with your changed script
```
shelly_ota.cpp:523      fw: Plus1PM ver 12.5.0 20231221-162638/tasmota-12.5.0
shelly_update.cpp:305   incoming fw signatures: 00
shelly_ota_esp32_ba:477 Will write to slot 1
shelly_ota.cpp:403      bootloader-tasmota.bin: 15648 bytes (hdr 80 desc 0 crc 160e0a45)
shelly_notification:209 Event from sys: {"component":"sys", "event":"ota_begin", "msg":"in_progress", "ts":1703176052.80}
shos_rpc_inst.c:230     ota.write via WS_in 192.168.2.202:54349
shos_rpc_inst.c:230     ota.write via WS_in 192.168.2.202:54349
shos_rpc_inst.c:230     ota.write via WS_in 192.168.2.202:54349
shos_rpc_inst.c:230     ota.write via WS_in 192.168.2.202:54349
shos_rpc_inst.c:230     ota.write via WS_in 192.168.2.202:54349
shos_rpc_inst.c:230     ota.write via WS_in 192.168.2.202:54349
shos_rpc_inst.c:230     ota.write via WS_in 192.168.2.202:54349
shos_rpc_inst.c:230     ota.write via WS_in 192.168.2.202:54349
shelly_ota_esp32_ba:681 Boot: cur 00040127, new 00000000, min 00000000, update? 0
shelly_ota.cpp:403      partition-table-tasmota.bin: 3072 bytes (hdr 85 desc 0 crc 5991bd2d)
shos_rpc_inst.c:230     ota.write via WS_in 192.168.2.202:54349
shos_rpc_inst.c:230     ota.write via WS_in 192.168.2.202:54349
shelly_ota.cpp:311      Aborting update: backend error while processing data: -1 (-10: fs_1 not found in the new PT)
shelly_ota.cpp:234      OTA status error, 1 %
```

RavenSystem, 2023-12-21:
@Jason2866 Something is wrong with the partition table file and filesystem file. You have a "fs_1" in your manifest, but "fs_1" is missing in partition table file binary. This is a new issue introduced now. Maybe a little mess testing with files. My script is the same as main branch, with right order of zipped files.

Jason2866, 2023-12-21:
Progress, when using the orig images and just replacing the firmware image it is successfull

RavenSystem, 2023-12-21:
Change fs_0 with spiffs.

Jason2866, 2023-12-21:
No, using the Tasmota FS image does work. Just tested. So will use the orig. other files. Just changing the FS and the firmware image. The safeboot conversion will fix and rewrite all other.
Thx for your great help. Now it is just updating some files...

RavenSystem, 2023-12-21:
fs_1 issue is caused because Tasmota partition table file lacks "fs_1", and Shelly firmware wants it.

Jason2866, 2023-12-21:
Fixed with release v13.3.0

RavenSystem, 2023-12-21:
@Jason2866 Many thanks!!!

closed 2023-12-21 by RavenSystem

Jason2866, 2023-12-21:
@RavenSystem  Added you to the release :-)

RavenSystem, 2023-12-21:
> @RavenSystem Added you to the release :-)

Great!! Many thanks!!