tasmota/mgos32-to-tasmota32 #36: Shelly Plus 2PM

quenthal, 2024-01-12:
Shelly with it's newest firmware. Nothing happens when you try to update from .zip?

Jason2866, 2024-01-12:
No info provided at all. What Shelly firmware version? What does mean nothing happens?

quenthal, 2024-01-12:
Oh, my bad :).

Firmware is 1.1.0.

I have succesfully upgraded some of the older devices before. I think it was 1.0.8 or even older.

With this it seems like that nothing just happens. In Shelly WebUI I go to Settings, Firmware. I either drag the latest mgos32-to-tasmota .zip for Plus 2PM or choose it manually by browsing local files -> Click "Update from file". Nothing happens . Of course there might be some quick uploading process, but no visible progress indication.

Kris1166, 2024-01-19:
Hi,
I have a similiar problem, maybe this helps.

First, thanks for your good and continious work.

I tried to flash a shelly plus 2pm, I followed your instructions:
- Shelly Firmware was: 1.0.3 -> it updated to 1.1.0
- All Steps till Step 8 worked fine. I used the tasmota auto config feature and the device rebootet.
- after that, no AccessPoint reapeared. Powercycle did not change anything.
- I attached the device to a FTDI Serial to USB
- it lists repeatedly that it cannot find several files:rst:
-------------------------------------------------------------- 
0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 188777542, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:184
load:0x40078000,len:12736
load:0x40080400,len:2716
entry 0x40080594

00:00:00.002 HDW: ESP32-U4WDH-D
00:00:00.029 UFS: FlashFS mounted with 336 kB free
00:00:00.032 TFS: File '.settings' not found
00:00:00.033 CFG: Loaded from NVS, Count 2
00:00:00.034 FRC: Some settings have been reset (113)
00:00:00.057 TFS: File '.drvset003' not found
00:00:00.057 CFG: Energy use defaults as file system not ready or file not found
00:00:00.061 Project tasmota - Tasmota Version 12.5.0(solo1)-2_0_7(2023-04-17T08               :07:06)
00:00:00.274 WIF: WifiManager active for 3 minutes

Brownout detector was triggered

ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 188777542, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:184
load:0x40078000,len:12736
load:0x40080400,len:2716
entry 0x40080594

00:00:00.002 HDW: ESP32-U4WDH-D
00:00:00.029 UFS: FlashFS mounted with 336 kB free
00:00:00.032 TFS: File '.settings' not found
00:00:00.033 CFG: Loaded from NVS, Count 2
00:00:00.034 FRC: Some settings have been reset (114)
00:00:00.057 TFS: File '.drvset003' not found
00:00:00.057 CFG: Energy use defaults as file system not ready or file not found
00:00:00.061 Project tasmota - Tasmota Version 12.5.0(solo1)-2_0_7(2023-04-17T08               :07:06)
00:00:00.274 WIF: WifiManager active for 3 minutes

Brownout detector was triggered

ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 188777542, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:184
load:0x40078000,len:12736
load:0x40080400,len:2716
entry 0x40080594

00:00:00.002 HDW: ESP32-U4WDH-D
00:00:00.029 UFS: FlashFS mounted with 336 kB free
00:00:00.032 TFS: File '.settings' not found
00:00:00.033 CFG: Loaded from NVS, Count 2
00:00:00.034 FRC: Some settings have been reset (115)
00:00:00.057 TFS: File '.drvset003' not found
00:00:00.057 CFG: Energy use defaults as file system not ready or file not found
00:00:00.061 Project tasmota - Tasmota Version 12.5.0(solo1)-2_0_7(2023-04-17T08               :07:06)
00:00:00.274 WIF: WifiManager active for 3 minutes

-------------------------------------------------------------------

If you want me to test something for you, feel free to ask.

Last but not least, where do I find a tasmota firmware for the plus 2PM to flash it by wire? Or is it possible to use a plain firmware from the tasmota page?

Thanks in advance.
Sincerely Kris

Jason2866, 2024-01-20:
Everything is normal, your FTDI does not deliver enough current to power the device
`Brownout detector was triggered`

TeHashX, 2024-01-29:
I successfully converted a Shelly Plus 2PM device with firmware v1.1.0 using step by step instruction by flashing v13.3.1 https://github.com/tasmota/mgos32-to-tasmota32/releases/download/v13.3.1/mgos32-to-tasmota32-Plus2PM.zip

Jason2866, 2024-01-29:
Closing, converting does work. Issue is not reproducable.

closed 2024-01-29 by Jason2866