tasmota/mgos32-to-tasmota32 #37: Shelly1 Plus migration failed

stetorm, 2024-01-13:
I followed the procedure to update firmware of my Shelly1 Plus with the tasmota one: mgos32-to-tasmota32-Plus1.zip. Everything worked fine until I executed Step 8:

"..Configure the device using Tasmota Auto configuration. (Configuration > Auto-configuration > Select new auto-configuration) Select your device and hit enter. This replaces the locked bootloader (without this step device gets bricked with the next firmware"

After that my Shelly1 never turned on again, no led light on, no online presence :(

stetorm, 2024-01-13:
I switched to AC power supply, now I see device online on a different ip (192.168.1.111) but the only port open is: 6668 but is not web server. Any suggestion?

Jason2866, 2024-01-20:
No logs, so don't know what happens. Can't reproduce the issue.

346L3, 2024-01-20:
I'm having the same issue on a Shelly Plus 2PM.
Booted fine into tasmota first time, but after using the autoconfiguration it never came back...

Can I provide any information that might help analyze the issue?

Jason2866, 2024-01-20:
Logs ;-)

346L3, 2024-01-20:
For the failed process? That's too late, I guess. Or are logs persistent in flash?
Or current bootlog (if there are any?) via serial console?

I have no experience with this hardware platform. ;)

Jason2866, 2024-01-20:
The log of the failed process would have been interesting. But that's history. Nothing is stored. The log of the current start up, maybe gives an idea what's happening (or not).

346L3, 2024-01-26:
My apologies. At least my issue was completely my fault (the OPs might be not!):

After trying procedures for resetting the device using the button etc. (no luck).
I finally came around to connect my USB-serial adaptor (I was on vacation and the adapter is in my office...):
After bootlooping (brown out detection, the dongle can't provide enough power on its 3.3V OUT for WiFi apparently...) and connecting it to a different PSU after that, it stated "some options were reset" and started a WiFi AP - which AFAIR didn't show up after the supposed softbrick, i configured it as WiFi Client and it never reconnected.
It still came up as "ESP32-DevKit", so the autoconfig wasn't fully applied.
Going to auto-configure it prompted me to reapply the last config, which was for the Shelly Pro 2PM, not plus. So I misclicked, which is quite embarassing as i converted multiple Shelly so far :-(

Jason2866, 2024-01-29:
Can not reproduce the issue. My tries where all successful.

closed 2024-01-29 by Jason2866