tasmota/mgos32-to-tasmota32 #38: Shelly Plus 1 Mini update to tasmota not working

rakeshnsw, 2024-01-15:
I followed all the steps to update firmware of my Shelly Plus 1 mini with tasmota firmware. 
As per the directions, I used mgos32-to-tasmota32-Plus1Mini.zip firmware zip file.
But when I hit update button, I get this error message in console "no update in progress" and Shelly interface does nothing.
Shelly Plus 1 mini is already updated to it's latest firmware 1.1.0
Error snapshot is attached. 

[image]

Jason2866, 2024-01-15:
Using wrong update zip. You have a gen3 Mini1. So use https://github.com/tasmota/mgos32-to-tasmota32/releases/download/v13.3.1/mgos32-to-tasmota32-Mini1G3.zip

rakeshnsw, 2024-01-15:
My bad. Thanks. I was able to successfully update it to tasmota using Mini1G3 firmware

closed 2024-01-15 by rakeshnsw