tasmota/mgos32-to-tasmota32 #40: Plus1PMMini - Partition Wizard fails Step 2

Mugga6315, 2024-01-18:
I've flashed 1 out of 3 Plus1PMMini successfully, but the other 2 are stuck at Step 2 of the Partition Wizard.

This is a part of the log:
```
15:43:38.822 QPC: Reset
15:43:39.997 UPL: Step 1 Done
15:43:40.001 UPL: Starting step 2
15:43:40.181 UPL: flashing `safeboot` from http://ota.tasmota.com/tasmota32/release/tasmota32c3-safeboot.bin 712kB
15:43:40.182 UPL: error (internal_error) `app0` offset is not 0x10000
15:48:43.334 RSL: RESULT = {"OtaUrl":"http://ota.tasmota.com/tasmota32/release/tasmota32c3.bin"}
15:50:39.093 RSL: RESULT = {"OtaUrl":"http://ota.tasmota.com/tasmota32/release/tasmota32c3.bin"}
15:50:39.886 RSL: RESULT = {"OtaUrl":"http://ota.tasmota.com/tasmota32/release/tasmota32c3.bin"}
```
Any ideas? Can I just ota update normally?

Jason2866, 2024-01-18:
Delete everything in Tasmota Filesystem. Do the Autoconfig step again. Does PartitionWizard still shows `UPL: error (internal_error) `app0` offset is not 0x10000`?

Mugga6315, 2024-01-18:
Tried the steps, but still the same error:
`UPL: error (internal_error) `app0` offset is not 0x10000`

Jason2866, 2024-01-18:
Mhh, download this version (should be the same!) via the github download button NOT via mouse click "save as" and upload in FS (remove before any file with PartitionWizard in name)
https://github.com/arendst/Tasmota/blob/development/tasmota/berry/modules/Partition_Wizard.tapp

Mugga6315, 2024-01-18:
That worked!
Trying it now with the other one.

Jason2866, 2024-01-18:
Very very strange, since this PartitionWizard version should be loaded via Autoconfig too.

Mugga6315, 2024-01-18:
Yeah, I've also been confused. Because the first one flashed without any issues. And the other two, with that strange behaviour.

closed 2024-01-18 by Mugga6315