tasmota/mgos32-to-tasmota32 #41: Cannot flash on Shelly 1PM Pro

vpant, 2024-01-19:
I bought multiple shelly devices (mostly 1PMs and 2PMs) and I flashed everything with tasmota. The only one I cannot flash is the 1PM Pro that I bought to control my water heater.

The ota flash starts but then it is almost immidiately stops without any error except one in the console:
[image]

I also enabled the debug logs:
[diagnostics-shelly-pro1pm-debug-log.txt](https://github.com/tasmota/mgos32-to-tasmota32/files/13986641/diagnostics-shelly-pro1pm-debug-log.txt)

Jason2866, 2024-01-19:
Thx, for the log. Shelly changed (again) something.
This
```Aborting update: backend error while processing data: -1 (-10: New app_1 overlaps with current app_0)"```
shows that probably the partition scheme does not fit to the latest firmware.

macSE1990, 2024-03-10:
I have the same problem on Shelly Pro 1 v.1 (without PM)

Jason2866, 2024-04-12:
Bug identified. Working on the fix.

Jason2866, 2024-04-12:
Should be fixed with latest release 13.4.1

Jason2866, 2024-04-24:
Closing since no feedback

closed 2024-04-24 by Jason2866

macSE1990, 2024-04-24:
Confirming success for Shelly Pro 1 v.1
Sorry for the late answer.