tasmota/mgos32-to-tasmota32 #42: Shelly Plus1Mini - No Partition Manager

petereit, 2024-01-24:
Got to step 10: Under consoles, open the Partition Wizard and start the Safeboot Conversion by hitting the button "Start Migration". The conversion will update to the latest Tasmota version too.

I click Console, but there is no Partition Wizard option. These are my only options:

Console
Berry Scripting console
Manage File system
Main Menu

Jason2866, 2024-01-24:
So step 8 failed. Does the device has Internet access?

petereit, 2024-01-24:
Yes. The device has internet access. 

I was able to copy the Partition Wizard .tapp, reboot and start the Safeboot Conversion, but now it’s stuck in Safeboot.

Jason2866, 2024-01-24:
When step 8 is not completely successful (replacing bootloader) it stuck.
Only serial flash is the way to solve.

Jason2866, 2024-01-29:
Can not reproduce the issue. Tried a few times, always successful.

closed 2024-01-29 by Jason2866