tasmota/mgos32-to-tasmota32 #43: Shelly Plus Uni

elngejw, 2024-01-24:
Hi good day!

Do you also have a convert file I can use for Shelly Plus Uni?
https://www.shelly.com/en-nl/products/product-overview/shelly-plus-uni-1

Many thanks and best regards,

Geert-Jan

Jason2866, 2024-01-24:
There is a chance to get it done. Can you post a screenshot of the firmware update page?
In upper left corner there should be the intern Shelly device name shown.
Knowing the name, there should be possible to download a update package. All needed Infos to build a convert package should be there.

elngejw, 2024-01-24:
First of all thx for your quick reply!
Find attached the requested screenshot.

[image]

MinGeunJang, 2024-02-20:
I look forward to the tasmota firmware for Shelly Plus Uni.

Jason2866, 2024-04-05:
Added Shelly Plus Uni
The Autoconfig does not set any GPIOs since I don't have the device.
You have to find out the configuration yourself. Would be great if you post your findings.

MenesesPT, 2024-04-16:
I followed the instructions in the readme and was able to convert a Shelly Plus Uni to Tasmota successfully.

I then managed to find the following pinout:
| **Shelly** | **GPIO** | **Tasmota** |
|------|------|------|
| Data  |   13   | AM2301 (or DS18x20)
| Led    |   18   | Led_i 1
| Out 1 |   19   | Relay 1
| Out 2 |   21   | Relay 2
| N/A   |   25   | ADC Range 1
| Analog In   |   36   | ADC Input 1
| In 2  |   37   | Switch_n 2
| In 1  |   38   | Switch_n 1
| Count In  |   36   | Switch_n 3

Which gives the following template: 
`{"NAME":"Shelly Plus Uni","GPIO":[0,0,0,0,0,0,0,0,0,1216,0,0,0,0,320,224,0,225,0,0,0,4864,0,0,0,0,0,0,0,0,0,0,4704,193,192,194],"FLAG":0,"BASE":1}`

> [!NOTE]  
> 1. I did not test any sensors on **GPIO13** since I don't own any;
> 2. Changing **GPIO25** output value seems to change the values read on **GPIO36**, I used *ADC Range* but I'm not sure if that is correct;
> 3. The **Count In** pin was defined as a normal switch input.

Given that my shelly has an ESP32-U4WDH-D with two cores, I then replaced the solo version of Tasmota with the dual-core one without any issues.

Thanks for the conversion firmware!

Jason2866, 2024-04-17:
Thx, added to Autoconfig.

closed 2024-04-17 by Jason2866