tasmota/mgos32-to-tasmota32 #44: Template für Shelly Plus H&T

Joylancer, 2024-01-28:
Hallo Jason2866,
Ich habe mal einen Shelly Plus H&T nach der Anleitung geflasht, Tasmota läuft.
Wie ich dann auch mitbekommen habe, gibts noch kein Template.
Glaube auch, das noch die Display Driver fehlen.
Was ich schon zum laufen gebracht habe, ist der Temp-Hum Sensor.
Das ist ein SHT30.
SDA: GPIO15
SCL: GPIO02
Power über: GPIO4 (habe hier mal "Output High" eingesetzt.
Beim Display hört meine Fähigkeit leider auf.
Wenn du möchtest, kann ich dir gerne das Teil zuschicken, kannst es auch behalten, zum testen etc.
Habe davon noch einige im Einsatz und möchte hier lieber alles einheitlich.
VG
Peter

Jason2866, 2024-01-28:
Ich weiß nicht ob es grundsätzlich eine gute Idee ist Tasmota auf dieses Device zu installieren. Tasmota ist nicht auf Batteriebetrieb ausgelegt. Wenn auch die Unterstützung für das Display realisiert wäre, wird die Laufzeit mit Batterien wahrscheinlich sehr kurz ausfallen.
Damit ist der eigentliche Charme dieses Devices dahin.
Wenn also noch ein Backup der orig. Firmware vorhanden ist, diese zurück flashen und die Shelly mqtt Integration nutzen. Sollte diese nicht vorhanden sein, nehme ich das Angebot an mal zu schauen was mit Tasmota möglich ist. Kontakt über PM in Discord. Mein Username ist dort auch `Jason2866`

Joylancer, 2024-01-28:
Hast PM
Ich mach dann hier zu.

closed 2024-01-28 by Joylancer