tasmota/mgos32-to-tasmota32 #45: Migrate to safeboot partition layout problem

bilbolodz, 2024-02-03:
I've converted my Shelly I4 DC to Tasmota using mgos32-to-tasmota32 and it's working fine but I've tasmote version 12.5
I've tried to OTA update to newest (13.3.0) but upgrade fails due to: "Program flash size is larger than real flash size".
I've tried with Partition Wizard to "Migrate to safeboot partition layout" but after running it I've ended with ..... back to original Shelly firmware. Any ideas how to get newest Tasmota on Shelly I4 and possibility to upgrade it with OTA?

Jason2866, 2024-02-03:
Step 8 was not done immediately or was not successful. Serial flash is needed.

closed 2024-02-03 by Jason2866

bilbolodz, 2024-02-04:
Indeed step 8 was not done but which device profile should I use? There is no Shelly I4 profile:
[image]

bilbolodz, 2024-02-04:
Yep, my fault. Now all is working fine!