tasmota/mgos32-to-tasmota32 #46: Shelly 1 Plus without Web UI but accepting commands

Orkote, 2024-02-14:
I’ve installed Tasmota through OTA on Shelly 1 Plus. All the process seems OK but, after last reboot, I can see the tasmota wifi and connect with the device, but if I try to connect via web on 192.168.4.1 I’ve obtain a message ‘File not found’.

However, seems that the device is completely empty of configuration but accepts commands, so I can send commands like http://192.168.4.1/cm?cmnd=WifiConfig%200 I can see ｛"wifiConfig" ：｛"o"："Restart"｝｝and the device is restarted.

So, Can I fix it or configure my WiFi in order to connect? The SSId1 and Password1 commands seems not working, the device restarts and start is own WiFi again.

Jason2866, 2024-02-16:
Device connected to AC mains? If yes, only serial flashing is the way to get Tasmota running.

TD-er, 2024-02-16:
And make sure to **not** have it powered via mains when performing serial flashing, or else things may get seriously damaged (or you can get killed)

optio50, 2024-03-18:
This is what worked for me
curl -X POST -d '{"id":1,"method":"WiFi.SetConfig","params":{"config":{"sta":{"ssid":"YOUR SSID","pass":"YOUR PASSWORD","enable":true}}}}'

http://YOUR SHELLY IP/ota?url=https://github.com/tasmota/mgos32-to-tasmota32/releases/download/v13.3.1/mgos32-to-tasmota32-PlugUS.zip

closed 2024-04-01 by Jason2866

Orkote, 2024-04-02:
Not for me, is it not possible to pass all this parameters... and I think that you're using a very old parameters because some of them are changed. Following the official documentation is possible to send some commands via CURL but the device is not able to save on flash and all is lost on restart :(

Seems that is necessary to re-flash with the USB cable.