tasmota/mgos32-to-tasmota32 #48: Shelly Pro 2 - Firmware 1.2.2 - not possible to flash

RaidObServer, 2024-02-28:
My new Shelly Pro 2 has been updated to version 1.2.2. 
With this version it is not possible to download the current mgos32-to-tasmota32-Pro2.zip file.

[image](https://github.com/tasmota/mgos32-to-tasmota32/assets/112802138/94dfef01-3d6c-4e7c-a794-147312682c28)

Jason2866, 2024-02-28:
To rule out misunderstandings. The convert package was provided via file upload or via drop down in the marked area?
It is **not** possible to provide the convert package via Link!!

RaidObServer, 2024-02-28:
correct, the file was dragged and dropped into the marked area.

It is also interesting to note that the mgos32-to-tasmota32-Pro2 version 13.1 update runs up to 73%. However, it does not start with the current version 13.3.1. I have added both diagnostics files.

[diagnostics-shelly-pro2-debug-log_v13_1.txt](https://github.com/tasmota/mgos32-to-tasmota32/files/14439988/diagnostics-shelly-pro2-debug-log_v13_1.txt)
[diagnostics-shelly-pro2-debug-log_v13_3_1.txt](https://github.com/tasmota/mgos32-to-tasmota32/files/14439990/diagnostics-shelly-pro2-debug-log_v13_3_1.txt)

RavenSystem, 2024-02-29:
@Jason2866 Error is the same type of other times:

`app / fs already written, can't update boot`

Meaning that `boot` part must be added to zip file before than `app` and `fs`.

RavenSystem, 2024-02-29:
@Jason2866 FYI. I downloaded last Plus 1PM official firmware 1.2.2, and the official order of the files is:
- bootloader.bin
- partition-table.bin
- otadata.bin
- Plus1PM.bin
- fs.img

RavenSystem, 2024-02-29:
@Jason2866 I think current issue is the size of Tasmota binary.

Jason2866, 2024-03-13:
@RavenSystem  The currently used partition table is incompatible.
Guess the partition table has changed again for Shelly fw 1.2.x.
Can currently not investigate, no time for.

jeroenst, 2024-03-18:
I also upgraded a Shelly 2PM Plus to 1.2.2 and I can't find any way to downgrade it so I can flash Tasmota to it. Hopefully it will be possible in the future. I think I return the unit and get a new one with stock firmware.

Jason2866, 2024-04-04:
Try again, Issue should be fixed.

closed 2024-04-04 by Jason2866

dr-m, 2024-07-10:
For what it is worth, I successfully upgraded a freshly purchased Shelly 2 Pro to firmware 1.3.0 and then to Tasmota using the appropriate package. Thank you!