tasmota/mgos32-to-tasmota32 #51: no Plus10V under auto-configuration

Peeri90, 2024-03-04:
Hello,
i have a Problem with the step 8. i cant find the Shelly Plus10V under  Configuration > Auto-configuration > Select new auto-configuration. 

i use the mgos32-to-tasmota32-Plus10V.zip from Dec 31, 2023

[image]

berndh1979, 2024-03-06:
Same issue here. I guess that is why the device after flashing is not working anymore.

Jason2866, 2024-03-06:
Yes, there is no Autoconfig for the Plus10V. Simple reason idk what to configure.

Jason2866, 2024-04-05:
Added the device to the Autoconfig to prevent bricking during convert process.
There are NO GPIOs configured since I don't have the device.
You have to find out this yourself. It would be great if you post your found config.

Jason2866, 2024-04-13:
Update Autoconfig with GPIO settings

closed 2024-04-13 by Jason2866