tasmota/mgos32-to-tasmota32 #52: Unable to flash Mini1G3 on Firmware 1.14

arcoast, 2024-03-13:
I've just purchased a Mini1G3 with the hope of flashing to Tasmota as I tend to do with all my devices.

Unfortunately I think the v1.14 firmware prevents this, unless I'm missing something really simple, as there's no webui element to upload the zip to.

[image]

Not sure if there's any way around this, whilst I've flashed a lot of devices to Tasmota in the past, including some original Shelly devices with the obscenely small pin headers :smile:  this is my first outing with a Shelly 1 Mini Gen 3.

Jason2866, 2024-03-13:
You have to look in settings menu

arcoast, 2024-03-13:
I must be missing something, that is the setting menu in the screenshot I posted. 

I've gone through every setting in every menu and there's no firmware upload section anywhere, unless I'm missing a step.

arcoast, 2024-03-19:
OK, time for me to put my hands up.  I was trying to flash a Mini1G3, the issue I was hitting is I'd actually wired up my old Shelly 1L.  I'm an idiot, looked at it again tonight and massive facepalm.  So sorry :man_facepalming:

closed 2024-03-19 by arcoast