tasmota/mgos32-to-tasmota32 #53: 2PM schaltet selbstständig aus

Marc9485, 2024-03-13:
Hey, ich hab meinen Shelly 2PM mit Tasmota geflasht aber habe das Problem, das dieser beide Kanäle selbst ausschaltet. Home Assistant schließe ich als Fehler aus da dort kekn Dienst erscheint der ausschaltet. Kann mir da jemand helfen?
[image]

Jason2866, 2024-03-13:
Not the correct place for discussions which are not related to the convert process.

closed 2024-03-13 by Jason2866

Marc9485, 2024-03-13:
OK, my thought was that there was a problem with the flash because Home Assistant, as you can see in the picture, doesn't give a command to switch it off but only recognizes that it has been switched off, so the Shelly must have a life of its own. And there is no field for “discussions” here otherwise I would ask it there.