tasmota/mgos32-to-tasmota32 #54: Fixing issue with Shelly firmware v1.2.0 and newer

RavenSystem, 2024-03-22:
@Jason2866 I updated my Shelly Plus 1PM to v1.2.2, and I was able to install Tasmota (including Partition Wizard step) without any issue using current `mgos32-to-tasmota32-Plus1PM.zip` file. I see that you added to README that Shelly FW 1.2.0 and newer is incompatible. I can help but, what is the issue?

Jason2866, 2024-03-22:
I added the warning because of user feedbacks. Currently i have no time to investigate.
Support highly appreciated. Imho best would be, replacing all "helper" files with the ones from Shelly update packages. So there are no differences in the update package, just the Trojan horse Tasmota.

RavenSystem, 2024-03-22:
I bought a new Shelly Plus 2PM just now from Amazon. I hope it has a recent firmware version to do tests. I will feedback here, but for now, I can not reproduce any issue with 1.2.2. All is working fine with Tasmota OTA installation.

Zumili, 2024-03-22:
I was just facing the same situation. Got an actual Shell Plus 2PM just 2 hours ago from Amazon. I intentionally updated the firmware to the latest 1.2.2, cannot surely say what initial firmware was installed. Then I read here: https://templates.blakadder.com/shelly_plus_2PM.html, that OTA will not work with versions greater v1.10+, tried to downgrade to 0.14, but the provided links are no more working (you will always get the latest version), so I tied to do the OTA anyway, and it also worked to update from Shelly 1.2.2 to Tasmota.

RavenSystem, 2024-03-22:
Is it possible that some users miss the auto-configuration step?

Zumili, 2024-03-22:
I need to write an update. At first, it seemed to work and Tasmota was running on the device, but after running the Auto-configuration the original Shelly firmware was showing up instead of Tasmota. I tried to reflash Tasmota again but now it is not working anymore. After flashing the mgos32-to-tasmota32-Plus2PM.zip the original firmware is back again and Tasmota is not loading. I repeated the steps I did before several times and was no longer lucky enough to get Tasmota running.

RavenSystem, 2024-03-22:
> I need to write an update. At first, it seemed to work and Tasmota was running on the device, but after running the Auto-configuration the original Shelly firmware was showing up instead of Tasmota. [...]

Thanks for your feedback. It is important.

RavenSystem, 2024-03-23:
I have my new Shelly Plus 2PM. Chip is ESP32-U4WDH (revision v3.1).

It comes with firmware v1.0.7 (20231031-152228/1.0.7-g5db02bd).

Bootloader log is:
```
rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 188777542, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:6288
load:0x40078000,len:13996
load:0x40080400,len:3592
entry 0x40080664
I (28) boot: ESP-IDF 4.4.4-a5 2nd stage bootloader
I (28) boot: compile time 08:49:29
I (28) boot: chip revision: v3.1
I (31) boot_comm: chip revision: 3, min. bootloader chip revision: 0
I (38) boot.esp32: SPI Speed      : 80MHz
I (43) boot.esp32: SPI Mode       : DIO
I (47) boot.esp32: SPI Flash Size : 4MB
I (52) boot: Enabling RNG early entropy source...
I (57) boot: Partition Table:
I (61) boot: ## Label            Usage          Type ST Offset   Length   F
I (69) boot:  0 nvs              WiFi data        01 02 00009000 00004000 0
I (76) boot:  1 otadata          OTA data         01 00 0000d000 00002000 1
I (84) boot:  2 app_0            OTA app          00 10 00010000 00180000 1
I (92) boot:  3 fs_0             FS               01 82 00190000 00070000 1
I (99) boot:  4 app_1            OTA app          00 11 00200000 00180000 1
I (107) boot:  5 fs_1             FS               01 82 00380000 00070000 1
I (115) boot:  6 aux              unknown          55 00 003f0000 0000c000 0
I (122) boot:  7 shelly           Unknown data     01 88 003fc000 00004000 1
I (130) boot: End of partition table
I (134) boot: OTA data 0: seq 0x00000001, st 0x10, CRC 0x157a2b85, valid? 1
I (142) boot: OTA data 1: seq 0x00000002, st 0x11, CRC 0xddd0de27, valid? 1
I (150) boot_comm: chip revision: 3, min. application chip revision: 0
I (157) esp_image: segment 0: paddr=00200020 vaddr=3f400020 size=2a3e8h (173032) map
I (218) esp_image: segment 1: paddr=0022a410 vaddr=3ff80000 size=00004h (     4) load
I (218) esp_image: segment 2: paddr=0022a41c vaddr=3ffbdb60 size=03858h ( 14424) load
I (228) esp_image: segment 3: paddr=0022dc7c vaddr=40080000 size=0239ch (  9116) load
I (235) esp_image: segment 4: paddr=00230020 vaddr=400d0020 size=1308a0h (1247392) map
I (617) esp_image: segment 5: paddr=003608c8 vaddr=4008239c size=1b400h (111616) load
I (671) boot: Loaded app from partition at offset 0x200000
I (671) boot: Disabling RNG early entropy source...
```

I continue...

Jason2866, 2024-03-24:
@RavenSystem Don't think it is such a different bootloader (still IDF 4.4.x). The just named it. As long Tasmota boots up and is able to replace the bootloader (Step 8) and the partition scheme is adopted to Tasmota needs (with deleting Shelly OTA content), there is no active Shelly code left.

RavenSystem, 2024-03-24:
@Jason2866 Yes, it is a modified bootloader. Shelly added a boot counter to know if the new firmware is working fine. But, as I said, conversion to Tasmota works perfectly. I don't know how to reproduce issue. All my tries worked right.

Jason2866, 2024-03-24:
@RavenSystem  Thx for your tests and investigations. Could you provide a PR with the new Shelly files, you used in your convert zip? So we have anything as close as possible to the orig. update packages.

RavenSystem, 2024-03-24:
@Jason2866 Sorry if I was not clear. Current mgos32-to-tasmota32 files work right. Bootloader is not used because it is older than currently installed, but there is not any issue with this. Tasmota boots, and auto-configuration erase Shelly bootloader and install Tasmota bootloader. All is working fine with current conversion files.

Jason2866, 2024-03-24:
Ahh, okay. So we have to wait if we get an info how to reproduce a fail.

RavenSystem, 2024-03-24:
I have a report of an user with Shelly Mini 1 gen 3. Conversion to Tasmota was fine, without any issue. Done few hours ago.

michalpk, 2024-03-25:
> I have a report of an user with Shelly Mini 1 gen 3. Conversion to Tasmota was fine, without any issue. Done few hours ago.

Hi, today I tried to migrate a Mini 1 Pm Gen 3 and after the migration (step 10) my device became bricked... so I would say I managed replicate the issue. :D

tux1337, 2024-03-26:
Maybe the issue is device specific. I am not able to flash my Shelly Pro 2 with Shelly Firmware v1.2.2. 
There was an firmware 0.10.x - beta before which do not offer firmware upload via web interface, I updated then to v1.2.2 via Shelly OTA. After that I have exactly the same issue than described in #48

RavenSystem, 2024-03-27:
@Jason2866 Issue could be produced in the case that Tasmota firmware is not flashed at partition app0.

Auto-configurator installs new bootloader from Tasmota. This new bootloader doesn't understand Shelly OTA partition and boots from app0 by default. If Tasmota is flashed at partition app1, Shelly firmware will be booted, and uploading again Tasmota installer will be useless, because bootloader will not be flashed, and Tasmota bootloader will boot Shelly firmware again, leaving device stuck with Shelly firmware forever.

I suggest that, before reboot, auto-configurator downloads and flashes Tasmota firmware at partition app0 in the case Tasmota is running at app1 when it downloads and flashes bootloader.

Jason2866, 2024-03-27:
@RavenSystem 
It is done or should? do like this. Tasmota is copied in "other" OTA slot to overwrite Shelly firmware.

> I suggest that, before reboot, auto-configurator downloads and flashes Tasmota firmware at partition app0 in the case Tasmota is running at app1 when it downloads and flashes bootloader.

But, yes this is probably failing. The result is exactly what is described.

RavenSystem, 2024-03-27:
@Jason2866 Tasmota must be copied in "other" OTA slot immediately before or after flashing bootloader, BEFORE reboot. If device reboots with new bootloader keeping Shelly firmware, process will fail if Shelly firmware is at app0.

Jason2866, 2024-03-27:
@RavenSystem yes, i did understand already before ;-)

RavenSystem, 2024-03-27:
> @RavenSystem yes, i did understand already before ;-)

@Jason2866 Sorry :)

Jason2866, 2024-03-27:
@RavenSystem btw. you find the Berry scripts doing the lifting for example for the Shelly Plus 1 mini here:
https://github.com/tasmota/autoconf/tree/main/raw/esp32c3/Shelly_Plus1Mini
For the others it is the same.

s-hadinger, 2024-03-27:
I'm not sure to get all the details here. Do you know how to reproduce on a non-shelly device so I can try?

RavenSystem, 2024-03-31:
@Jason2866 Reading current installed Tasmota from running OTA slot and coping it to the other OTA slot will work?

Jason2866, 2024-03-31:
@RavenSystem quiet sure. Had some time to test. Could rebuild a Shelly factory image v1.2.2 and reproduce the issue. Shelly changed somehow the standard logic to set the OTA which is booted next time after reboot. So the only way to make sure it does always boot Tasmota is to have it in both OTA slots.
To answer your question, yes it is possible to copy Tasmota in "the other" OTA slot too. When Tasmota is active we have no restrictions to write anywhere in flash (sdkconfig "dangerous write).
This function needs to be added in the Berry script.

RavenSystem, 2024-03-31:
@Jason2866 I have done some tests. This steps will work:
- Flash new bootloader (this is already done in the current script).
- Read from running partition and erase and write to the other OTA partition (I have done using 4KB block size).
- Erase OTA Data partition (2 sectors size).
- Reboot.

You can remove all code about modify OTA Data partition. With these steps, it is useless.

Shelly will boot in an initial OTA status, booting from OTA-0, and bootloader will create all needed data in the OTA Data partition.

Jason2866, 2024-03-31:
@RavenSystem  Great! Seems you have already done the coding in Berry. Are you going to do a PR in repo https://github.com/tasmota/autoconf ?

RavenSystem, 2024-03-31:
No, sorry. I understand Berry, but I coded it in C using ESP-IDF.

s-hadinger, 2024-04-02:
I could reproduce but I'm a bit lost about the Shelly conversion process (I don't have any Shelly).

I flashed the Shelly1PM on ESP32. I then flashed `mgos32-to-tasmota32-Plus1PM.zip` from the repo.

There is normally `migrate_shelly.be` that flashes the new bootloader, but I don't know when this is supposed to happen?

s-hadinger, 2024-04-02:
After flashing the Tasmota firmware, the first 3 boots are made on Tasmota, and then it reverts to Shelly firmware. This is probably a crash resilient behavior until the firmware confirms a sucessful boot (which Tasmota doesn't).

However if I immediately install the `Shelly Pro 1PM` autoconf file from the web interface, then the bootloader is correctly installed and everything works fine.

So I don't understand why `mgos32-to-tasmota32-Plus1PM.zip` does not contain the autoconf file in its filesystem.

Jason2866, 2024-04-02:
@s-hadinger  We can't provide our own filesystem anymore. It is not accepted and flashed.
So the user has to do the Autoconfig step. In this step the bootloader is replaced and tried to set active boot partition for next boot.
Setting active boot partition does not work anymore. So to make sure Tasmota is booted next time it has to be in both OTA slots.

Staars, 2024-04-03:
I guess @Jason2866 was right with his assumption about overwriting OTA_0 with Tasmota on OTA_1 after the first step of the firmware conversion.
What I did for testing:

1. Have a fresh Shelly-FW with no failed update tries.
2. Use the Tasmota package the "usual" way until you have the first boot with correct WiFi credentials.
3. Then start the Berry console and launch this code snippet:
```
import partition_core
import flash
var p = partition_core.Partition()

print("Using brute force by copying ota_1 over ota_0, fingers crossed ... there is zero error handling for now!!")

var ota_0 = p.slots[2]
var ota_1 = p.slots[4]

for i:0..((ota_1.sz/0x1000)-1)
    var buf = flash.read(ota_1.start + (0x1000 * i), 0x1000)
    flash.write(ota_0.start + (0x1000 * i),buf)
end

print("Done! Please proceed with the usual AUTOCONF step.")
```
4. Then do the "usual" Autoconf.
5. Then migrate to Safeboot with the embedded Partition wizard.

Tested on a ESP32 dev board.

Jason2866, 2024-04-03:
@Staars Verified that your script which copies content from OTA_1 to OTA_0 slot does solve the issue.

s-hadinger, 2024-04-03:
We need to automate this and add first a check to know which ota slot is used. 

Can you please hold on. I'm still confident that we can fix the boot slot information after flashing the bootloader, and that we don't need to copy Tasmota to the other slot.

Jason2866, 2024-04-03:
@s-hadinger  Yes, we need to know in which OTA slot Tasmota is installed to copy it correctly. I am for the solution to copy Tasmota. If Shelly changes the OTA set active logic again we have a problem again. Overwriting Shelly firmware solves this permanently.

Jason2866, 2024-04-04:
Convert Berry scripts are updated. Issue should be fixed.

closed 2024-04-04 by Jason2866