tasmota/mgos32-to-tasmota32 #55: mgos32 not working on Shelly 1 mini Gen3

carpe07, 2024-03-29:
Hi,

I have bought 4 Shelly mini Gen3 at Amazon...
I was able to update 3 of them without any issue!? Now with the 4. I setup shelly, update to the latest firmware. 
After I flush "mgos32-to-tasmota32-Mini1G3.zip" nothing happen!? The led is on, but no tasmota nor shelly, no ping, nothing....
After some power circles shelly is booting!?!? 
So it looks like, that the tasmota flush was not working, but why? On the other 3 minis I done the same!?

Any ideas?

carpe07, 2024-03-29:
btw: the shelly FW is 1.2.2

Jason2866, 2024-04-04:
Try again. Issue should be fixed.

closed 2024-04-04 by Jason2866

Jason2866, 2024-04-05:
The fix is in the Berry migrate script which is executed by doing the "Autoconfig" step.

Jason2866, 2024-04-05:
@carpe07 It is soft bricked. Needs to be flashed via hardware serial.
This is the behaviour which is fixed now with latest change in the "Autoconfig" step.
A device which shows the behaviour is probably not possible to recover without hardware serial flashing. One last thing you can try is to download the Shelly update package.zip
and provide it local with drag and drop in the firmware update area. **Do this two times**
Now provide the Shelly convert package zip. If lucky the device has been recovered and the convert will run through.

carpe07, 2024-04-05:
the URL is for the 1Plus Gen 2 (https://fwcdn.shelly.cloud/**gen2/Plus1Mini**/9bab)

Jason2866, 2024-04-05:
gen3 mini models are all plus variants. Please post a screenshot from the Shelly Web page.
In upper left corner the intern Shelly device name is visible. Only this name is relevant.

carpe07, 2024-04-05:
I was ab´le to flush this version: https://updates.shelly.cloud/update/Mini1G3....I done this twice as intended, but after this two times flush i try to flush the "mgos32-to-tasmota32-Mini1G3.zip"...but again it hang after the flush!?
Thanks for your help, it looks like i need to flush this shelly with the serial adapter :-(

alexstarke, 2024-09-14:
I am stuck after loading mgos32-to-tasmota32-Mini1G3.zip, no tasmota AP appears. Did you manage to serial flash the Shelly Mini Gen3? If so, how? Thanks!

alexstarke, 2024-09-23:
> I am stuck after loading mgos32-to-tasmota32-Mini1G3.zip, no tasmota AP appears. Did you manage to serial flash the Shelly Mini Gen3? If so, how? Thanks!

I created a new [issue](https://github.com/tasmota/mgos32-to-tasmota32/issues/77)