tasmota/mgos32-to-tasmota32 #58: Cannot flash on Shelly PM Mini Gen3

hdfox, 2024-04-05:
The ota flash starts but then it is almost immidiately stops without any error except one in the console:

[image]

Jason2866, 2024-04-05:
Need to leave this open. Don't have a PM Mini Gen3 for doing tests.

Qowy, 2024-04-11:
Same except different error message
[image]

Qowy, 2024-04-11:
After updating the shelly to 1.2.2 the error changed: 
[image]

Jason2866, 2024-04-12:
Please try https://github.com/tasmota/mgos32-to-tasmota32/releases/download/v13.4.1/mgos32-to-tasmota32-MiniPMG3.zip
If this version does not work i am out of ideas. Just to be shure. You have downloaded the zip and drag and drop them in the firmware update area?

Qowy, 2024-04-12:
I am sorry at least on my part it is user error I retried with v13.4.0  mgos32-to-tasmota32-MiniPMG3.zip  and it worked.
Before I used mgos32-to-tasmota32-PlusPMMini.zip  I did not consider the other since there was no plus in the name.

So I sadly can't tell you if 13.4.1 works :D

Qowy, 2024-04-12:
Ok so this is not the end of it however.
Flashing the mentioned image worked.
I used autoconfigure but again for PlusPMMini (since I didn'd find MiniPMG3)
That "worked" it rebooted but only displayed the temperature sensor.
So autoconfigure again this time something with PMG3. But now the device is completely dead, no led no wifi ap no nothing.

I am usually not that inept with tasmota, but this hole mini G3 stuff is somewhat confusig :D

Jason2866, 2024-04-12:
Without the device i can't investigate.

Qowy, 2024-04-14:
[image]
I managed to get the device working again, by flashing it wia wire. Autoconfig also worked this time.

Anything I can dump you /log output from the device that would help you?

Jason2866, 2024-04-14:
@Qowy Full serial log when Shelly firmware is running (updated to latest version) and convert package.zip is applied may help to get an idea what wents wrong

Qowy, 2024-04-14:
@Jason2866 Can I revert to stock shelly? I can get the Image from https://[updates.shelly.cloud/update/MiniPMG3](https://updates.shelly.cloud/update/MiniPMG3) but can I flash that back? If so how? The web installer only takes a single factory image.

Jason2866, 2024-04-14:
@Qowy Not possible to build a full identic firmware which has been on the device from the update package. And the rebuild from the update package is a manual putting together of "the pieces".

Qowy, 2024-04-14:
Ok then I guess I will leave it be for now. Especially since your method worked for me, I just bricked the bootloader but I have a feeling that was my own fault, from flashing the wrong image /autoconfigure first). So I could never reproduce hdfox issue in the first place (apart from using the wrong image)

Jason2866, 2024-04-14:
@Qowy Thx for Trying!

Jason2866, 2024-04-14:
@hdfox Can you please try following. Download the Shelly update package for your device from here. https://updates.shelly.cloud/update/MiniPMG3
Provide the update file to the device by drag & drop in the update are. Start update.
If device does the update, wait until finished and the Shelly page is back. Do the update again. Yes a second time. When the device is back online it is fully recovered to Shelly firmware.
Now do provide the convert package in the drag&drop update area. Follow exactly the described steps.

hdfox, 2024-04-15:
Thank you so much. it worked, now I can use tasmota on the miniPMG3

closed 2024-04-15 by hdfox

jotwg, 2024-04-19:
@Qowy: some questions regarding flashing via serial:
- Does your photo show a PM mini Gen3 (no relay)?
- Do you have kind of pinout description/schematic of the shelly? (your photo is a bit blurry...)
- How did you activate the fw download mode? (just holding down the button during power on?)

Thx!

Qowy, 2024-04-19:
@jotwg It is a PM mini Gen3 without relay. However the pinout seems to be identical to other mini devices (with or without Gen3)
I used the pinout from here: https://devices.esphome.io/devices/Shelly-Plus-PM-Mini

Bodengriller, 2024-08-04:
@Qowy  Did you first flash [mgos32-to-tasmota32-MiniPMG3.zip](https://github.com/tasmota/mgos32-to-tasmota32/releases/download/v13.4.1/mgos32-to-tasmota32-MiniPMG3.zip) over Serial and dollowed on with auto-config and Partition Wizard
Or did you flash Tasmota directly by using [Tasmota-Web-Flasher](https://tasmota.github.io/install/) or something similar?

Qowy, 2024-08-04:
@Bodengriller As far as I remember the Serial flash was done using Web Flasher