tasmota/mgos32-to-tasmota32 #59: Shelly Pro Dimmer 2PM

justspacedog, 2024-04-16:
Hi,

please add a build for this device:
https://kb.shelly.cloud/knowledge-base/shelly-pro-dimmer-2pm

Thank you!

Jason2866, 2024-04-16:
Need the device name visible on the Shelly device Web page.
Please post a screenshot from.

Jason2866, 2024-04-24:
There is no update package available for this device from Shelly update server.
https://updates.shelly.cloud/update/ProDimmer2PM
To make a convert package I need infos from this package. As long there is no update is available no dice.

Jason2866, 2024-04-24:
There are flash pin headers. If you can dump the installed firmware (8MB), and provide me the dump I could try do too.

justspacedog, 2024-04-24:
That would be awesome! How can I do the dump?

Jason2866, 2024-04-24:
See https://docs.espressif.com/projects/esptool/en/latest/esp32/esptool/basic-commands.html#read-flash-contents-read-flash
You need hardware for. Recommended https://tasmota.github.io/docs/Getting-Started/

justspacedog, 2024-04-24:
I have a serial interface somewhere, but we just moved and I don’t know where I put it… :/

I activated debugging on the WebInterface and found out, that the update URL uses the Identifier „ProDimmerx“:
https://updates.shelly.cloud/update/ProDimmerx

[image]

So the stable Firmware should be found under:
https://fwcdn.shelly.cloud/gen2/ProDimmerx/d6d452c462e2e60e7106ff8996c0ac6184c0ba984d96a5490f5adf9874b6ea4e

It’s a 2,2 MB Zip Archive. Can you work with that or do you need the dump?

Jason2866, 2024-04-25:
If it is similar to #17 it is not supported from Tasmota

Jason2866, 2024-12-16:
Closing support not planned.

closed 2024-12-16 by Jason2866