tasmota/mgos32-to-tasmota32 #60: can't drag and drop

mokkin, 2024-04-16:
Unfortunately there is no field to drag & drop the new firmware image:

[image]

Am I looking at the wrong place?

Jason2866, 2024-04-17:
Do as in the workflow told. Update Shelly firmware to latest.

closed 2024-04-17 by Jason2866

mokkin, 2024-04-19:
Please reopen.
As you can see in the screenshot, I already had the latest firmware installed. I updated even to the latest BETA, but I still can't find any field for drag and drop:

[image]

Jason2866, 2024-04-19:
Which device is it? It is not visible in the screenshot nor you mentioned it.
All Shelly esp32 driven devices supported by this tool do show the intern Shelly name on the main page.

mokkin, 2024-04-19:
It's a Shelly Plug S.

mokkin, 2024-04-19:
sorry, I have no plus...

Jason2866, 2024-04-19:
Look here https://github.com/arendst/mgos-to-tasmota