tasmota/mgos32-to-tasmota32 #63: Stuck with the Partition Wizard step

Dilergore, 2024-06-19:
Hi,

I have successfully upgraded multiple Shelly Plus 2PMs following the same steps. That said, I have this single device where I am having issues, it seems I am stuck and cannot proceed after the partition wizard:
[image]

Any idea how I could resolve this?

Jason2866, 2024-06-19:
Since the tool can't find the app sizes, probably only a serial flash can recover the device.
Looks like the partition info has been corrupted.

Dilergore, 2024-06-19:
Thanks @Jason2866!

So after FS increase, the Partition Manager disappeared. When I checked the auto-config it was showing like i never applied it, so I reapplied it. Once I reapplied the auto-config, the partition manager was there again, and it was not showing the ERROR for Step 1, so I was able to run it with success and everything is working and I am on the latest version.

closed 2024-06-19 by Dilergore