tasmota/mgos32-to-tasmota32 #65: request support for shelly plug plus S V2

Roksvaag, 2024-06-24:
Hi there,
is it possible to add support for this device? Unless i'm mistaken it's not supported yet
link to Shelly KB: https://kb.shelly.cloud/knowledge-base/shelly-plus-plug-s-v2 
Screenshot of device info
[image]

Jason2866, 2024-06-24:
What happens with this one https://github.com/tasmota/mgos32-to-tasmota32/releases/download/v13.4.2/mgos32-to-tasmota32-PlusPlugS.zip ?

Roksvaag, 2024-06-24:
As there's no option to "drag 'n drop" the zip into the UI, I've tried just using the link. It loads for about 15 sec, then I get an error message saying "Your Shelly has failed to update". please refer to attached image. 

[image]

Jason2866, 2024-06-24:
The Shelly firmware is not the actual one. Download from here https://updates.shelly.cloud/update/PlusPlugS and provide this original Shelly update package.

Th0maz, 2024-06-24:
I had the same problem and solved it by hosting the update package locally using https://github.com/TheWaWaR/simple-http-server and put the link from my local network into the custom firmware field.

After I read something about "[Devices with firmaware v1.10+ cannot be flashed OTA anymore](https://templates.blakadder.com/shelly_plus_2PM.html)" I thought it would not be a good idea to let the Shelly device update itself to the latest firmware version.

Roksvaag, 2024-06-24:
Thank you guys, I'll look into it. 
 @Th0maz , did you use the update package linked by Jason? Are you also using the V2-edition of the plug plus s?

Th0maz, 2024-06-24:
I used different devices (I think Shelly Plus 2PM or Shelly Plus PM Mini). And since there was no drag-and-drop area, I used the local web server to "inject" the matching firmware update package.

Jason2866, 2024-07-10:
Closing since no feedback. Flashing the Shelly Plus Plug S is supported.

closed 2024-07-10 by Jason2866