tasmota/mgos32-to-tasmota32 #67: after update and restart the device fallback to shelly firmware

Martin4us3, 2024-07-29:
Hallo,
today I updated Shelly Plus UNI to Tasmota firmware.

After update I can see the Tasmota frontend.

If I restarting now, the system fallback to shelly firmware?!?

Migration to safeboot partition has no effect?

What's wrong?

Jason2866, 2024-07-29:
Needs serial flash to recover. Two things may be the reason for
- Something did not work following the described workflow to convert 
- Workflow not followed exactly as described

Jason2866, 2024-08-22:
New version is online in release section supporting Shelly fw >= v.14.x

Jason2866, 2024-08-26:
Issue was caused by using old convert ZIPs with new Shelly fw >= 1.4.x
The Converter is updated to support Shelly fw >= 1.4.x

A Shelly device which is/was tried to convert with the previous convert packages is soft bricked. Needs recover with serial flash. Sorry for this. Shelly changed Partition layout starting with fw 1.4.0

closed 2024-08-26 by Jason2866