tasmota/mgos32-to-tasmota32 #68: Cant flash on shelly 1pm plus with fw 1.4.0

breakingflower, 2024-08-02:
Hi

Following the guide, I have updated a shelly 1 plus pm to latest shelly firmware
[image]
Tried flashing with latest zip from releases. Getting error 

```
  {
    "ts": 1722589187.248,
    "level": 0,
    "data": "shelly_ota.cpp:261      Aborting update: backend error while processing data: -1 (-11: PT update is req'd but unsupported by the new fw)",
    "fd": 1
  },
```
Full logs
[diagnostics-shelly-plus1pm-debug-log.txt](https://github.com/user-attachments/files/16468324/diagnostics-shelly-plus1pm-debug-log.txt)

I've tried the following:
- downgrade (does not work, variety of errors with OTA and zip)
- reset to factory and try again
- OTA

I could not find any related articles, any advice is welcome.

Jason2866, 2024-08-02:
New firmware dated from end of July. Looks, like Shelly changed "something" again.

cicatrix87, 2024-08-21:
So for now its better to not update the Shellys FW and just make the conversion as long as its not FW 1.4.0?

Jason2866, 2024-08-21:
Yes, do not update to fw v1.4.x

fmarzocca, 2024-08-22:
I have just received a Mini with FW 1.4.2....
Is there any solution for this? I can't upload Tasmota

Jason2866, 2024-08-22:
Know what needs to be changed. Updated convert ZIPs the next days...

fmarzocca, 2024-08-22:
Thanks! Will you post here when done?

Jason2866, 2024-08-22:
Yes, i will leave a note in the regarding open issues.

Jason2866, 2024-08-22:
New version is online in release section supporting Shelly fw >= v.14.x

fmarzocca, 2024-08-22:
> New version is online in release section supporting Shelly fw >= v.14.x

1.14.x? Wasn't it 1.4.x?

Jason2866, 2024-08-22:
yes. 1.4.x

fmarzocca, 2024-08-23:
Just done on my Shelly Mini

1- upgraded Shelly firmware to 1.4.2
2 - downloaded conversion file from your site
3 - uploaded to Shelly
4 - The Shelly web interface is no more responding and no "Tasmota" wifi AP is shown.

Is my new shelly device dead now?

Jason2866, 2024-08-23:
I have checked the convert zip for the Shelly Mini1G3. It is correct. How long have you waited? If you see no Tasmota AP after 10 minutes, disconnect device from AC mains. Wait 10 seconds and connect the device again. You can do this one time only!!! After the next attempt the Shelly locked bootloader will block the device. So do not disconnect the device if you are not 1000% sure there is really really no Tasmota AP!!!
My tests with the new version where all successful.

fmarzocca, 2024-08-23:
No way. I waited 30 minutes. Then disconnected and reconnected the device but I have no new AP.
Isn't there a way to factory reset it?

Jason2866, 2024-08-23:
No, the only way to recover is via serial flash

fmarzocca, 2024-08-23:
Where can I find the pin assignments for serial flashing it?

Jason2866, 2024-08-23:
The factory image via Tasmota WebInstaller

rp2019433, 2024-08-24:
Hi.
Just tried to convert Shelly Mini1 G3 to Tasmota using firmware 20240819-074315/1.4.2-gc2639da and was not successfull. Shelly got stuck and it's not possible to access it anymore.
Have done this procedure successfully several times in the past Shelly firmware.

Jason2866, 2024-08-24:
Mhh, triple checked the package. Do not see anything wrong. Can't check this build variant with a real device since not having a Mini1 G3.  Looks like something is wrong. Honestly idk what. Would need a Mini1 G3 for testing.

rp2019433, 2024-08-24:
HI @Jason2866. Thank you for checking again. If there is something I can do for help, please let me know - I mean, get any log or information. Just let me know how to do it.

fmarzocca, 2024-08-24:
@rp2019433 , I had the same experience and I had to throw away my Mini1 G3. I tried to serial-flash it, but the printed circuits and the pads are too small to avoid messing everything up.

I think that a temporary solution could be *NOT* to upgrade the Shelly FW if you have one with FW<1.4.x and install previous release of conversion file.

rp2019433, 2024-08-24:
> @rp2019433 , I had the same experience and I had to throw away my Mini1 G3. I tried to serial-flash it, but the printed circuits and the pads are too small to not mess everything up.
> 
> I think that a temporary solution could be _NOT_ to upgrade the Shelly FW if you have one with FW<1.4.x and install previous release of conversion file.

Thank you for sharing. Yes, I would recommend not to convert Shelly to Tasmota if firmware is 1.4 at this moment until it's confirmed the process is working properly with this device.

fmarzocca, 2024-08-26:
I managed (with great effort) to arrange wirings for serial flashing to my dead Mini1 G3. I have falshed tasmota32.bin, but this is the resulting log (in Web Installer):
```
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x7 (TG0WDT_SYS_RST),boot:0xd (SPI_FAST_FLASH_BOOT)
Saved PC:0x40047ed2
Invalid chip id. Expected 5 read 0. Bootloader for wrong chip?
ets_main.c 333 
```
Could it be so damaged that it cannot be flashed by serial?

Jason2866, 2024-08-26:
Convert is working. See successful attempt #72

closed 2024-08-26 by Jason2866

fmarzocca, 2024-08-26:
I broke 2 of those and I am experienced with this procedure, so I am not using the wrong file.
Anyway I have succesfully flashed via serial.

alexstarke, 2024-09-23:
> I broke 2 of those and I am experienced with this procedure, so I am not using the wrong file. [#72](https://github.com/tasmota/mgos32-to-tasmota32/issues/72) refers to 1PMMini, not 1Mini. Maybe the "PM" variant is different. Anyway I have succesfully flashed via serial.

I believe I am facing the same issue, cf. [#77](https://github.com/tasmota/mgos32-to-tasmota32/issues/77). How did you manage to serial flash the Mini Gen3? (note: it is not about the PM version)