tasmota/mgos32-to-tasmota32 #69: Failed to boot after flashed Shelly Plus 1 Mini

axyzs, 2024-08-03:
Hello,
Thanks for the awesome project! 

I have an unbootable Shelly Plus 1 Mini after flashing it using v13.4.1. 
I followed the instructions 
- updated to the latest stable firmware with WiFi connected. 
- Then, I used v13.4.1 mgos32-to-tasmota32-Mini1G3.zip.
I waited about 2 minutes, the LED stopped blinking, and I couldn't find the Tasmota AP and I waited another 5 minutes, but still no Tasmota AP.

I unplugged the device and plugged it in again, but still no Tasmota AP and no LED indicator on.

I found this article https://devices.esphome.io/devices/Shelly-Plus-1-Mini and tried to flash it via the serial method, but I couldn't get the device detected. I used a multimeter to check each connection point to the device, and every connection was fine.

So, I'd like to understand if this means the device is totally bricked and unrecoverable? 
Thanks in advance

Jason2866, 2024-08-03:
Idk what happened. Even when there was an OTA update from Shelly securing the device and burning efuses (i think this would be not legal, without asking before) the device is still discovered via USB serial adapter and esptool.py 
So if you are 100% sure your hardware setup is correct and working, the device is dead.
I highly doubt this. Probably "something" is not correctly setup/working.

axyzs, 2024-08-03:
Hi @Jason2866,

Many thanks for your reply. 
After retrying for almost 2 hours since the pins are too small to be connected properly, I managed to flash ESPHome successfully. I tried flashing Tasmota but have no idea why it couldn't be flashed, always had different errors. Then I 
tried ESPHome web installer, and it worked on the first try. 
I'm glad the device is finally functional.

axyzs, 2024-08-03:
Hi @Jason2866 
Could you please suggest the correct way to migrate from ESPhome to Tasmota (32c)? I don't want to serial flash again because those tiny pins are quite painful. I tried OTA from ESPhome using `tasmota32c3.bin`, but I received an error stating that there wasn't enough space. Then I tried `tasmota32c3-safeboot.bin`, but it didn't work either and reverted back to ESPhome.

I suspect there might be a partition issue. Should I first flash the minimal firmware (http://ota.tasmota.com/tasmota32/release-9.2.0/tasmota32-minimal.bin) and then upgrade to the latest `tasmota32c3.bin`?

Jason2866, 2024-08-03:
Sorry, idk anything about esphome. Can't tell if there is a OTA migration path to Tasmota.

axyzs, 2024-08-04:
Thanks for your reply. 
Just spend some more time, I managed flashing `tasmota32c3.factory.bin` successfully by using serial method.
I will close this issue.

closed 2024-08-04 by axyzs