tasmota/mgos32-to-tasmota32 #70: Shelly pro 4pm dead after "Start Migration"

Chellux, 2024-08-04:
hi, 

my pro 4pm died after "Start Migration"

it opens the wifi, but dont starts a webserver to config... 

is there any way to recover it? can i programm it via FTDI?

or is it now a piece of waste (with wifi :/ )

Jason2866, 2024-08-04:
Recovering via serial is possible as long there are no secure fuses burnt forcing signed firmware. This is the case for devices which does have native support from Shelly for Matter.
Afaik this device is not available with Matter support, so flashing Tasmota via serial is possible and Tasmota will start.

Chellux, 2024-08-04:
But there are no normal headers like on the shelly 1pm ? 🤔

Jason2866, 2024-08-26:
If the device had already a firmware fw version >= 1.4.0 it is soft bricked.
The recent released new convert packages can OTA this new Shelly firmware release.
Serial flash is possible for the Shelly pro 4pm too. 
There is a pin header inside for flashing. I found no Internet side where a picture shows.
Best I found is this site https://karlquinsland.com/shelly-pro-4pm-teardown/

closed 2024-08-26 by Jason2866

TD-er, 2024-08-26:
Just make sure to not break the flat cable to the display.
It is a real pain in the *ss to open this Shelly pro 4pm