tasmota/mgos32-to-tasmota32 #74: Failed to flash Shelly 1PM Mini Gen3

wildekek, 2024-08-27:
Thanks for the effort getting Gen3 devices supported! I'm running into an issue though.

Followed the instructions:
- Updated to Shelly FW 4.2
- Downloaded mgos32-to-tasmota32-Mini1PMG3.zip
- Selected above file in Shelly web ui and click update
- Waited for update to finish, then WebUI becomes unavailable
- No "tasmota-*" Access Point appears and I'm stuck :(

I've waited a good 10 minutes, opened up WiFi scanner, but I see no access point appearing. After that I rebooted the device (yes, I know this might brick it, but it was unavailable already, so the only option left imo), still no luck. I'll try to hook up and FTDI adapter and flash it hardwired.

I really paid attention to the instructions, not sure if this is an issue with the binary or that I fail to understand the steps.

Jason2866, 2024-08-27:
Weird. Honestly idk why the Mini Gen 3 is making issues to OTA convert.
Checked again the packages and see nothing wrong. But "something" seems to be not as it should be :-(  There are success messages and fail feedback.
Will add a warning in the readme

wildekek, 2024-08-27:
I've hooked up an FTDI adapter and flashed an ESPHome image successfully. Unfortunately the device seems to be stuck in a boot loop:
```
[22:50:43]ets_loader.c 78 
[22:50:44]ESP-ROM:esp32c3-api1-20210207
[22:50:44]Build:Feb  7 2021
[22:50:44]rst:0x10 (RTCWDT_RTC_RST),boot:0xd (SPI_FAST_FLASH_BOOT)
[22:50:44]SPIWP:0xee
```

wildekek, 2024-08-27:
Am I correct that by flashing the ESP via the ESPHome web interface -> USB -> UART should write the bootloader too? Or is this a fuse issue now?

wildekek, 2024-08-27:
After erasing and flashing mgos32-to-tasmota32-Mini1PMG3.zip via UART:

```
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x1 (POWERON),boot:0xd (SPI_FAST_FLASH_BOOT)
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
```

Jason2866, 2024-08-28:
The OTA convert zip is not a firmware which can be flashed.
Use Tasmota Installer https://tasmota.github.io/install/ to flash Tasmota. You just have to select "Tasmota" and hit install. That's all.

wildekek, 2024-08-30:
Thanks Jason! I used the tasmota binary and that worked. My shelly got in a bootloop since it had a brownout on start, but i fixed that by using an external power supply instead of the ftdi adapter.

Jason2866, 2024-08-30:
Great you managed to get the device working with Tasmota.
One question how was the Shelly device powered when you try to do the OTA convert process?

wildekek, 2024-08-30:
Powered via 230v inputs on the shelly.

wildekek, 2024-08-30:
I have received a new one, I could try and help debug this by trying again and investigating when it borks again.

Jason2866, 2024-08-30:
@wildekek Yes, you can help. Make a full backup of the firmware via esptool.py 
The flash is 8MB! After you have done please connect a terminal programm which can save the output to a file. Power the device with a reliable 3.3V power supply.
Update to latest Shelly fw. After that please start logging. Now provide the OTA package.
Try to do all described steps until you can't do anything. Please post this log file. Maybe "something" can be seen

wenningerk, 2024-08-31:
Same thing just happened to me with a 1 mini Gen 3. I've read about similar issues in other threads. Some experience it just after a week or so. Can it be that the power-supply over  230v is on the edge. Is it possible that the chip can be run at differenct clocks as per configuration or that the wifi power can be controlled? I had that on one of these cloud-based serial connectors used for solar-inverters once I had my own software flashed. Could as well just be the startup sequence.
If the above theory is true I probably won't even have to hook up the whole serial adapter. 3.3v supply might be sufficient unless of course the conversion firmware didn't die early enough and had enough time to brick something.

wenningerk, 2024-08-31:
ok - more juice wasn't the answer. The shelly-bootloader was just complaining about an invalid image. So I soldered all the connections and after flashing it is looking fine. No 12V relay when driven from just 3.3v. Comes up on 230v as well. But still no relay :-( Maybe GPIO5 is wrong? Anyway button and led are at least working without additional 3.3v.

Btw that is the pattern I saw after the OTA attempt:
```ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x3 (RTC_SW_SYS_RST),boot:0xd (SPI_FAST_FLASH_BOOT)
Saved PC:0x403cfc5a
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd5810,len:0x16dc
load:0x403cc710,len:0x9cc
load:0x403ce710,len:0x2f04
entry 0x403cc8cc
I (35) boot: Shelly OS loader 0.8.2 (Dec 19 2023 13:39:56 IDF 4.4.4-a7)
I (35) boot: chip revision: v0.4
I (35) boot_comm: chip revision: 4, min. bootloader chip revision: 3
I (42) boot.esp32c3: SPI Speed      : 80MHz
I (47) boot.esp32c3: SPI Mode       : DIO
I (51) boot.esp32c3: SPI Flash Size : 4MB
I (56) boot: Enabling RNG early entropy source...
E (62) flash_parts: partition 4 invalid - offset 0x3a0000 size 0x280000 exceeds flash chip size 0x400000
E (72) boot: Failed to verify partition table
E (77) boot: load partition table error!```

wenningerk, 2024-08-31:
Aah autoconfiguration does the trick ;-)

Jason2866, 2024-08-31:
Autoconfiguration does way more than config the device with Shelly devices.
It is mandatory since it replaces the locked bootloader from Shelly. Without doing the step the device gets 100% bricked!

wenningerk, 2024-08-31:
I got that but with the first device that was already bricked on the first step I couldn't start the Autoconfiguration.
And after having erased and flashed the bootstrap image I already had the bootloader switched and thus had skipped this step.
Btw. meanwhile I've tried the exact same procedure (according to the instructions in the readme with Autoconfiguration of course ;-) ) on a 2nd device. Only difference was that I didn't upgrade to the newest Shelly version (iirc 1.4.2) first. And it went smoothly. Unfortunately I of course didn't spend the effort of soldering the wires first so I can't tell which Shelly OS loader it had.

matzman666, 2024-09-16:
While updating a Shelly 1PM Gen3 via OTA I was getting the same error as posted above: 

> E (62) flash_parts: partition 4 invalid - offset 0x3a0000 size 0x280000 exceeds flash chip size 0x400000

What confused me that the error message said chip size is 0x400000 but Shelly Gen3 devices are using a modified ESP32-C3 that features 8MB of flash. So I investigated further and found an incorrectly specified flash size in bootloader-tasmota-c3.bin. Byte 3 in this file indicates the flash size among other things and it incorrectly said 4 MB of flash. After modifying this byte to indicate 8 MB of flash OTA was working successfully.

Jason2866, 2024-09-16:
Not updating Shelly firmware to latest is not a good idea. The partition layout and sizes have changed. Latest Tasmota convert expects this new layout. When doing OTA convert with old scheme there is a 50% chance it fails. It will only work when update is done with partition at address 0x10000. If the other one is the one to update it will fail and brick the device.
The size setting for flash in the bootloader is not relevant, since not used.

fmarzocca, 2024-09-16:
> It will only work when update is done

Sometimes.... 
Not in my 2 cases.

rp2019433, 2024-09-18:
> > It will only work when update is done
> 
> Sometimes.... Not in my 2 cases.

Yes, same here - as described on issue #68.

wenningerk, 2024-09-18:
> It will only work when update is done with partition at address 0x10000. 
Guess that is the case before any update. So the trick is then maybe not to upgrade any shelly firmware and ota flash tasmota right away. This btw. just worked again. So my current count is then 3x success and 1x failure (needed to serial-flash).

nemesis03, 2024-09-29:
I also tried to convert my Shelly PM Mini Gen3 and it resulted in the following error:
`(62) flash_parts: partition 4 invalid - offset 0x3a0000 size 0x280000 exceeds flash chip size 0x400000
(72) boot: Failed to verify partition table
(77) boot: load partition table error!`

The error is the same regardsless whether app_0 or app_1 was the active partition when doing the OTA upgrade. (I switched the active partition by doing another firmware upgrade with 1.4.2). I do not see an error in the 

Here the log for when app_0 was active:
`shos_rpc_inst.c:242     ota.start via WS_in 192.168.33.91:54788
shelly_ota.cpp:248      Starting update (0x3fcc7a14), to=600 cd=0/ct=0 ignore_same=0
shos_rpc_inst.c:242     ota.write via WS_in 192.168.33.91:54788
shelly_ota.cpp:375      manifest.json: 1776 bytes (hdr 71 desc 0 CRC 25b03f19)
shelly_ota.cpp:554      fw: MiniPMG3 ver 12.5.0 20240822-170213/tasmota-12.5.0
shos_rpc_inst.c:242     ota.write via WS_in 192.168.33.91:54788
shelly_ota.cpp:182      OTA status in_progress, 0 %
shelly_update.cpp:332   incoming fw signatures: 00
shelly_ota_esp32_ba:512 Will write to slot 1
shelly_notification:208 Event from sys: {"component":"sys", "event":"ota_begin", "msg":"in_progress", "ts":1727602907.36}
shos_rpc_inst.c:242     ota.write via WS_in 192.168.33.91:54788
shelly_ota.cpp:375      bootloader-c3.bin: 20480 bytes (hdr 75 desc 0 CRC f18a07bc)
shos_rpc_inst.c:242     ota.write via WS_in 192.168.33.91:54788
[...]
shelly_ota_esp32_ba:709 Boot: cur 010000ff, new 000802ff, min 00000000, update? 1
shos_rpc_inst.c:242     ota.write via WS_in 192.168.33.91:54788
shelly_ota.cpp:375      C3-8MB-partition-table.bin: 4096 bytes (hdr 84 desc 0 CRC 2207497b)
shos_rpc_inst.c:242     ota.write via WS_in 192.168.33.91:54788
shos_rpc_inst.c:242     ota.write via WS_in 192.168.33.91:54788
shelly_ota_esp32_ba:733 PT is being moved from 0xf000 to 0x8000
shelly_ota_esp32_ba:302 Installing PT 4096@fs_1(0x620000) -> pt(0x8000)...
shos_rpc_inst.c:242     ota.write via WS_in 192.168.33.91:54788
shelly_ota.cpp:375      otadata.bin: 8192 bytes (hdr 69 desc 0 CRC 40773c40)
shos_rpc_inst.c:242     ota.write via WS_in 192.168.33.91:54788
[...]
shelly_ota.cpp:375      tasmota32c3.bin: 1605040 bytes (hdr 73 desc 0 CRC 3a6d211f)
shelly_ota_esp32_ba:302 Installing BL 20480@app_1(0x3a0000) -> boot(0)...
shos_init.c:94          New min heap free: 83304
shos_rpc_inst.c:242     ota.write via WS_in 192.168.33.91:54788
[...]
shelly_ota.cpp:375      fs-1024.img: 1048576 bytes (hdr 69 desc 0 CRC c6351727)
shos_rpc_inst.c:242     ota.write via WS_in 192.168.33.91:54788
[...]
shelly_ota.cpp:182      OTA status in_progress, 100 %
ota                     Write BS1: seq 23 bst 0x11 seq2 23 | as 1 rs 0 ba 3 mfs 1 c 0 | v 1
ota                     Write BS0: seq 23 bst 0x11 seq2 23 | as 1 rs 0 ba 3 mfs 1 c 0 | v 1
shelly_ota_esp32_ba:115 Set NVS boot state to 0x310 'LFS' '{"bs":4096,"is":128}' (from 0x0 'LFS' '{"bs":4096,"is":128}')
shelly_ota.cpp:771      Update succeeded, will reboot, uba=3, cd=0/ct=0 s
shos_system.c:60        Rebooting in 3000 ms
shelly_notification:208 Event from sys: {"component":"sys", "event":"scheduled_restart", "time_ms": 2995, "ts":1727602968.17}
shelly_ota.cpp:182      OTA status success, 100 %
shelly_update.cpp:416   FW signatures: want 07 got 00
efuse                   Bits are not empty. Write operation is forbidden.
shelly_update.cpp:443   EFUSE write error bit[0]: 5636
efuse                   Bits are not empty. Write operation is forbidden.
shelly_update.cpp:443   EFUSE write error bit[1]: 5636
shelly_ota_source_rp:25 leftover RPC source 0x3fcc8de4
shelly_notification:208 Event from sys: {"component":"sys","event":"config_changed","restart_required":false,"ts":1727602968.41,"cfg_rev":9}
shelly_notification:162 Status change of sys: {"cfg_rev":9}
shelly_notification:208 Event from sys: {"component":"sys", "event":"ota_success", "msg":"success", "ts":1727602968.41}
wifi                    I (125243) wifi:state: run -> init (0)
wifi                    I (125243) wifi:pm stop, total sleep time: 0 us / 122950832 us
wifi                    I (125253) wifi:<ba-del>idx:0, tid:0
wifi                    I (125263) wifi:<ba-del>idx:1, tid:1
wifi                    I (125263) wifi:new:<1,0>, old:<1,1>, ap:<1,1>, sta:<1,1>, prof:2
shos_wifi.c:60          WiFi STA: Disconnected, reason: 8
shos_lwip.cpp:171       ws(0x3fcbf810): Link down
shos_lwip.cpp:110       Default: ws(0x3fcbf810) -> (null) ()
wifi                    W (125293) wifi:Haven't to connect to a suitable AP now!
wifi                    W (125303) wifi:Haven't to connect to a suitable AP now!
shos_vfs.c:1072         Unmounting filesystems
shos_lwip.cpp:175       wa(0x3fcbce70): Status down
[... 126 lines cut ...]
boot: ## Label            Usage          Type ST Offset   Length   F
boot:  0 otadata          OTA data         01 00 00010000 00002000 1
boot:  1 nvs              WiFi data        01 02 00014000 0000c000 0
boot:  2 app_0            OTA app          00 10 00020000 00280000 1
boot:  3 fs_0             FS               01 82 002a0000 00100000 1
boot:  4 app_1            OTA app          00 11 003a0000 00280000 1
boot:  5 fs_1             FS               01 82 00620000 00100000 1
boot:  6 scratch          Unknown data     01 80 00720000 00080000 1
boot:  7 shelly           Unknown data     01 88 007f0000 00010000 1
boot: End of partition table`

partition table 1.4.2:
`boot: Partition Table @ 0xf000 (93d694b5):
[... cut ...]
boot:  3 fs_0             FS               01 82 002a0000 00100000 1
boot:  4 app_1            OTA app          00 11 003a0000 00280000 1
boot:  5 fs_1             FS               01 82 00620000 00100000 1
boot:  6 scratch          Unknown data     01 80 00720000 00080000 1
boot:  7 shelly           Factory data     01 88 007f0000 00010000 1
boot: End of partition table`

I still have my Shelly PM Mini Gen3 hooked up for serial flashing and serial console. Is there anything that I should test?

Jason2866, 2024-09-29:
Yes, the partition table changed with newer firmware. That's why it is mandatory to update Shelly fw to latest version. The OTA convert zip expects the new scheme.
Why it works for some to convert and for others it does not i have no idea.

Jason2866, 2024-09-29:
@nemesis03 Can you please verify if the address of the partition table has really moved from 0x8000 to 0xf000. Did not see this when i updated the the script to fw 1.4.x
Maybe Shelly changed this after my investigation.
If this is the case that's the reason for failing OTA convert!

nemesis03, 2024-09-29:
Well the partition table was copied from the serial console while booting the Shelly firmware.
Here for firmware 1.3.3:
```
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x3 (RTC_SW_SYS_RST),boot:0xd (SPI_FAST_FLASH_BOOT)
Saved PC:0x40382970
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd5810,len:0x1724
load:0x403cc710,len:0x15a4
load:0x403ce710,len:0x2a20
entry 0x403cc8cc
␛[0;32mI (35) boot: Shelly OS loader 1.0.0 (Jun 25 2024 12:26:04 IDF 4.4.4-a9)␛[0m
␛[0;32mI (35) boot: chip revision: v0.4␛[0m
␛[0;32mI (35) boot_comm: chip revision: 4, min. bootloader chip revision: 3␛[0m
␛[0;32mI (42) boot.esp32c3: SPI Speed      : 80MHz␛[0m
␛[0;32mI (47) boot.esp32c3: SPI Mode       : DIO␛[0m
␛[0;32mI (52) boot.esp32c3: SPI Flash Size : 8MB␛[0m
␛[0;32mI (56) boot: Enabling RNG early entropy source...␛[0m
␛[0;32mI (62) boot: Partition Table @ 0x8000 (2077e0d9):␛[0m
␛[0;32mI (67) boot: ## Label            Usage          Type ST Offset   Length   F␛[0m
␛[0;32mI (75) boot:  0 otadata          OTA data         01 00 00010000 00002000 1␛[0m
␛[0;32mI (82) boot:  1 nvs              WiFi data        01 02 00014000 0000c000 0␛[0m
␛[0;32mI (90) boot:  2 app_0            OTA app          00 10 00020000 00280000 1␛[0m
␛[0;32mI (98) boot:  3 fs_0             FS               01 82 002a0000 00100000 1␛[0m
␛[0;32mI (105) boot:  4 app_1            OTA app          00 11 003a0000 00280000 1␛[0m
␛[0;32mI (113) boot:  5 fs_1             FS               01 82 00620000 00100000 1␛[0m
␛[0;32mI (121) boot:  6 scratch          Unknown data     01 80 00720000 00080000 1␛[0m
␛[0;32mI (128) boot:  7 shelly           Unknown data     01 88 007f0000 00010000 1␛[0m
␛[0;32mI (136) boot: End of partition table␛[0m
␛[0;32mI (141) ota: Read BS0: seq 8 bst 0x10 seq2 8 | as 0 rs 1 ba 0 mfs 0 c 1 | v 1␛[0m
␛[0;32mI (148) ota: Read BS1: seq 8 bst 0x10 seq2 8 | as 0 rs 1 ba 0 mfs 0 c 1 | v 1␛[0m
␛[0;32mI (156) ota: Booting app 0␛[0m
␛[0;32mI (159) boot_comm: chip revision: 4, min. application chip revision: 3␛[0m
␛[0;32mI (167) esp_image: segment 0: paddr=00020020 vaddr=3c150020 size=34600h (214528) map␛[0m
␛[0;32mI (209) esp_image: segment 1: paddr=00054628 vaddr=3fc94600 size=02e58h ( 11864) load␛[0m
␛[0;32mI (212) esp_image: segment 2: paddr=00057488 vaddr=40380000 size=08b90h ( 35728) load␛[0m
␛[0;32mI (222) esp_image: segment 3: paddr=00060020 vaddr=42000020 size=14b8c8h (1358024) map␛[0m
␛[0;32mI (440) esp_image: segment 4: paddr=001ab8f0 vaddr=40388b90 size=0b988h ( 47496) load␛[0m
␛[0;32mI (449) esp_image: segment 5: paddr=001b7280 vaddr=50000010 size=00004h (     4) load␛[0m
␛[0;32mI (456) boot: Loaded app from partition at offset 0x20000␛[0m
␛[0;32mI (456) boot: Disabling RNG early entropy source...␛[0m
␛[0;32mI (469) cpu_start: Pro cpu up.␛[0m
␛[0;32mI (478) cpu_start: Pro cpu start user code␛[0m
␛[0;32mI (478) cpu_start: cpu freq: 160000000␛[0m
␛[0;32mI (478) cpu_start: Application information:␛[0m
␛[0;32mI (481) cpu_start: Project name:     MiniPMG3␛[0m
␛[0;32mI (486) cpu_start: App version:      1.3.3␛[0m
␛[0;32mI (490) cpu_start: Compile time:     Jun 25 2024 12:26:24␛[0m
␛[0;32mI (497) cpu_start: ELF file SHA256:  89bf582ccd5654c8...␛[0m
␛[0;32mI (503) cpu_start: ESP-IDF:          4.4.4-a9␛[0m
␛[0;32mI (508) heap_init: Initializing. RAM available for dynamic allocation:␛[0m
␛[0;32mI (515) heap_init: At 3FC9FF70 len 0003C7A0 (241 KiB): DRAM␛[0m
␛[0;32mI (521) heap_init: At 3FCDC710 len 00002950 (10 KiB): STACK/DRAM␛[0m
␛[0;32mI (528) heap_init: At 500000A8 len 00001F58 (7 KiB): RTCRAM␛[0m
␛[0;32mI (534) spi_flash: detected chip: gd␛[0m
␛[0;32mI (538) spi_flash: flash io: qio␛[0m
␛[0;32mI (553) sleep: Configure to isolate all GPIO pins in sleep state␛[0m
␛[0;32mI (554) cpu_start: Starting scheduler.␛[0m
```

And here for firmware 1.4.2:
```
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0xc (RTC_SW_CPU_RST),boot:0xd (SPI_FAST_FLASH_BOOT)
Saved PC:0x4038069c
SPIWP:0xee
mode:DIO, clock div:1
[... 31 lines cut ...]
boot: Disabling RNG early entropy source...
cpu_start: Unicore app
cpu_start: Pro cpu start user code
cpu_start: cpu freq: 160000000 Hz
cpu_start: Application information:
cpu_start: Project name:     MiniPMG3
cpu_start: App version:
cpu_start: Compile time:     Aug 19 2024 07:43:15
cpu_start: ELF file SHA256:  24998333a...
cpu_start: ESP-IDF:          5.2.1-s2
cpu_start: Min chip rev:     v0.3
cpu_start: Max chip rev:     v1.99
cpu_start: Chip rev:         v0.4
heap_init: Initializing. RAM available for dynamic allocation:
heap_init: At 3FC9FFA0 len 00020060 (128 KiB): RAM
heap_init: At 3FCC0000 len 0001C710 (113 KiB): Retention RAM
heap_init: At 3FCDC710 len 00002950 (10 KiB): Retention RAM
heap_init: At 50000088 len 00001F60 (7 KiB): RTCRAM
spi_flash: detected chip: gd
spi_flash: flash io: qio
sleep: Configure to isolate all GPIO pins in sleep state
sleep: Enable automatic switching of GPIO sleep configuration
```

Ist there any other way that I can double Check?

Jason2866, 2024-09-30:
@nemesis03 Please try https://github.com/tasmota/mgos32-to-tasmota32/releases/tag/v14.1.0 The partition table address has indeed been changed from 0x8000 to 0xF000 for gen3 min devices (with latest firmware). Verified by investigating the orig. Shelly update zip package.
Thx for the hint!

nemesis03, 2024-09-30:
I have just successfully tested the new version.

```
shos_rpc_inst.c:242     ota.start via WS_in 192.168.33.91:59241
shelly_ota.cpp:248      Starting update (0x3fcc8b68), to=600 cd=0/ct=0 ignore_same=0
shos_rpc_inst.c:242     ota.write via WS_in 192.168.33.91:59241
shelly_ota.cpp:375      manifest.json: 1776 bytes (hdr 71 desc 0 CRC ca68bfd4)
shelly_ota.cpp:554      fw: MiniPMG3 ver 12.5.0 20240930-100201/tasmota-12.5.0
shos_rpc_inst.c:242     ota.write via WS_in 192.168.33.91:59241
shelly_ota.cpp:182      OTA status in_progress, 0 %
shelly_update.cpp:332   incoming fw signatures: 00
shelly_ota_esp32_ba:512 Will write to slot 1
shelly_notification:208 Event from sys: {"component":"sys", "event":"ota_begin", "msg":"in_progress", "ts":1727723793.02}
shos_rpc_inst.c:242     ota.write via WS_in 192.168.33.91:59241
shelly_ota.cpp:375      bootloader-c3.bin: 20480 bytes (hdr 75 desc 0 CRC f18a07bc)
shos_rpc_inst.c:242     ota.write via WS_in 192.168.33.91:59241
[...]
shelly_ota_esp32_ba:709 Boot: cur 010000ff, new 000802ff, min 00000000, update? 0
shos_rpc_inst.c:242     ota.write via WS_in 192.168.33.91:59241
shelly_ota.cpp:375      C3-8MB-partition-table.bin: 4096 bytes (hdr 84 desc 0 CRC 2207497b)
shos_rpc_inst.c:242     ota.write via WS_in 192.168.33.91:59241
shos_rpc_inst.c:242     ota.write via WS_in 192.168.33.91:59241
shelly_ota_esp32_ba:449 Installing new PT1 (93d694b5)
ota                     Write BS1: seq 16 bst 0x10 seq2 16 | as 0 rs 1 ba 2 mfs 0 c 1 | v 1
ota                     Write BS0: seq 16 bst 0x10 seq2 16 | as 0 rs 1 ba 2 mfs 0 c 1 | v 1
shos_rpc_inst.c:242     ota.write via WS_in 192.168.33.91:59241
shelly_ota.cpp:375      otadata.bin: 8192 bytes (hdr 69 desc 0 CRC 40773c40)
shos_rpc_inst.c:242     ota.write via WS_in 192.168.33.91:59241
shos_rpc_inst.c:242     ota.write via WS_in 192.168.33.91:59241
shos_rpc_inst.c:242     ota.write via WS_in 192.168.33.91:59241
shos_rpc_inst.c:242     ota.write via WS_in 192.168.33.91:59241
shos_rpc_inst.c:242     ota.write via WS_in 192.168.33.91:59241
shelly_ota.cpp:375      tasmota32c3.bin: 1605040 bytes (hdr 73 desc 0 CRC 3a6d211f)
shos_rpc_inst.c:242     ota.write via WS_in 192.168.33.91:59241
[..]
shelly_ota.cpp:375      fs-1024.img: 1048576 bytes (hdr 69 desc 0 CRC c6351727)
shos_rpc_inst.c:242     ota.write via WS_in 192.168.33.91:59241
[...]
shelly_ota.cpp:182      OTA status in_progress, 100 %
ota                     Write BS1: seq 17 bst 0x11 seq2 17 | as 1 rs 0 ba 3 mfs 1 c 0 | v 1
ota                     Write BS0: seq 17 bst 0x11 seq2 17 | as 1 rs 0 ba 3 mfs 1 c 0 | v 1
shelly_ota_esp32_ba:115 Set NVS boot state to 0x310 'LFS' '{"bs":4096,"is":128}' (from 0x0 'LFS' '{"bs":4096,"is":128}')
shelly_ota.cpp:771      Update succeeded, will reboot, uba=3, cd=0/ct=0 s
shos_system.c:60        Rebooting in 3000 ms
shelly_notification:208 Event from sys: {"component":"sys", "event":"scheduled_restart", "time_ms": 2996, "ts":1727723862.32}
shelly_ota.cpp:182      OTA status success, 100 %
shelly_update.cpp:416   FW signatures: want 07 got 00
efuse                   Bits are not empty. Write operation is forbidden.
shelly_update.cpp:443   EFUSE write error bit[0]: 5636
efuse                   Bits are not empty. Write operation is forbidden.
shelly_update.cpp:443   EFUSE write error bit[1]: 5636
shelly_ota_source_rp:25 leftover RPC source 0x3fcbf744
shelly_notification:208 Event from sys: {"component":"sys","event":"config_changed","restart_required":false,"ts":1727723862.56,"cfg_rev":7}
shelly_notification:162 Status change of sys: {"cfg_rev":7}
shelly_notification:208 Event from sys: {"component":"sys", "event":"ota_success", "msg":"success", "ts":1727723862.56}
wifi                    I (141189) wifi:state: run -> init (0)
wifi                    I (141189) wifi:pm stop, total sleep time: 0 us / 138501044 us
wifi                    I (141199) wifi:<ba-del>idx:0, tid:0
wifi                    I (141209) wifi:new:<1,0>, old:<1,1>, ap:<1,1>, sta:<1,1>, prof:2
shos_wifi.c:60          WiFi STA: Disconnected, reason: 8
wifi                    W (141219) wifi:Haven't to connect to a suitable AP now!
shos_lwip.cpp:171       ws(0x3fcbf80c): Link down
shos_lwip.cpp:110       Default: ws(0x3fcbf80c) -> (null) ()
wifi                    W (141239) wifi:Haven't to connect to a suitable AP now!
shos_vfs.c:1072         Unmounting filesystems
shos_lwip.cpp:175       wa(0x3fcbce6c): Status down
shos_lwip.cpp:168       wa(0x3fcbce6c): Removed
shos_lwip.cpp:175       ws(0x3fcbf80c): Status down
shos_lwip.cpp:168       ws(0x3fcbf80c): Removed
wifi                    I (141279) wifi:mode : null
[... 27 lines cut ...]
esp_image: segment 1: paddr=003e9230 vaddr=3fc8da00 size=03028h ( 12328) load
esp_image: segment 2: paddr=003ec260 vaddr=40380000 size=03db8h ( 15800) load
esp_image: segment 3: paddr=003f0020 vaddr=42000020 size=12e150h (1237328) map
esp_image: segment 4: paddr=0051e178 vaddr=40383db8 size=09c14h ( 39956) load
esp_image: segment 5: paddr=00527d94 vaddr=50000000 size=00018h (    24) load
boot: Loaded app from partition at offset 0x3a0000
boot: Disabling RNG early entropy source...

00:00:00.001-244/09 HDW: ESP32-C3
IDF/components/esp_littlefs/src/littlefs/lfs.c:1229:error: Corrupted dir pair at {0x1, 0x0}
00:00:00.055-246/10 UFS: FlashFS mounted with 1016 kB free
00:00:00.056-246/10 TFS: File '.settings' not found
00:00:00.057-246/10 TFS: File '.settings.lkg' not found
00:00:00.067-246/10 CFG: Use defaults
00:00:00.129 QPC: Reset
[... cut ...]
00:00:00.196 TFS: File '.drvset003' not found
00:00:00.196 CFG: Energy use defaults as file system not ready or file not found
00:00:00.204 Project tasmota - Tasmota Version 12.5.0(tasmota32)-2_0_7(2023-04-17T08:06:55)
00:00:00.517 WIF: WifiManager active for 3 minutes
00:00:01.075 HTP: Web server active on tasmota-C6C974-2420 with IP address 192.168.4.1
```

tasmota-AP is available. Auto-configuration works. I do not get readings from the power meterin module yet. I have to look into that issue tomorrow.
Thank you for providing the fix!

closed 2024-10-01 by Jason2866

wildekek, 2024-10-06:
Sir, you are a gentleman and a scholar! 🎩