tasmota/mgos32-to-tasmota32 #75: Shelly Plus Plug S (EU) bricked using the method in the Readme

zeus86, 2024-08-31:
Following the `README` unfortunately bricked 2 of my shellys. 

Steps to reproduce:
* Connect to shelly, provide wifi-credentials, continue with ip in the lan with internet access
* upgraded vanilla fw, upgrades to (IIRC) 1.3.x
* after device is back up, another upgrade is available, upgrade to 1.4.x
* after device is back up, locally flashed provided `mgos32-to-tasmota32-PlusPlugS.zip` (`sha256:dc5c10d50026a8b795bde72ca0ffaa6693243a950ae84ca3b8ebb5ee97ac49d7`)
* device comes back up with tasmota 12.5 in AP-Mode.
* provide wifi-credentials, continuing with LAN IP with internet access
* configure->auto-configure->choose shelly plus plug s 
* device reboots into AP mode with shelly-ssid (not tasmota-ssid)
* acessing the web-ui just gives a `Not Found`
  * Full Output as seen with `curl`:
```
❯ curl -v http://192.168.33.1/ 
*   Trying 192.168.33.1:80...
* Connected to 192.168.33.1 (192.168.33.1) port 80
> GET / HTTP/1.1
> Host: 192.168.33.1
> User-Agent: curl/8.6.0
> Accept: */*
> 
< HTTP/1.1 404 Not Found
< Content-Length: 9
< Server: ShellyHTTP/1.0.0
< Connection: close
< 
* Closing connection
Not Found
~ 
```

* device is essentially bricked

can anyone tell me, what i've been missing? i can only assume, that the fact, that i flashed the OG-firmware twice, that it has different firmwares in both slots before atempting to flash tasmota, which may the problem here. 

I flashed a second shelly, because on the first one i had to forgot to do the "autoconfigure" step. "My bad" I thought, and was willing to do it again, but the outcome was exactly the same on the second device. the readme explicitly states **latest stable version first**, so pleace clarify someone here.

However, it would be really nice, if the Readme would be more specific in this regard, because if serial flashing is the only way to recover from this, this is basically useless, because you cannot open these devices without destroying them, which makes them unusable from then on for most people (me included).

bigbutler, 2024-09-11:
same here Web Ui Not Found i mean is damge thx Jason 140 Dollers dead!

PsychoRS, 2024-09-20:
Exact same issue and same result here, in my case with a Plus 2PM.

bigbutler, 2024-09-20:
Same on Shelly 4pm

lukesav1, 2024-09-21:
Same on Shelly Plus 1PM

zeus86, 2024-09-26:
it would be really nice, if someone from the team could have a look onto this. i don't even know how this behaviour is possible in the first place, now that we have safeboot instead of two mirrored partitions...

lukesav1, 2024-09-27:
> it would be really nice, if someone from the team could have a look onto this. i don't even know how this behaviour is possible in the first place, now that we have safeboot instead of two mirrored partitions.
> 
> With models like 1pm or 4pm or dimmer or something, you at least have the possible route of manually flashing the ESP via it's debug header, but on the plug-s-plus: no way without destroying it physically...
> 
> At least a warning in the `README` would be suitable.

I also tried flashing via the serial header after the error but with no luck, it failed with no comms, in my opinion it was properly bricked. I binned it in the end.

From now on im flashing via serial from the start

fawick, 2024-09-27:
Last night I ran into the same problem: 

- Updated Shelly Plug Plus S v2 to 1.4.2 (first from 1.1.x to 1.3.3 and only then the update to 1.4.2 was offered in Shelly WebUI)
- Installed Tasmota
- Ran auto-configuration

After reboot, the Shelly AP appeared again, and now most HTTP requests result in `Not found`. 

However, I made two additional observations that are not yet listed here:

1) In the wifi network provided by Shelly AP, the endpoint `http://192.168.33.1/ota` is still special. Without any parameters it will complain that it is missing an URL. With one provided (e.g. http://192.168.33.1/ota?url=https://github.com/tasmota/mgos32-to-tasmota32/releases/download/v14.0.0/mgos32-to-tasmota32-PlusPlugS.zip), the plug reponds with something like `Installing firmware from https://github.com/tasmota/mgos32-to-tasmota32/releases/download/v14.0.0/mgos32-to-tasmota32-PlusPlugS.zip`. However, nothing really happens.
2) The plug is still be found via bluetooth by [the Shelly Smart Control App](https://www.shelly.com/en-de/app/shelly-smart-control) and I can get to the point in which I am asked to provide SSID and wifi passwort. But the actual integration of the device into my wifi is not successful.

I wonder whether there are other endpoints to try.

fawick, 2024-09-27:
Good news: I was unable to unbrick my Plus Plus S (v2).

Here's how:
- Download a shelly stock firmware for your device. I used http://rojer.me/files/shelly/stock/1.3.3/PlusPlugS.zip
- Connect to the Shelly AP, let's assume you get the IP 192.168.33.42

- In a second console, listen with netcat to an arbitrary UDP port, e.g. ` nc -u -l 56789`
- Run `curl -X POST -d '{"id":1,"method":"Sys.SetConfig","params":{"config":{"debug":{"udp":{"addr":"192.168.33.42:56789"}}}}}' http://192.168.33.1/rpc` to let the Shelly firmware send debug messages via UDP
- In the folder, where you downloaded the ZIP, run `python -m http.server` to serve this folder via HTTP
- curl "http://192.168.33.1/ota?url=http://192.168.33.42:8000/PlusPlugS.zip" 
- Wait a bit, there should be progress messages coming in via UDP, check the `nc` console
- Afterwards, you can access the Shelly Web UI again.

Hope that helps all of you a bit.

Glotzon, 2024-10-03:
Same issue with one out of four Shelly PlusPlugS.
Successfully recovered using this method: https://github.com/tasmota/mgos32-to-tasmota32/issues/75#issuecomment-2379974905

Now I need to decide if I'll give it another try, but thank you @fawick for the awesome work to figure out how recovery works.

zeus86, 2024-10-15:
> Good news: I was unable to unbrick my Plus Plug S (v2).
> 
> Here's how:
> [...]
> Hope that helps all of you a bit.

I have no Idea how you came up with this, but this totally works. I was able to unbrick both plugs by hitting the http-api as described. the debug-messages did not work for me, but the flashing itself did. Awesome that the seemingly defunct webserver takes OTA-requests anyways. You see the plug hitting the webserver for the firmware-file and blinking red afterwards, indicating an OTA-Flash. 

To be honest, i much prefer this  method of flashing anyways instead of this "drag this file to this textbox", which produces drastically different results depending on the OS, the browser and things like wayland, flatpaked browser, network setup and so on. 

However, thank you alot for this solution.

Whatsoever: this of course does not solve the original problem, that the readme-method is broken für plugplus-s, which you cannot easily flash via serial, so this issue itself remains valid.

coaxx77, 2024-10-17:
fawick's method to get bricked device back to shelly firmware worked for me, too. I assume 

> There is no way back to Shelly firmware if you have initiated the convert process!

must be wrong then.

majbom, 2024-12-09:
I seem to have the same problem with a PM Mini gen 3 - I haven't been able to find a stock fw for that device.
I tried with the Mini1PMG3 from rojer.me, but it refuses to update with that file, which of cours is a good thing.
Does anyone know where else to get stock fw?

Jason2866, 2024-12-09:
There is no stock firmware available for esp32x based devices. If the OTA process failed for whatever reason only alternative firmwares are the way. Flash via serial the Tasmota32 factory image.

majbom, 2024-12-09:
Did one final search after posting, and found this: https://updates.shelly.cloud/update/MiniPMG3 and now the device is back at stock fw 🎉
So now it's running 1.4.4. Should I try with the [mgos32-to-tasmota32-MiniPMG3.zip](https://github.com/tasmota/mgos32-to-tasmota32/releases/download/v14.1.0/mgos32-to-tasmota32-MiniPMG3.zip) once more or will it just fail again?

Jason2866, 2024-12-09:
@majbom Thats an update package, not an full firmware. So the Autoconfig process did not work and did not replaced the Shelly bootloader and Shelly partitions (in this case the Shelly update package would not work)
Honestly idk why it did not work first time. So when you don't want to risk the device don't try.

majbom, 2024-12-09:
@Jason2866 Thanks for your time.
The Autoconfig is a part of the mgos->tasmota zip? And if it HAD been working, I wouldn't have been able to revive the device this way?
The readme says that the Mini PM Gen 3 has been tested okay, but is that with an older shelly fw maybe?
If I choose to go with the serial solution, which file do I need to flash?

Jason2866, 2024-12-09:
You can try again via OTA. If it fails you can use the Tasmota32 C3 factory image

majbom, 2024-12-10:
> You can try again via OTA. If it fails you can use the Tasmota32 C3 factory image

Thanks, it seems to be working now. I think I might missed a step in the conversion guide during the first try.

closed 2024-12-11 by Jason2866

copperfield74, 2025-03-05:
@fawick Thanks so much for your work. I was able to recover two of my devices but now I have one that still shows NOT FOUND when I try to enter the configuration page after your workaround. Any ideas what firther steps I can try?