tasmota/mgos32-to-tasmota32 #77: Shelly Mini Gen3 - No AP appears after loading mgos32-to-tasmota32-Mini1G3.zip

alexstarke, 2024-09-21:
I am pretty sure that I followed the [instructions](https://github.com/tasmota/mgos32-to-tasmota32?tab=readme-ov-file#lets-start) thoroughly to convert the Shelly Mini Gen3 modules. Still I am stuck after loading mgos32-to-tasmota32-Mini1G3.zip, no tasmota AP appears. 
Anything I can do to recover the module? Is it possible to serial flash the Shelly Mini Gen3? 

Any help is much appreciated, Thanks!

rp2019433, 2024-09-22:
Hi.
Please take a look at #68 and #74 .

Jason2866, 2024-09-30:
Please try https://github.com/tasmota/mgos32-to-tasmota32/releases/tag/v14.1.0

alexstarke, 2024-09-30:
> Please try https://github.com/tasmota/mgos32-to-tasmota32/releases/tag/v14.1.0

Thanks a lot for looking into this! I need to order new modules, so feedback will take a while ...

StomaNL, 2024-10-09:
Flashed my PM Mini Gen3, Tasmota AP is up, filled in my network credentials but never connects to my network. AP shows up again but now I can't connect to the device, keeps throwing me off when it evens connect at all (on laptop and phone). 

Is flashing Tasmota over serial the way to resurrect?

Jason2866, 2024-10-10:
Probably the solution is #80

closed 2024-11-16 by Jason2866

alexstarke, 2024-12-07:
> > Please try https://github.com/tasmota/mgos32-to-tasmota32/releases/tag/v14.1.0
> 
> Thanks a lot for looking into this! I need to order new modules, so feedback will take a while ...

I finally got to testing it successfully with a new device, thanks for you support!