tasmota/mgos32-to-tasmota32 #79: Cannot update Shelly Mini1G3 12.5 to 14.0

jankkm, 2024-10-09:
I converted the Shelly Mini 1 G3 a while ago and it worked good for a while and today I did an OTA update which I now know was a mistake.

I can still use it but it is stuck on Tasmota 12.5 and I can do an update to 14.0 from the release page but after the reboot it is still on 12.5

I need Matter support so this is quite a problem for me.

Is there a way to fix this without serial access?
[image]
[image]

Jason2866, 2024-10-09:
If the Autoconfig step was not done right after the OTA convert process, the locked Shelly bootloader is not replaced and device can't not be updated via OTA.
Anyways you can try the Autoconfig step again. This step will update to latest Tasmota version if it is possible (no locked Shelly bootloader). If this does not work serial flash is the only way to update to a newer Tasmota version

jankkm, 2024-10-09:
I‘m not sure if you unterstood me correctly. Is did the full conversion correctly a while ago and it worked fine (with Matter). Today I clicked on the OTA Firmware Button
[image]
Does the bootloader have to be replaced again?

Jason2866, 2024-10-10:
If the Autoconfig step was not done the locked bootloader is not replaced. With the Shelly bootloader there is no update of the Tasmota firmware possible. So the step was recommended from me to be sure to have a unlocked bootloader. There is no reason when all this is done that a OTA update does not work.
Are you sure the device can reach the OTA Server? The device needs Internet access.
You can alternatively download the newer firmware and provide via file firmware upgrade.

jankkm, 2024-10-10:
Thanks for your response.

The autoconfig step was done so I guess I have a different problem.

I downloaded the newest release here (https://github.com/tasmota/mgos32-to-tasmota32/releases/download/v14.1.0/mgos32-to-tasmota32-Mini1G3.zip), unpacked it and used the "tasmota32c3.bin" for the firmware upgrade. This seems to work fine but afterwards I am still on Tasmota 12.5

My suspicion is that the upload does not work for some reason so it uses a firmware bin that is still on the device. But I don't know how to check for that.

Jason2866, 2024-10-10:
As expected, the convert package does have v12.5 Tasmota firmware.
Download actual firmware from https://ota.tasmota.com/tasmota32/

jankkm, 2024-10-10:
Silly me, of course it works. Thank you so much for your patience!

closed 2024-10-10 by Jason2866