tasmota/mgos32-to-tasmota32 #81: Tasmota SSID - No web page

yaba, 2024-10-12:
After following the tutorial, I can see the new Tasmota SSID, connect to it and ping 192.168.4.1 as expected.

However, no webpage is displayed and device enters a reboot loop.

Shelly PM Mini Gen 3 previously running 1.4.x firmware.

BTW, this can be fixed by old method? (with serial flash)

Thanks.

mr-awk, 2024-11-24:
I did also follow the instructions with firmware 1.4.4, and now I only see the Shelly SSID, and can't connect to that network. So I would say that PMMiniG3 with firmware 1.4.4 is not compatible with the installation instructions for now.

Jason2866, 2024-12-11:
@mr-awk if the Shelly AP comes up the mandantory steps 8-10 are not done.

closed 2024-12-11 by Jason2866