tasmota/mgos32-to-tasmota32 #83: Shelly Plug US Gen 2 to Tasmota fails halfway

spockatieee, 2024-10-20:
Hi. I'm trying to transition a Shelly PlugUS Gen2 (Shelly firmware 1.4.4) to Tasmota32.

I followed the README instructions, and got as far as generating the Autoconf for the Shelly Plug US, while booted into Tasmota 12.5.0 Solo1. While I did have a bricked moment, doing OTA on local network with the mgos32-to-tasmota32 revived the Shelly and brought me back into Tasmota 12.5.0 (supposedly with the locked bootloader).

The next step, clicking on the Partition Wizard, just hangs. Looking at the console, these are the Berry errors that are generated (tried setting as high as WebLog 5), but no difference:

BRY: Exception> 'type_error' - unsupported operand type(s) for <=: 'nil' and 'int'
stack traceback:
<unknown source>: in function `factory_migrate_eligible`
<unknown source>: in function `show_migrate_to_factory`
<unknown source>: in function `page_part_mgr`
<unknown source>: in function `<lambda>`

Any clues on what may be wrong here?

Thanks!

Jason2866, 2024-10-21:
Only serial flash is the way out. There are 3 boot attempts when a not Shelly signed firmware is installed (with the Shelly bootloader) until it always starts the firmware at address 0x10000. The Autoconfiguration step does replace the Shelly bootloader against a bootloader without firmware sign checks.

spockatieee, 2024-10-21:
I see. So serial flash straight to factory Tasmota, or something else?

Jason2866, 2024-10-23:
When serial flashing, directly flash Tasmota. No other step needed.