tasmota/mgos32-to-tasmota32 #84: Question regarding (un)-locked bootloader on Shelly Plus 1 and Plus 1PM

gilex-dev, 2024-10-21:
Hi, first of all THANK YOU SO MUCH  for making it possible to easily flash tasmota to these devices. 
I am thinking about using these devices with other Platform IO projects but the part about the locked bootloader freaked me out.
I read that it uses an eFuse and cannot really be disabled and only allows signed binaries in that state.
Now I am confused, because tasmota clearly works but is not signed (or is it???).
Would I brick the ESP32 if I now tried to flash a custom script or did the conversion process somehow truly unlock the bootloader?

Jason2866, 2024-10-21:
The bootloader is replaced with the Autoconfiguration step. Until now Shelly has not locked the devices to flash other firmwares. Just when the bootloader is not replaced the device gets soft bricked. Currently this is no issue since Shelly has not blocked flashing via serial (esptool.py). But only Shelly knows if this will be possible in future.

gilex-dev, 2024-10-21:
Thanks for the quick and helpful answer.
Yeah, let's hope they will never do that. I only see the need for encrypted / locked down IOT devices in a professional environment. But that probably not Shelly's targeted audience with these products.

closed 2024-11-16 by Jason2866