tasmota/mgos32-to-tasmota32 #85: Convert to Tasmota failed from 1.4.4 - Shelly Pro 4 PM

cDev1987, 2024-10-24:
Hi, 
today i received my Shelly Pro 4 PM and i tried to flash Tasmota fw to the Shelly, but after update it came back to my Wifi Network and i still see the Shelly GUI. 
I read the thread where many users reported to have a bricked device after update from 1.4.2  
From 1.4.4 it seems that the update is not possible, but device is not bricked. 

20241011-114451/1.4.4-g6d2a586
tried mgos32-to-tasmota32-Pro4PM from latest release 3 weeks ago

cDev1987, 2024-10-27:
I saw there is pro 4pm v1 and v2 - is there a difference wrt flashing Tasmota? 

Device name: Shelly Pro 4PM V2
Device model: SPSW-104PE16EU

Device name: Shelly Pro 4PM
Device model: SPSW-004PE16EU

Jason2866, 2024-11-12:
`"data": "shelly_update.cpp:421 FW signatures: want 07 got 00",` The log shows everything okay. We don't have signed firmware. The message is always there when using the convert script

cDev1987, 2024-11-15:
Hi @Jason2866 
thanks for your feedback. Anything else i could check or i can try? 
Or is there a way to get more debug data from Shelly device?

Jason2866, 2024-11-16:
Without the device idk how to find the reason for.

Jason2866, 2024-12-16:
Issue was caused from the new Shelly fw. 1.4.x Latest convert packages do support this firmware version

closed 2024-12-16 by Jason2866

ramponis, 2024-12-18:
> Issue was caused from the new Shelly fw. 1.4.x Latest convert packages do support this firmware version

So now the procedure is update it to last 1.4.x (1.4.4 for 4PM) then convert using last build.
Can you confirm it?

Jason2866, 2024-12-18:
@ramponis To use the actual convert packages a Shelly firmware with version 1.4.x is needed.
I had actual feedback for successful convert. As always no guarantees, you have to be fine to flash via serial if something goes wrong during OTA convert.

ramponis, 2024-12-18:
Thanks @cDev1987 for your feedback.
What do you think to try to downgrade shelly firmware
[downgrade](https://community.shelly.cloud/topic/454-downgrade-links-to-014x/)
and then try again to convert using 
[v13.4.1](https://github.com/tasmota/mgos32-to-tasmota32/releases/tag/v13.4.1)
Can this procedure work?

Jason2866, 2024-12-18:
@cDev1987 What and/or when the convert process fails?
Checked the convert package multiple times. I see nothing wrong there. As it is, it should do the convert when there is a Shelly fw 1.4.x running when applying the convert package.

Can't test myself since I don't have this device. So when the process reproducable fails there is **no solution in sight**

Jason2866, 2024-12-18:
@ramponis There are positive feedbacks for convert the Pro4PM when the Shelly firmware is pre 1.4 with convert packages v13.4.1.

The Pro4PM we made ours tests is in productive use and can not be used for further tests.

Jason2866, 2024-12-18:
As I wrote a few posts before, closed this issue since i have got a positive feedback of converting a Pro4PM with Shelly fw 1.4.x.

ramponis, 2024-12-18:
I have the same issue as @cDev1987
My 4PM seems to apply the firmware but after reboot it reloads shelly firmware.
On it there is ver. 20240726-114529/1.4.0-gb2aeadb, also it show me 1.4.4 is available.

cDev1987, 2024-12-18:
@ramponis exactly the same here. 

@Jason2866 
Running FW 20241011-114451/1.4.4-g6d2a586

After uploading the FW it will go down for a reboot but came back with same shelly FW. 
If there is a option to get more debug data from Pro 4PM remotely, let me know. 

Everything what i can do remotly i will do. But for now i will not disconnect everything and remove it from my fuse box and open the device to connect serial port and so on.

Jason2866, 2024-12-18:
@cDev1987 Thx for the friendly offer. But no, need the device physically and see what happens and trying changes depending on what I see. Not a straitforward process. Ideas do come when seeing what's happen (or not).
So there is nothing in sight. I do not buy the device, since i have no use for.

ramponis, 2025-01-09:
I did the test, but the downgrade fails from 1.4.0, tried many times.
After the reboot the firmware is unchanged.
Also, I have upgraded it to 1.4.4 an tested agian the downgrade with the same result.
Tested tasmota conversion using mgos32 v14.4.1 but same result.
So now the only way is using the serial interface