tasmota/mgos32-to-tasmota32 #88: Support for Shelly Plug S Gen3

jenrg, 2024-11-05:
Requesting support for Shelly Plug S Gen3. Only has FW 1.2.3-matter22 available at the moment as far as I know.

KB link: https://kb.shelly.cloud/knowledge-base/shelly-plug-s-mtr-gen3

Screenshot:
[image]

Jason2866, 2024-11-05:
Well would be surprised when a Matter capable device would be able to flash a non signed firmware.

Jason2866, 2024-11-05:
For building and testing a full firmware dump is needed.

oxynatOr, 2024-12-01:
if u need, i can give u a full-dump in a few...

```
Connecting....
Detecting chip type... ESP32-C3
Chip is ESP32-C3 (QFN32) (revision v0.4)
Features: WiFi, BLE, Embedded Flash 8MB (GD)
Crystal is 40MHz
MAC: b0:81:84:a4:f0:40
Uploading stub...
Running stub...
Stub running...
Warning: ESP32-C3 has no Chip ID. Reading MAC instead.
MAC: b0:81:84:a4:f0:40
Hard resetting via RTS pin...
```

8MB take a while... 

btw i do not use tasmota, so i cant build u a fw file for tasmota, but for esphome. at least that my plan.

Jason2866, 2024-12-01:
Don't have a esp32-c3 with 8MB flash, so I can't use the dump.

oxynatOr, 2024-12-03:
(y) all righty did it! 

for me it looks encrypted, but i don't have skill about FW-Hacking. just used binwalk. 
if you can figure something out, like Pinout of LEDs, Relay and Temp sensor, pls share it with me ;) 

Ciao

oxynatOr, 2024-12-03:
hmm, maybe i'm wrong, but in theory.. could i use the OTA-Zip from MiniG3 (its the same ESP-Chip), the internal name of the Plug is: PlugSG3. 

` 
 "name": "PlugSG3",
  "platform": "esp32c3",
  "version": "12.5.0",
  "build_id": "20241109-141726/tasmota-12.5.0",
  "build_timestamp": "2024-11-09T14:17:26Z",
`

i try it with no luck. maybe you have an idea/tipp? 

`
shos_rpc_inst.c:230 ota.start via WS_in 10.255.84.193:52495
shelly_ota.cpp:239 Starting update (0x3fcd0560), to=600 cd=0/ct=0 ignore_same=0
shos_rpc_inst.c:230 ota.write via WS_in 10.255.84.193:52495
shelly_ota.cpp:357 bootloader-c3.bin: 20480 bytes (hdr 47 desc 0 CRC f18a07bc)
shelly_ota.cpp:246 Aborting update: manifest.json was not first in archive
shos_rpc_inst.c:230 ota.write via WS_in 10.255.84.193:52495
shelly_ota.cpp:173 OTA status error, 0 %
shelly_ota_source_rp:23 leftover RPC source 0x3fcd0474
shelly_notification:209 Event from sys: {"component":"sys", "event":"ota_error", "msg":"error", "ts":1733212624.58}
shos_rpc_inst.c:230 ota.write via WS_in 10.255.84.193:52495
shos_rpc_inst.c:230 ota.abort via WS_in 10.255.84.193:52495
`

oxynatOr, 2024-12-03:
```
Matter-SWU:2 Default Provider timer handler is invoked
09:23:30
Matter-SWU:1 No suitable OTA Provider candidate found
09:23:30
Matter-SWU:2 Stopping the watchdog timer
09:23:30
Matter-SWU:2 Starting the periodic query timer, timeout: 300 seconds
09:23:30
shos_rpc_inst.c:230 ota.start via WS_in 10.255.84.193:52495
09:23:31
shelly_ota.cpp:239 Starting update (0x3fcd03f8), to=600 cd=0/ct=0 ignore_same=0
09:23:31
shos_rpc_inst.c:230 ota.write via WS_in 10.255.84.193:52495
09:23:31
shelly_ota.cpp:357 manifest.json: 1775 bytes (hdr 71 desc 0 CRC efd3b12c)
09:23:31
shelly_ota.cpp:531 fw: PlugSG3 ver 12.5.0 20241203-092222/tasmota-12.5.0
09:23:31
shelly_update.cpp:330 incoming fw signatures: 00
09:23:31
shelly_ota_esp32_ba:470 Will write to slot 0
09:23:31
shelly_notification:209 Event from sys: {"component":"sys", "event":"ota_begin", "msg":"in_progress", "ts":1733214211.16}
09:23:31
shos_rpc_inst.c:230 ota.write via WS_in 10.255.84.193:52495
09:23:31
shelly_ota.cpp:357 bootloader-c3.bin: 20480 bytes (hdr 75 desc 0 CRC f18a07bc)
09:23:31
shos_rpc_inst.c:230 ota.write via WS_in 10.255.84.193:52495
09:23:31
shos_rpc_inst.c:230 ota.write via WS_in 10.255.84.193:52495
09:23:31
shos_rpc_inst.c:230 ota.write via WS_in 10.255.84.193:52495
09:23:31
shos_rpc_inst.c:230 ota.write via WS_in 10.255.84.193:52495
09:23:31
shos_rpc_inst.c:230 ota.write via WS_in 10.255.84.193:52495
09:23:31
shos_rpc_inst.c:230 ota.write via WS_in 10.255.84.193:52495
09:23:31
shos_rpc_inst.c:230 ota.write via WS_in 10.255.84.193:52495
09:23:31
shos_rpc_inst.c:230 ota.write via WS_in 10.255.84.193:52495
09:23:31
shos_rpc_inst.c:230 ota.write via WS_in 10.255.84.193:52495
09:23:31
shos_rpc_inst.c:230 ota.write via WS_in 10.255.84.193:52495
09:23:31
shelly_ota_esp32_ba:668 Boot: cur 010000ff, new 000802ff, min 00000000, update? 1
09:23:31
shos_rpc_inst.c:230 ota.write via WS_in 10.255.84.193:52495
09:23:31
shelly_ota.cpp:357 C3-8MB-partition-table.bin: 4096 bytes (hdr 84 desc 0 CRC 2207497b)
09:23:31
shos_rpc_inst.c:230 ota.write via WS_in 10.255.84.193:52495
09:23:32
shos_rpc_inst.c:230 ota.write via WS_in 10.255.84.193:52495
09:23:32
shelly_ota_esp32_ba:403 PT0 unchanged
09:23:32
shos_rpc_inst.c:230 ota.write via WS_in 10.255.84.193:52495
09:23:32
shelly_ota.cpp:357 otadata.bin: 8192 bytes (hdr 69 desc 0 CRC 40773c40)
09:23:32
shos_rpc_inst.c:230 ota.write via WS_in 10.255.84.193:52495
09:23:32
shos_rpc_inst.c:230 ota.write via WS_in 10.255.84.193:52495
09:23:32
shos_rpc_inst.c:230 ota.write via WS_in 10.255.84.193:52495
09:23:32
shos_rpc_inst.c:230 ota.write via WS_in 10.255.84.193:52495
09:23:32
shos_rpc_inst.c:230 ota.write via WS_in 10.255.84.193:52495
09:23:32
shelly_ota.cpp:357 tasmota32c3.bin: 1605040 bytes (hdr 73 desc 0 CRC 3a6d211f)
09:23:32
shelly_ota_esp32_ba:510 PT is being moved from 0x10000 to 0xf000
09:23:32
shelly_ota_esp32_ba:282 Installing PT 20480@app_0(0x20000) -> pt(0x10000)...
09:23:32
shelly_ota.cpp:246 Aborting update: backend 0x3fcd05dc aborts update at file tasmota32c3.bin: -7: failed to copy 20480@app_0 -> pt : Invalid write offset 0
09:23:32
shos_rpc_inst.c:230 ota.write via WS_in 10.255.84.193:52495
09:23:32
shelly_ota.cpp:173 OTA status error, 1 %
09:23:32
shelly_ota_source_rp:23 leftover RPC source 0x3fcd0a40
09:23:32
shelly_notification:209 Event from sys: {"component":"sys", "event":"ota_error", "msg":"error", "ts":1733214212.38}
09:23:32
shos_rpc_inst.c:230 ota.write via WS_in 10.255.84.193:52495
09:23:32
shos_rpc_inst.c:230 ota.abort via WS_in 10.255.84.193:52495
09:23:32
Shelly PlugSG3 B08184A4F040| 1.2.3-matter22| 88e1d930| 09:23:59 | [Feedback](https://forms.office.com/Pages/ResponsePage.aspx?id=4nYJ453e502VZA-uzySuPVkpSEOEYHdOhR2DIwiMd01UOVFOOE5OVlhCMzdEMUFQUjdPOFBDRVo3TSQlQCN0PWcu)
```

Jason2866, 2024-12-03:
This is not looking good. Should not happen

```
09:23:32
shelly_ota.cpp:246 Aborting update: backend 0x3fcd05dc aborts update at file tasmota32c3.bin: -7: failed to copy 20480@app_0 -> pt : Invalid write offset 0
```

Jason2866, 2024-12-03:
Mixing firmware versions is not gonna working, since there are different partition layouts.

oxynatOr, 2024-12-04:
yea true.  is there anything you guys need or i can test/do?

Jason2866, 2024-12-05:
@oxynatOr No, will do investigations when i am able to do. Don't hold your breath ;-)
I think there will be no option for OTA converting devices with Matter support.

cociweb, 2024-12-11:
In the meantime, should we add into the readme.md as an unsupported device? - I've almost started the procedure as I saw plugS is tested, but gen3 is different....

Jason2866, 2024-12-11:
@cociweb Good idea. PR welcome

Jason2866, 2024-12-16:
Matter devices are not possible to convert.

sqanch, 2025-03-30:
Can you give some more details? Will the Plug S Gen 3 never be convertible or just at this time? And does that apply to all matter devices?