tasmota/mgos32-to-tasmota32 #89: Shelly Plus Plug S dosn't work

MCSE-2000, 2024-11-07:
My new Shelly Plus Plg S had firmware 0.13.0. I have update the Firmware to Shelly 1.14.4.
I have selected at Point 8 Configuration > Auto-configuration >  "Shelly Plus PlugS".
After pressing  "Apply configuration" the Shelly come with a WiFi "ShellyPlusPlugS-{MAC-Adress}".
I can connect with the Shelly and become a IP 192.168.33.68 but i can't get a WEB-Gui at 192.168.33.1!
What is wrong?

Jason2866, 2024-11-16:
Something went wrong. To recover device needs to be flashed via serial to recover.

closed 2024-11-16 by Jason2866

joshinils, 2024-12-10:
I think I've done the same.

I had the tasmota interface working, and wanted to enter the configuration `{"NAME":"Shelly Plus Plug S","GPIO":[0,0,0,0,224,0,32,2720,0,0,0,0,0,0,0,2624,0,0,2656,0,0,288,289,0,0,0,0,0,0,4736,0,0,0,0,0,0],"FLAG":0,"BASE":1}` from https://templates.blakadder.com/shelly_plus_plug_S.html 

after that I waited a bit, nothing, tried again, but nothing. I think I unplugged it and then it came up with the previous AP, but connecting to it gives no web gui, only "Not Found". It's blinking blue and that's it.

Did I enter the wrong config? how can I avoid this with the second plug I have (still untouched)?

Jason2866, 2024-12-10:
Do as described in the readme here in the repo. Just when all steps are done it is safe to use!