tasmota/mgos32-to-tasmota32 #90: Shelly 1 Gen3

ulferi, 2024-11-09:
I have a shelly 1 Gen 3. If im reading correctly i cant find it in supported devices
Witch file or method shall i use to upgrade to tasmota,
I have the latest firmware 1.4.4 from Shelly on the device.

Jason2866, 2024-11-09:
Which device exactly? Please post a screenshot of the main Shelly Web page.
In upper left corner there is the internal Shelly device name, needed to make a convert package if the device is really new and different.

RavenSystem, 2024-11-09:
S1G3

Jason2866, 2024-11-09:
Try https://github.com/tasmota/mgos32-to-tasmota32/releases/download/v14.1.1/mgos32-to-tasmota32-S1G3.zip
It is an untested build! I don't have the device. It is build from the infos of the manifest json of the orig. Shelly firmware update package.
**If convert works do the Autoconfiguration step!** Mandantory to replace the locked bootloader!
After that you have to find the config to setup the device correctly.
Don't try if you can not flash the device by serial if OTA convert does not work and device gets soft bricked!

ulferi, 2024-11-09:
Device name: Shelly 1 Gen3

    Device model: S3SW-001X16EU

    Device SSID: Shelly1G3-XXXXXXXXXXXX

    BLE Model ID: 0x1018

ulferi, 2024-11-09:
I will try the beta!

ulferi, 2024-11-09:
Hi again!
The upload went fine!

And the i did se the Tasmota wifi, But after a short while it disapered!
(Before i was able to connect)
What shall i try in this situation?

Jason2866, 2024-11-09:
Serial flash is probably the only way. I don't know what happened.

ulferi, 2024-11-10:
When i run the web https://tasmota.github.io/install/ to install i get this message.
I have soldered wires to the board and the led is flashing. But i dont succeed to program the unit.
Is it a special way to get into programming mode!

[image]

TD-er, 2024-11-10:
Typically keep the button pressed while powering on (or reset) and keep pressed till you have the flashing started.

For ESP8266/ESP32/ESP32-S2/ESP32-S3 the button is wired to GPIO-0 and if not present, you need to pull GPIO-0 to GND to enter flash mode.

ulferi, 2024-11-17:
Can someone guide me to the right firmware when i flash my device!

RavenSystem, 2024-11-24:
@ulferi Could you try this config?

```
{"NAME":"Shelly 1 Gen3","GPIO":[576,32,0,4736,0,224,0,0,0,0,192,0,0,0,0,0,0,0,0,0,0,0],"FLAG":0,"BASE":1,"CMND":"AdcParam1 2,5600,4700,3350"}
```

ulferi, 2024-11-30:
> @ulferi Could you try this config?
> 
> ```
> {"NAME":"Shelly 1 Gen3","GPIO":[576,32,0,4736,0,224,0,0,0,0,192,0,0,0,0,0,0,0,0,0,0,0],"FLAG":0,"BASE":1,"CMND":"AdcParam1 2,5600,4700,3350"}
> ```
Thanks!
On off works fine but the temperature is not according to the indoor temperature. Its about 22 deg C.
[image]

RavenSystem, 2024-11-30:
@ulferi could you check if physical SW input works? Built-in button? And LED, please?

ulferi, 2024-12-01:
> @ulferi could you check if physical SW input works? Built-in button? And LED, please?

Built in button works. LED is lit the whole time

RavenSystem, 2024-12-01:
@ulferi and physical SW input?

ulferi, 2024-12-01:
> @ulferi and physical SW input?

Physical input works fine! But not the temperature!

RavenSystem, 2024-12-01:
Ok. First, check if LED works with this:
```
{"NAME":"Shelly 1 Gen3","GPIO":[0,32,576,4736,0,224,0,0,0,0,192,0,0,0,0,0,0,0,0,0,0,0],"FLAG":0,"BASE":1,"CMND":"AdcParam1 2,5600,4700,3350"}
```

ulferi, 2024-12-01:
> Ok. Check if LED and temp work with this:
> 
> ```
> {"NAME":"Shelly 1 Gen3","GPIO":[0,32,576,4736,0,224,0,0,0,0,192,0,0,0,0,0,0,0,0,0,0,0],"FLAG":0,"BASE":1,"CMND":"AdcParam1 2,10000,10000,3350"}
> ```

The LED is still on. Nothing happens with LED. What shall i expect to see

RavenSystem, 2024-12-01:
Well, I am not sure. I don't use Tasmota firmware. It is a "Link led".

And temperature?

ulferi, 2024-12-01:
> Well, I am not sure. I don't use Tasmota firmware. It is a "Link led".
> 
> And temperature?

Temperature is 48 deg C but the real temp is about 10 deg C

RavenSystem, 2024-12-01:
Try this and check temp and LED again, please. I think that now, the LED will have the same status as relay:

```
{"NAME":"Shelly 1 Gen3","GPIO":[320,32,0,4736,0,224,0,0,0,0,192,0,0,0,0,0,0,0,0,0,0,0],"FLAG":0,"BASE":1,"CMND":"AdcParam1 2,10000,10000,4000"}
```

ulferi, 2024-12-01:
> Try this and check temp and LED again, please. I think that now, the LED will have the same status as relay:
> 
> ```
> {"NAME":"Shelly 1 Gen3","GPIO":[320,32,0,4736,0,224,0,0,0,0,192,0,0,0,0,0,0,0,0,0,0,0],"FLAG":0,"BASE":1,"CMND":"AdcParam1 2,10000,10000,4000"}
> ```

The LED is always on. In open closed mode.
The temperature is now 43 degC

RavenSystem, 2024-12-01:
Ok. First, try to find the LED. Use this, and then, try LED (currently at GPIO 0) with all unused GPIOs in the Template Parameters, one by one.

```
{"NAME":"Shelly 1 Gen3","GPIO":[320,32,0,0,0,224,0,0,0,0,192,0,0,0,0,0,0,0,0,0,0,0],"FLAG":0,"BASE":1}
```

ulferi, 2024-12-01:
> Ok. First, try to find the LED. Use this, and then, try LED (currently at GPIO 0) with all unused GPIOs in the Template Parameters, one by one.
> 
> ```
> {"NAME":"Shelly 1 Gen3","GPIO":[320,32,0,0,0,224,0,0,0,0,192,0,0,0,0,0,0,0,0,0,0,0],"FLAG":0,"BASE":1}
> ```

LED working ; {"NAME":"VentFlakttilluftVind","GPIO":[0,32,0,0,0,224,0,0,0,0,192,0,0,0,0,0,0,0,0,320,0,0],"FLAG":0,"BASE":1}

RavenSystem, 2024-12-01:
Now, same way. Use this and check all free GPIOs with ADC:
```
{"NAME":"Shelly 1 Gen3","GPIO":[0,32,0,4736,0,224,0,0,0,0,192,0,0,0,0,0,0,0,0,320,0,0],"FLAG":0,"BASE":1,"CMND":"AdcParam1 2,10000,10000,3350"}
```

ulferi, 2024-12-01:
> Now, same way. Use this and check all free GPIOs with ADC:
> 
> ```
> {"NAME":"Shelly 1 Gen3","GPIO":[0,32,0,4736,0,224,0,0,0,0,192,0,0,0,0,0,0,0,0,320,0,0],"FLAG":0,"BASE":1,"CMND":"AdcParam1 2,10000,10000,3350"}
> ```

GPIO3 is the only one that gives a positive value but the temperature is not right.

ulferi, 2024-12-01:
> Which device exactly? Please post a screenshot of the main Shelly Web page. In upper left corner there is the internal Shelly device name, needed to make a convert package if the device is really new and different.

https://www.shelly.com/products/shelly-1-gen3?_gl=1*xqw760*_up*MQ..*_gs*MQ..&gclid=Cj0KCQiAr7C6BhDRARIsAOUKifhWwnVtMG-WaR-G9_gES3Xn-Ro9dDerG3VkQn7EZkIkST39TN42c9YaAvtzEALw_wcB

RavenSystem, 2024-12-02:
@ulferi Ok. Next step is to find right parameters for ADC, but unfortunately I cannot help you with this. Maybe @Jason2866 could.

Jason2866, 2024-12-02:
@ulferi just ask in Tasmota discussions for the needed settings and post there the actual findings so that everyone can see. This repo is just a helper repo for the main Tasmota repo.
https://github.com/arendst/Tasmota/discussions

Jason2866, 2024-12-10:
Closing, since not an convert issue.