tasmota/mgos32-to-tasmota32 #91: Shelly Plus i4 successfully flashed to Tasmota opens Shelly AP after restart

waldorffsmama, 2024-11-26:
I have bought a couple of Shelly Plus I4. Now after updating the Shelly to firmware 1.4.4, I managed to successfully flash Tasmota firmware with the mgos32-to-tasmota32-PlusI4a.zip file. It restarts, I find the Tasmota AP as usual, set it up and change all parameters in the console as needed.

After restarting, it does not login to my WiFi-Network but opens again a Shelly HotSpot. There is no WebUI anymore on 192.168.33.1 and I am stuck. This is the second device where it happens.

Any ideas what could have gone wrong?

Jason2866, 2024-11-26:
I would need the device to do investigations. Without it is hard / impossible to find the reason for failing.

waldorffsmama, 2024-11-26:
I can send one to you.

Jason2866, 2024-11-26:
You can contact me on Tasmota Discord Server https://discord.gg/Ks2Kzd4
via PM for details. My nick is Jason2866 there too.

gerritb, 2024-11-27:
Shelly 1 Mini Gen3 has the exact same behavior if that helps debugging

Tillomar, 2024-11-29:
Shelly Plus Plug S (on 20241011-114442/1.4.4-g6g2a586) also has this behavior. Bricked one, recovery procedure ([https://github.com/tasmota/mgos32-to-tasmota32/issues/75#issuecomment-2379974905]) does not work; waiting for your analysis.

copperfield74, 2024-12-14:
Same problem here with Shelly Plus Plug S. Was able to recover but same problem after second try.

Jason2866, 2024-12-16:
@waldorffsmama thx for sending a Shelly i4. Can investigate now.

Tillomar, 2024-12-16:
Jason, if having a Shelly Plus Plug S too would help, let me know.

Jason2866, 2024-12-16:
@Tillomar Dont know if I find something with the Shelly Plus Plug S. If you don't mind sending me one (don't send anything back). Ping me on Discord. For sure I will investigate!!
The Shelly Plus I4 investigation did not show any problems. So maybe I find no problem with the Shelly Plug S too. There is the chance that a WiFi network problem during the convert process occurs. If this happens device is soft bricked.

waldorffsmama, 2024-12-17:
> @waldorffsmama Rebuilded the Shelly i4 firmware (from the leftovers on the device and the orig. Shelly update firmware zip) and flashed on the device. The I4 booted up successfully with Shelly firmware. Dragged & Dropped the Shelly I4 convert zip. Device converted successfully to Tasmota. After Autoconfig converted to safeboot scheme without any problems. Honestly idk what went wrong in your attempts. Maybe wrong conver zip used? The device you sent me is an I4 with an esp32 solo1 (No 3rd gen device).
Amazing. And how did I manage to do the same error twice? Will start with another Shelly I4 from scratch and let you what I am doing step-by-step.

waldorffsmama, 2024-12-18:
@Jason2866  
Here is how I am flashing to Tasmota. Would you mind checking if there is anything wrong or unusual?

1) Connect Shelly i4 to 12V
2) Access AP and enter Wifi information, turn on Wifi
3) Access user interface from new IP address, navigate to Firmware / Upgrade and update to 1.3.3...
4) It reboots
5) Update to 1.4.4 (stable)
6) Now I would download http://ota.tasmota.com/tasmota32/release/tasmota32solo1.bin and drag and drop the file into the "firmware file image" box

I am stopping at this point and would like to ask you to confirm that I have followed the correct procedure. All correct?

Tillomar, 2024-12-18:
Comment from my side:
I DID follow the steps -- however, after step 8 (autokonfiguration) my device DID NOT come up again.

closed 2024-12-18 by Jason2866

Jason2866, 2024-12-18:
@Tillomar You can send me a (bricked) Shelly Plus Plug S for free if you want. I will not send back the device! 
Will try to find the reason for the Shelly Plus Plug S fail attempt in your case.