tasmota/mgos32-to-tasmota32 #92: tasmota32-to-mgos32

noripman, 2024-12-09:
Hi,
I accidentally found a way to flash back to the original Shelly software. (Please test and confirm.)
Shells Plus 1 PM=> Firmware 1.1.4-g6d2a568 
flashed with Tasmota >= mgos32-to-tasmota32-Plus1PM.zip fully configured with
{"NAME":"Shelly Plus 1PM","GPIO":[0,0,0,0,192,2720,0,0,0,0,0,0,0,0,2656,0,0,0,0,2624,0,32,224,0,0,0,0,0,4736,0,0,0,0,0,0,0],"FLAG":0,"BASE":1,"CMND":"AdcParam1 2,10000,10000,3350"}
then go to Tasmota WebIF => Firmware Upgrade and don't enter anything in the OTA URL (or as in my case "Attention! No update")
and after the restart Shelly is back... I had to update several times in some attempts... When I reinstalled Tasmota all the settings were back...
I tested it several times on 2 Shells plus 1 PM...

Jason2866, 2024-12-09:
Not an good idea. You have left out the needed "Autoconfig" step. This steps removes the locked Shelly bootloader and wipes any Shelly leftovers.
Without the replacing the Shelly bootloader the device gets 100% bricked after a few restarts.

noripman, 2024-12-09:
Sorry,
who can read...sorry, I have completed the last steps 8 and 10...
Question: can I now install tasmota32-DE.bin or do I have to continue updating tasmota32solo1.bin?

Jason2866, 2024-12-09:
Ohh, if the steps 8-10 are done and still Shelly comes up, "something" in the Autoconfig step is not working.
If it is a solo1 esp32 with one core only you can not flash a normal esp32 build (with two cores). You can see in Tasmota information page

noripman, 2024-12-09:
No no, I only carried out steps 8-10 after your post. Now the downgrade no longer works...

closed 2024-12-10 by Jason2866

kallejoken, 2024-12-18:
> Not an good idea. You have left out the needed "Autoconfig" step. This steps removes the locked Shelly bootloader and wipes any Shelly leftovers. Without the replacing the Shelly bootloader the device gets 100% bricked after a few restarts.

bricked means that it can no longer be flashed serially?

Jason2866, 2024-12-18:
@kallejoken Flashing via serial is always working.

kallejoken, 2024-12-18:
> @kallejoken Flashing via serial is always working.

Does it matter which tasmota version I have to use? Shelly firmware was 1.4.4 - device is a first generation pm mini.

Jason2866, 2024-12-18:
Use latest Tasmota32 factory image